VeriSilicon/TIM-VX #16: insmod: ERROR: could not insert module galcore.ko: Invalid module format

wuqiangch, 2021-02-24:
AA311D,ubuntu18.04
sudo insmod galcore.ko
insmod: ERROR: could not insert module galcore.ko: Invalid module format

thezha, 2021-02-24:
Please share result of dmesg and `uname -a`.

wuqiangch, 2021-02-24:
@thezha 
Linux Khadas 5.5.0-rc2 #0.8.2 SMP PREEMPT Fri Jan 3 09:48:06 CST 2020 aarch64 aarch64 aarch64 GNU/Linux

thezha, 2021-02-24:
The provided "galcore.ko" is built using Fenix 0.8.1 based on Kernel 4.9.0, so it might not be compatible with 5.5.0 kernel.

Linux Khadas 4.9.206 #13 SMP PREEMPT Tue Dec 31 00:37:47 CST 2019 aarch64 aarch64 aarch64 GNU/Linux

wuqiangch, 2021-02-24:
@thezha 
[tmp.txt](https://github.com/VeriSilicon/TIM-VX/files/6034832/tmp.txt)
Maybe I have to built it myself.

thezha, 2021-02-24:
Have you tried --force option with insmod?

wuqiangch, 2021-02-24:
@thezha I have try it,but failed.

thezha, 2021-02-24:
@wuqiangch The fastest way is to switch to 4.9.0 system image until we can supply a prebuilt-sdk for 5.5.0. You may also have trouble building it yourself since the galcore used by TIM-VX might not match the opensource version you have. Thanks!

wuqiangch, 2021-02-24:
@thezha Ok.Thanks!

kalcohol, 2021-02-25:
@wuqiangch your system need to switch to VIM3_V0.9-20200530(i guess your board is vim3).
download last fireware and flash your vim3, than do apt full-update.
KHADAS VIM3 fireware download page: https://docs.khadas.com/firmware/Vim3Firmware.html
update your board:
```
sudo apt update
sudo apt full-upgrade
```
you can switch to other apt source, such as Ali source(https://developer.aliyun.com/mirror/).
sample is below:
```
deb-src https://mirrors.aliyun.com/ubuntu-ports/ focal main restricted universe multiverse

deb https://mirrors.aliyun.com/ubuntu-ports/ focal-security main restricted universe multiverse
deb-src https://mirrors.aliyun.com/ubuntu-ports/ focal-security main restricted universe multiverse

deb https://mirrors.aliyun.com/ubuntu-ports/ focal-updates main restricted universe multiverse
deb-src https://mirrors.aliyun.com/ubuntu-ports/ focal-updates main restricted universe multiverse

deb https://mirrors.aliyun.com/ubuntu-ports/ focal-proposed main restricted universe multiverse
deb-src https://mirrors.aliyun.com/ubuntu-ports/ focal-proposed main restricted universe multiverse

deb https://mirrors.aliyun.com/ubuntu-ports/ focal-backports main restricted universe multiverse
deb-src https://mirrors.aliyun.com/ubuntu-ports/ focal-backports main restricted universe multiverse
```

other reference: https://github.com/OAID/Tengine/blob/tengine-lite/doc/npu_tim-vx_user_manual.md
our qq group: 829565581.

kalcohol, 2021-03-03:
after remote assistance, issue solved.

wuqiangch, 2021-03-03:
It's solved.
https://docs.khadas.com/firmware/Vim3Firmware.html
Latest system：
VIM3_Ubuntu-gnome-focal_Linux-4.9_arm64_EMMC_V0.9-20200530(Latest release)

closed 2021-03-03 by wuqiangch