VeriSilicon/TIM-VX #24: Is there the way to deal with graph verification errors?

janoslim, 2021-03-12:
Hello, I am using tim-vx for tensorflow lite delegate and tengine,
And I encountered graph verification fail error.
What can I try to do for verification error? what may I check for this error 
Thank you!

thezha, 2021-03-16:
Please enable the following environment variables during runtime, and post full logs here. 

`export VSI_NN_LOG_LEVEL=5`

janoslim, 2021-03-16:
Thank you. Is there another way to check verification error? I cannot get some reason of failing verification from printed logs.
I tried Tengine & tflite vx-delegate both with mobilenet-v2 quantized model provided by official tensorflow.

thezha, 2021-03-18:
Did you set export VIVANTE_SDK_DIR to point to the SDK path at runtime?

janoslim, 2021-03-19:
No I don't set VIVANTE_SDK_DIR but another models works well like vgg.
And What is the sdk path you are said ? I am using libtim-vx.so.

thezha, 2021-03-19:
Some networks require runtime compilation to happen on device, and you need to point VIVANTE_SDK_DIR to the location of the pre-built SDK which you pushed onto the device. In my environment, it looks like something below. 

```
root@Khadas:/home/kainan/vx_delegate/vim3_aarch64# pwd
/home/kainan/vx_delegate/vim3_aarch64
root@Khadas:/home/kainan/vx_delegate/vim3_aarch64# tree -L 2
.
├── include
│   ├── CL
│   └── VX
└── lib
    ├── galcore.ko
    ├── libArchModelSw.so
    ├── libCLC.so
    ├── libGAL.so
    ├── libNNArchPerf.so
    ├── libNNGPUBinary.so
    ├── libNNVXCBinary.so
    ├── libOpenVXC.so
    ├── libOpenVX.so -> libOpenVX.so.1.3.0
    ├── libOpenVX.so.1 -> libOpenVX.so.1.3.0
    ├── libOpenVX.so.1.3.0
    ├── libOpenVXU.so
    ├── libOvx12VXCBinary.so
    └── libVSC.so

4 directories, 14 files
root@Khadas:/home/kainan/vx_delegate/vim3_aarch64# export | grep VIVANTE_SDK_DIR
declare -x VIVANTE_SDK_DIR="/home/kainan/vx_delegate/vim3_aarch64"
```

janoslim, 2021-03-19:
still same issues :(
Could you check if I upload the tflite models that causes problems?

[tflite_quant_models.zip](https://github.com/VeriSilicon/TIM-VX/files/6168329/tflite_quant_models.zip)

Thank you alot.

thezha, 2021-03-19:
I can run all three of this models fine. Did you copy the header files over onto the device? The runtime compiler needs to look for this file, so please make sure it is present on your system.

root@Khadas:/home/kainan/vx_delegate/vim3_aarch64# ls -l include/CL/
total 76
-rw-rw-r-- 1 993 992 76691 Mar 17 16:50 cl_viv_vx_ext.h

janoslim, 2021-03-19:
It's hard to find what I missed.
It's not a network problem that all networks work well on your device.
Did you test test them on tengine?
I tried tengine and tflite-vxdelegate and they all failed.

```
khadas@Khadas:~/tengine-lite/3rdparty/tim-vx$ tree -L 3
.
├── include
│   ├── CL
│   │   └── cl_viv_vx_ext.h
│   ├── tim
│   │   └── vx
│   └── VX
│       ├── viv_nn_compatibility.h
│       ├── vx_api.h
│       ├── vx_compatibility.h
│       ├── vx_ext_program.h
│       ├── vx_ext_target.h
│       ├── vx.h
│       ├── vx_helper.h
│       ├── vx_kernels.h
│       ├── vx_khr_cnn.h
│       ├── vx_khr_compatible.h
│       ├── vx_khr_dot.h
│       ├── vx_khr_icd.h
│       ├── vx_khr_import_kernel.h
│       ├── vx_khr_interp.h
│       ├── vx_khr_nn.h
│       ├── vx_khr_nn_internal.h
│       ├── vx_khr_node_memory.h
│       ├── vx_khr_opencl.h
│       ├── vx_khr_tiling.h
│       ├── vx_khr_variants.h
│       ├── vx_khr_xml.h
│       ├── vx_lib_debug.h
│       ├── vx_lib_extras.h
│       ├── vx_lib_xyz.h
│       ├── vx_nodes.h
│       ├── vx_types.h
│       ├── vxu.h
│       ├── vx_vendors.h
│       └── vx_viv_sys.h
└── lib
    ├── aarch64
    ├── galcore.ko
    ├── libArchModelSw.so
    ├── libCLC.so
    ├── libGAL.so
    ├── libNNArchPerf.so
    ├── libNNGPUBinary.so
    ├── libNNVXCBinary.so
    ├── libOpenVXC.so
    ├── libOpenVX.so.1.3.0
    ├── libOpenVXU.so
    ├── libOvx12VXCBinary.so
    └── libVSC.so

7 directories, 42 files
```

thezha, 2021-03-19:
It seems that ldd found these libraries in /lib, but your libraries are in ~/tengine-lite/3rdparty/tim-vx/lib

janoslim, 2021-03-19:
I push them both ~/tengine-lite/3rdparty/tim-vx/lib and /lib.
Does this can make the problem? However if it is the library error, every model would not works well but my vgg19 model works fine.

janoslim, 2021-03-28:
I fixed with os reset. thank you for your response.

closed 2021-03-28 by janoslim