VeriSilicon/TIM-VX #33: android for arm 32 bit

watchyou, 2021-04-20:
Do you have any plans to support Android (arm 32 bit)

thezha, 2021-04-22:
We target for Android support by end of May 2021. Thanks!

stevenwudi, 2021-04-24:
Yes, we the support of Android (arm 32 bit) will be quite essential for our team project.
Looking forward to the release

thezha, 2021-04-28:
Please let us know the exact Android SDK you are using? Is it from official VIM3? https://docs.khadas.com/vim3/DownloadAndroidSourceCode.html

stevenwudi, 2021-04-28:
We customised our android system based on Android 10. It is not from Khadas provided system.
Does the Android SDK matters in terms of TIMVX integration?

thezha, 2021-04-28:
Android SDK itself doesn't matter that much, but we need to know the kernel driver version.

stevenwudi, 2021-04-29:
Our kernel version is 4.9
Looking forward to hearing back any progress :)

stevenwudi, 2021-04-29:
My wechat ID is: stevenwudibbb
If it's ever needed for expedite the process of anything unclear.
Best regards

stevenwudi, 2021-04-29:
I am wondering if there is a certain release date for the support of Android system(arm 32 bit). Our team leader is actively thinking about alternatives for VSI  from Amlogic(which of course I personally didn't want another NPU provider given our team has spent quite some effort in using TIMVX for project development).
Really looking forward to the support of android 32 bit asap...

thezha, 2021-05-08:
Resolved with [SHA03a53](https://github.com/VeriSilicon/TIM-VX/commit/03a53cba7e90948b4b1cb59634d9f871d545d625)

mkdir build; cd build
cmake -DCMAKE_TOOLCHAIN_FILE=android-ndk-r21e/build/cmake/android.toolchain.cmake ..
make

closed 2021-05-10 by thezha