VeriSilicon/TIM-VX #105: Segmentation fault

bkovalenkoyandex, 2021-06-07:
Hi, I have A311D board with ubuntu on board (from khadas). My end goal is to compile tflite with vx-delegate support. What is the best way to do this? 

Also Im trying to compile TIM-VX (libtim-vx.so) it seems Bazel building is broken, so I try CMake. CMake works fine: compiles and links all targets, but when I try to run unit tests (under gdb), I get segfault from `*** _LoadStates() from libGAL.so` what causes this?

bkovalenkocomp, 2021-06-07:
when Im trying to build tf with vx-delegate I also get errors:

on x86:
```
ERROR: /home//vx-delegate/tensorflow/lite/delegates/vx-delegate/BUILD.bazel:3:1: C++ compilation of rule '//tensorflow/lite/delegates/vx-delegate:vx_delegate' failed (Exit 1)
tensorflow/lite/delegates/vx-delegate/delegate_main.cc:32:10: fatal error: tim/transform/layout_inference.h: No such file or directory
 #include "tim/transform/layout_inference.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

on A311D:
```
/aarch64_linux_toolchain/bin/aarch64-linux-gnu-gcc: 1: Syntax error: "(" unexpected
```

liyuenan2333, 2021-06-08:
@bkovalenkocomp For bazel build, we don't update the tim-vx in vx-delegate yet, so we suggest you to clone TIM-VX manually.
you can enable your local TIM-VX by modifying file `third_party/tim_vx/workspace.bzl`. Please make sure your local TIM-VX is the latest.
Let us know if you have any problem. Thanks.

bkovalenkocomp, 2021-06-08:
> @bkovalenkocomp For bazel build, we don't update the tim-vx in vx-delegate yet, so we suggest you to clone TIM-VX manually.
> you can enable your local TIM-VX by modifying file [...]

I compiled TIM-VX using CMake, then I fix paths in TFLite CMake files and compile it with vx-delegate ON, using CMake. During linking of target `benchmark_model` I get errors:
```
*/libOpenVX.so.1: undefined reference to gcFreeCLPatchLibrary
*/libOpenVX.so.1: undefined reference to gcSHARED_LoadKernel
*/libOpenVX.so.1: undefined reference to  gcSHARED_FreeRecompilerLibrary
```
and few more...

it seems libOpenVX.so from https://github.com/VeriSilicon/TIM-VX/blob/main/cmake/A311D.cmake#L18 doesn't have this symbols. How do I fix this? Thanks.

sunshinemyson, 2021-06-08:
> 
> 
> > @bkovalenkocomp For bazel build, we don't update the tim-vx in vx-delegate yet, so we suggest you to clone TIM-VX manually.
> > you can enable your local TIM-VX by modifying file `third_party/tim_vx/workspace.bzl`. Please make sure your local TIM-VX is the latest.
> > Let us know if you have any problem. Thanks.
> 
> I compiled TIM-VX using CMake, then I fix paths in TFLite CMake files and compile it with vx-delegate ON, using CMake. During linking of target `benchmark_model` I get errors:
> 
> ```
> */libOpenVX.so.1: undefined reference to gcFreeCLPatchLibrary
> */libOpenVX.so.1: undefined reference to gcSHADER_LoadKernel
> */libOpenVX.so.1: undefined reference to  gcSHADER_FreeRecompilerLibrary
> ```
> 
> and few more...
> 
> it seems libOpenVX.so from https://github.com/VeriSilicon/TIM-VX/blob/main/cmake/A311D.cmake#L18 doesn't have this symbols. How do I fix this @liyuenan2333 ? Thanks.

Those symbol is not in libOpenVX.so, could you try to add libGAL.so/libVSC.so/libCLC.so  to the linker input?

bkovalenkocomp, 2021-06-08:
ok, I figured I have 2 sets of libs needed by linker. System libs and from https://github.com/VeriSilicon/TIM-VX/blob/main/cmake/A311D.cmake#L18

Im trying to build libtim-vx.so, everything compiles and links, but when I try to run `unit_test` I get segmentation fault:

If I use system libs - `vxCreateProgramWithBinary () from /lib/libOpenVX.so`

If I use provided libs - `_LoadStates () from lib/libGAL.so`

also, when I use system libs, I can compile tflite with vx-delegate, otherwise I get error above (during linking)

I probably have incompatible libs, or wrong versions? Could give me a hint please?

thezha, 2021-06-08:
BTW, a change has been pushed to the Tensorflow fork to update TIM-VX to 1.1.30.3. 
https://github.com/VeriSilicon/tensorflow/commit/25ae27ea7e6e7ce52bf16bf1a4754256d4e712c6

With this change, you should be able to build using 

`bazel build --config A311D //tensorflow/lite/tools/benchmark:benchmark_model`

bkovalenkocomp, 2021-06-09:
> BTW, a change has been pushed to the Tensorflow fork to update TIM-VX to 1.1.30.3.
> [VeriSilicon/tensorflow@25ae27e](https://github.com/VeriSilicon/tensorflow/commit/25ae27ea7e6e7ce52bf16bf1a4754256d4e712c6)
> 
> With this change, you should be able to build using
> 
> `bazel build --config A311D //tensorflow/lite/tools/benchmark:benchmark_model`

thanks! I confirm this fixes build and I can build libs on x86 for A311D using Bazel.

I almost got it working, not I get error:

```
gcSHADER_LoadHeader: shader library files's version of 0.0.1:56 is not compatible with current version 0.0.1:57
Please recompile source.

*** Failed to verify graph
Aborted
```

thezha, 2021-06-09:
It's probably because the libraries on your A311D device is incompatible with the libraries used for cross compilation. Please try take the contents from this archive and replace the libraries on the devices. 

https://github.com/VeriSilicon/TIM-VX/releases/download/v1.1.28/aarch64_A311D_D312513_A294074_R311680_T312233_O312045.tgz

bkovalenkocomp, 2021-06-09:
> It's probably because the libraries on your A311D device is incompatible with the libraries used for cross compilation. Please try take the contents from this archive and replace the libraries on [...]

when I use provided libs I get segfault: ` _LoadStates () from lib/libGAL.so`

thezha, 2021-06-09:
Please run `ldd` on the device and lets see where the dynamic libraries are located.

bkovalenkocomp, 2021-06-09:
> Please run `ldd benchmark_model` on the device and lets see where the dynamic libraries are located.

```
ldd benchmark_model 
        linux-vdso.so.1 (0x0000007fb629e000)
        libArchModelSw.so => not found
        libCLC.so => not found
        libGAL.so => not found
        libNNArchPerf.so => not found
        libNNGPUBinary.so => not found
        libNNVXCBinary.so => not found
        libOpenVX.so.1 => not found
        libOpenVXC.so => not found
        libOpenVXU.so => not found
        libOvx12VXCBinary.so => not found
        libVSC.so => not found
        libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007fb619b000)
        libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007fb6187000)
        libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007fb6157000)
        librt.so.1 => /lib/aarch64-linux-gnu/librt.so.1 (0x0000007fb613f000)
        libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007fb5f5c000)
        libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007fb5f37000)
        libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007fb5dc4000)
        /lib/ld-linux-aarch64.so.1 (0x0000007fb626e000)
```
I run with command: `LD_LIBRARY_PATH=lib ./benchmark_model --graph=mobilenet_v1_1.0_224_quant.tflite --use_vxdelegate=true`

thezha, 2021-06-09:
Try export LD_LIBRARY_PATH=/home/khadas/TIM-VX/build/aarch64_A311D_D312513_A294074_R311680_T312233_O312045/lib and run ldd again to see if it finds the correct library. 

Also, you need to use the galcore.ko file from the lib as well, instead of using the galcore.ko supplied by the board.
`rmmod galcore; insmode galcore.ko`

bkovalenkocomp, 2021-06-09:
after `rmmod galcore; insmod galcore.ko` I get:

```
(gdb) run
Starting program: /home/khadas/benchmark_model --graph=mobilenet_v1_1.0_224_quant.tflite --use_vxdelegate=true
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
STARTING!
Log parameter values verbosely: [0]
Graph: [mobilenet_v1_1.0_224_quant.tflite]
Use VXdelegate : [1]
Loaded model mobilenet_v1_1.0_224_quant.tflite
2021-06-09 09:53:32.699717: I tensorflow/lite/delegates/vx-delegate/op_map.cc:1876] Initialize supported_builtins
2021-06-09 09:53:32.699870: I tensorflow/lite/delegates/vx-delegate/delegate_main.cc:390] vx_delegate Delegate::Init
2021-06-09 09:53:32.699897: I tensorflow/lite/delegates/vx-delegate/op_map.cc:1876] Initialize supported_builtins
2021-06-09 09:53:32.699954: I tensorflow/lite/delegates/vx-delegate/delegate_main.cc:470] Delegate::Prepare node:0xa5cae0
Explicitly applied VX Delegate delegate, and the model graph will be completely executed by the delegate.
The input model file size (MB): 4.27635
Initialized session in 1.264ms.
Running benchmark for at least 1 iterations and at least 0.5 seconds but terminate if exceeding 150 seconds.
2021-06-09 09:53:32.702711: I tensorflow/lite/delegates/vx-delegate/delegate_main.cc:477] Delegate::Invoke node:0xa5cae0
[     1] Failed to open device: No such file or directory, Try again...
[     2] Failed to open device: No such file or directory, Try again...
[     3] Failed to open device: No such file or directory, Try again...
[     4] Failed to open device: No such file or directory, Try again...
[     5] _OpenDevice(1130): FATAL: Failed to open device, errno=No such file or directory.
[Inferior 1 (process 175033) exited with code 01]

```

thezha, 2021-06-09:
Looks like your rmmod failed, please share some logs from dmesg and uname -a

bkovalenkocomp, 2021-06-09:
> Looks like your insmod failed, please share some logs from dmesg and uname -a

```
[    4.708012] BTRFS: selftest: Free space cache tests finished
[    4.708013] BTRFS: selftest: Running extent buffer operation tests
[    4.708015] BTRFS: selftest: Running btrfs_split_item tests
[    4.708029] BTRFS: selftest: Running extent I/O tests
[    4.708030] BTRFS: selftest: Running find delalloc tests
[    4.818516] BTRFS: selftest: Running extent buffer bitmap tests
[    4.869461] BTRFS: selftest: Extent I/O tests finished
[    4.869464] BTRFS: selftest: Running btrfs_get_extent tests
[    4.869573] BTRFS: selftest: Running hole first btrfs_get_extent test
[    4.869590] BTRFS: selftest: Running outstanding_extents tests
[    4.869622] BTRFS: selftest: Running qgroup tests
[    4.869624] BTRFS: selftest: Qgroup basic add
[    4.869658] BTRFS: selftest: Qgroup multiple refs test
[    4.869707] BTRFS: selftest: Running free space tree tests
[    4.891181] BTRFS: selftest: sectorsize: 4096  nodesize: 65536
[    4.891184] BTRFS: selftest: Running btrfs free space cache tests
[    4.891188] BTRFS: selftest: Running extent only tests
[    4.891199] BTRFS: selftest: Running bitmap only tests
[    4.891209] BTRFS: selftest: Running bitmap and extent tests
[    4.891222] BTRFS: selftest: Running space stealing from bitmap to extent
[    4.891543] BTRFS: selftest: Free space cache tests finished
[    4.891545] BTRFS: selftest: Running extent buffer operation tests
[    4.891546] BTRFS: selftest: Running btrfs_split_item tests
[    4.891563] BTRFS: selftest: Running extent I/O tests
[    4.891564] BTRFS: selftest: Running find delalloc tests
[    5.001919] BTRFS: selftest: Running extent buffer bitmap tests
[    5.052821] BTRFS: selftest: Extent I/O tests finished
[    5.052824] BTRFS: selftest: Running btrfs_get_extent tests
[    5.052935] BTRFS: selftest: Running hole first btrfs_get_extent test
[    5.052954] BTRFS: selftest: Running outstanding_extents tests
[    5.052987] BTRFS: selftest: Running qgroup tests
[    5.052989] BTRFS: selftest: Qgroup basic add
[    5.053025] BTRFS: selftest: Qgroup multiple refs test
[    5.053076] BTRFS: selftest: Running free space tree tests
[   10.467401] EXT4-fs (mmcblk0p6): mounted filesystem with writeback data mode. Opts: data=writeback
[   10.673940] systemd[1]: System time before build time, advancing clock.
[   10.690742] cgroup: cgroup2: unknown option "nsdelegate"
[   10.696772] systemd[1]: systemd 245.4-4ubuntu3.6 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD +IDN2 -IDN +PCRE2 default-hierarchy=hybrid)
[   10.697449] systemd[1]: Detected architecture arm64.
[   10.704685] systemd[1]: Set hostname to <Khadas>.
[   10.979966] systemd[1]: Created slice system-modprobe.slice.
[   10.980440] systemd[1]: Created slice system-serial\x2dgetty.slice.
[   10.980785] systemd[1]: Created slice User and Session Slice.
[   10.980892] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[   10.981342] systemd[1]: Set up automount Arbitrary Executable File Formats File System Automount Point.
[   10.981434] systemd[1]: Reached target User and Group Name Lookups.
[   10.981461] systemd[1]: Reached target Remote File Systems.
[   10.981479] systemd[1]: Reached target Slices.
[   10.981505] systemd[1]: Reached target Swap.
[   10.981523] systemd[1]: Reached target System Time Set.
[   10.981756] systemd[1]: Listening on Syslog Socket.
[   10.982019] systemd[1]: Listening on fsck to fsckd communication Socket.
[   10.982117] systemd[1]: Listening on initctl Compatibility Named Pipe.
[   10.982421] systemd[1]: Listening on Journal Audit Socket.
[   10.982566] systemd[1]: Listening on Journal Socket (/dev/log).
[   10.982757] systemd[1]: Listening on Journal Socket.
[   10.982928] systemd[1]: Listening on udev Control Socket.
[   10.983039] systemd[1]: Listening on udev Kernel Socket.
[   10.984789] systemd[1]: Mounting Huge Pages File System...
[   10.986707] systemd[1]: Mounting POSIX Message Queue File System...
[   10.988638] systemd[1]: Mounting Kernel Debug File System...
[   10.991052] systemd[1]: Mounting Kernel Trace File System...
[   10.993912] systemd[1]: Starting Journal Service...
[   10.995998] systemd[1]: Started Camera ISP 3A server service.
[   11.006498] systemd[1]: Starting Restore / save the current clock...
[   11.008476] systemd[1]: Starting Set the console keyboard layout...
[   11.014409] systemd[1]: Starting Create list of static device nodes for the current kernel...
[... 981 lines cut ...]
[   23.006128] hdmitx: hw: set channel status
[   23.006157] hdmitx: audio: Audio Type: PCM
[   23.006159] hdmitx: audio: set audio param
[   23.006162] aml_spdif_fifo_ctrl, bit depth:16, frddr type:1, toddr:type:1
[   23.020131] asoc-aml-card auge_sound: S/PDIF Playback enable
[   24.449952] meson_uart ffd24000.serial: ttyS1 use xtal(24M) 24000000 change 115200 to 2000000
[   26.220451] BT_RADIO going: on
[   26.220456] AML_BT: going ON
[   28.077143] asoc-aml-card auge_sound: S/PDIF Playback disable
[   28.077225] spdif_b keep clk continuous
[   28.077230] aml_spdif_close
[   28.077301] audio_ddr_mngr: frddrs[0] released by device ff642000.audiobus:spdif_b
[   48.420635] fb: osd_release now.index=1,open_count=1
[   48.420643] fb: osd_release now.index=2,open_count=1
[   48.420647] fb: osd_release now.index=3,open_count=1
[   51.347293] IPVS: Creating netns size=1928 id=4
[   52.481383] IPVS: Creating netns size=1928 id=5
[   54.099145] IPVS: Creating netns size=1928 id=6
[   54.111033] IPVS: Creating netns size=1928 id=7
[   62.490204] fb: mem_free_work, free memory: addr:800000
[   62.670120] IPVS: Creating netns size=1928 id=8
```
```
Linux Khadas 4.9.241 #20 SMP PREEMPT Thu May 20 18:06:42 CST 2021 aarch64 aarch64 aarch64 GNU/Linux
```

thezha, 2021-06-09:
After you run insmod galcore.ko, you should see something like this, please share this portion of the log.

```
[1564964.318490] npu_version: 0
[1564964.318493] not find power_version
[1564964.318525] galcore irq number is 36.
[1564964.318531] Galcore version 6.4.3.279124+
```

bkovalenkocomp, 2021-06-09:
> After you run insmod galcore.ko, you should see something like this, please share this portion of the log.
> 
> ```
> [1564964.318490] npu_version: 0
> [1564964.318493] not find power_version
> [1564964.318525] galcore irq number is 36.
> [1564964.318531] Galcore version 6.4.3.279124+
> ```

I think it works now. I rebooted and did `rmmod galcore; insmod galcore.ko`

Thank you very much guys!!!

closed 2021-06-09 by bkovalenkocomp