VeriSilicon/TIM-VX #107: Some queries about running tflite

rniranjan93, 2021-06-09:
1) What is the format of "TIM-VX/main/samples/lenet/lenet_asymu8_weights.h" this data, can it be considered as tflite file.
2) What is the best way to run tflite on verisilicon board, should we convert tflite to .nbg file and then use tim::lite to run that .nbg file.If yes, how to convert tflite to .nbg file?
3)Or using tflite library should we extract the weigths, and type of network ourseleves and using that information, should we build and run tflite using tim::vx.

rniranjan93, 2021-06-09:
How to use this delegate (Tensorflow-Lite Delegate (Unofficial) https://github.com/VeriSilicon/tensorflow)

thezha, 2021-06-09:
**What is the format of "TIM-VX/main/samples/lenet/lenet_asymu8_weights.h" this data, can it be considered as tflite file.**
It is not a tflite file, this is essentially the coefficient of a network.

**What is the best way to run tflite on verisilicon board, should we convert tflite to .nbg file and then use tim::lite to run that .nbg file.If yes, how to convert tflite to .nbg file?**
The best way is to use the tflite delegate https://github.com/VeriSilicon/tensorflow, Please try out the instructions below. 
```

# clone and cross build VeriSilicon tensorflow fork with TFlite delegate support
git clone --single-branch --branch vx-delegate.v2.4.1 git@github.com:VeriSilicon/tensorflow.git vx-delegate; cd vx-delegate
bazel build --config A311D //tensorflow/lite/tools/benchmark:benchmark_model

# push benchmark_model onto device and run
./benchmark_model --graph=mobilenet_v1_1.0_224_quant.tflite --use_vxdelegate=true
```

closed 2021-06-18 by thezha