VeriSilicon/TIM-VX #117: Can I create multi Contexts and graphs?

janoslim, 2021-06-17:
Hello,

I am using TIM-VX for running multi NN.
I'm referring to the sample code and I wonder what is the best solution for creating multiple nn in terms of performance?
I think about 
1) creating single Context multi graph,
2) creating multi graph and context each.

Thank you!

sunshinemyson, 2021-06-18:
Single Context with Multi Graph is prefered.
You can split the whole graph into multigraph with single layer, 100 graph is ok.

janoslim, 2021-06-18:
Hi @sunshinemyson,
Thank you for your reply.
I just have one more question that
I wonder creating tim::vx::tensor for input and output(not transient) allocate memory for activation? it might be memory-heavy job  for embedded boatd to split layers.

closed 2021-06-20 by janoslim