VeriSilicon/TIM-VX #119: compile NBG failed

janoslim, 2021-06-20:
Hello,

I followed source code defined at graph_test.cc that use NBG,
However I failed to compile nbg file.
I wonder my creating nbg file code is wrong. Thank you.

Compile as binary and save.
```
size_t bin_size = 0;

graph->CompileToBinary(nullptr, &bin_size); // bin size A  
std::vector<char> nbg_buf(bin_size); 

this->graph->CompileToBinary(nbg_buf.data(), &bin_size);  // bin size B
std::string file_path = this->makeChunkPath(nbgdir); 

std::ofstream FILE(file_path, std::ofstream::binary);
for (auto byte : nbg_buf)
    FILE << byte;

// while bin size A and B is slightly different, I wonder it does not matter.
```

Read NBG and compile

```
// read file size
FILE* nbg = NULL;    
nbg = fopen(file_path.c_str(), "rb");
fseek(nbg, 0, SEEK_END);
long rst = ftell(nbg);

fseek(nbg, 0, SEEK_SET);

this->nbg_buf = std::vector<char>(rst);

while((read_size = fread(nbg_buf.data(), 1, 4096, nbg)) > 0)
{ }

std::vector<std::shared_ptr<tim::vx::Tensor>> iVXTensors;
std::vector<std::shared_ptr<tim::vx::Tensor>> oVXTensors;
// Create tensors...
...
...
//

//  make f auto nbg_node = this->graph->CreateOperation<tim::vx::ops::NBG>(
  (nbg_buf.data()), /*num_of_input*/ iVXTensors.size(), /*num_of_output*/ oVXTensors.size());

(*nbg_node).BindInputs(iVXTensors).BindOutputs(oVXTensors);

iVXTensors[0]->CopyDataToTensor(nbg_buf.data(), shape);

std::cout << "Compile start." << std::endl;
this->graph->Compile();
// This goes seg fault...
```

sunshinemyson, 2021-06-21:
Hi Janoslim,

Could you share your NBG file and the backtrace for analysis? Thanks

janoslim, 2021-06-22:
[vgg19.zip](https://github.com/VeriSilicon/TIM-VX/files/6691054/vgg19.zip)
@sunshinemyson Thank you. I tried with this file.

sunshinemyson, 2021-07-08:
Hi janoslim,

Sorry for late reply. I just checked the NBG file, the nbg file is correct.

Did you try to run this nbg with your dev-board or not?

janoslim, 2021-07-10:
@sunshinemyson I tested with vim3 but it just failed to load.
Could I get a sample source code to use nbg file from timvx?
Thank you!

sunshinemyson, 2021-07-12:
@janoslim ,

We have unit test case for NBG in graph_test.cc.  You can take it as an example. I double checked your source code, it seems like correct.

Could you share the log while generate NBG with setup ENV VIV_VX_DEBUG_LEVEL=1 ? Thanks

closed 2021-07-22 by sunshinemyson