VeriSilicon/TIM-VX #121: How to profile network on vx-delegate?

bkovalenkocomp, 2021-06-22:
It seems with my graph I get following numbers (on A311D):

```
./benchmark_model --graph=model.tflite --use_vxdelegate=true                                            : Inference (avg): **191704**
./benchmark_model --graph=model.tflite --use_vxdelegate=false --num_threads=4            : Inference (avg): **48060.7**
```

the model is converted with `converter.optimizations = [tf.lite.Optimize.DEFAULT]` and with `representative_dataset`. Model size is around 10 Mb.

That's quite a difference, I expected much higher speedup. Is it possible Im doing something wrong? How to profile each op in the graph?

btw `./benchmark_model --graph=mobilenet_v1_1.0_224_quant.tflite --use_vxdelegate=true` is `Inference (avg): 6424.51`

sunshinemyson, 2021-06-23:
Yes, you can get profile information by "export VIV_VX_DEBUG_LEVEL=1 export CNN_PERF=1", you can find per layer performance from the log. 
Would you mind share the model also?

bkovalenkocomp, 2021-06-23:
> Yes, you can get profile information by "export VIV_VX_DEBUG_LEVEL=1 export CNN_PERF=1", you can find per layer performance from the log.
> Would you mind share the model also?

unfortunately I can't share the network, but here is part of log from debug run 
[per_op_log.log](https://github.com/VeriSilicon/TIM-VX/files/6702233/per_op_log.log)

if you need other info from the debug run, please ask.

thezha, 2021-06-23:
Is this a floating point model? From the log, it seems that this model is not quantized.

A311D contains an integer only engine of the NPU core, any none-quantized model will not yield good performances. Thanks!

bkovalenkocomp, 2021-06-23:
> Is this a floating point model? From the log, it seems that this model is not quantized.
> 
> A311D contains an integer only engine of the NPU core, any none-quantized model will not yield good performances. Thanks!

it's supposed to be integer model, I got it by converting with 

```
converter.optimizations = [tf.lite.Optimize.DEFAULT]
converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS_INT8]
converter.representative_dataset = representative_dataset
```

The graph contains dequantize ops and there is boost in tflite inference with xnnpack delegate.

input and output are float32 though, should they also be forced into int8?

thezha, 2021-06-23:
This is because the model is quantized using Per Channel Quantization method which is not optimized on this old version of the NPU core. On A311D, to get optimal performance, you will need to use Per Tensor Quantization.(mobilenet_v1_1.0_224_quant.tflite uses this method). 

Would you be able to quantized it using PerTensor Quantization method? You might need to use the tools in Tensorflow 1.x. Thanks!

bkovalenkocomp, 2021-06-29:
Confirm, with per tensor quantization I get around 10 ms, which is awesome.

FYI to control quantization type you can use flag:
https://github.com/tensorflow/tensorflow/blob/2c4ed56aca5becc6fd51347e694f151f8c46e8fd/tensorflow/lite/python/lite.py#L454

thanks!

closed 2021-06-29 by bkovalenkocomp