VeriSilicon/TIM-VX #122: Elemwise not work well

janoslim, 2021-06-22:
Hello, I wrote a basic program to verify npu is working, however it seems not work.
Could you notice me what i missed with it? Thank you!

```
int main(int argc, char** argv) {
  (void)argc, (void)argv;
  auto context = tim::vx::Context::Create();
  auto graph = context->CreateGraph();

  tim::vx::ShapeType input_shape({5});
  tim::vx::TensorSpec input_spec(tim::vx::DataType::FLOAT32, input_shape,
                                 tim::vx::TensorAttribute::INPUT);
  auto input = graph->CreateTensor(input_spec);

  tim::vx::ShapeType output_shape({5});
  tim::vx::TensorSpec output_spec(tim::vx::DataType::FLOAT32, output_shape,
                                  tim::vx::TensorAttribute::OUTPUT);
  auto output = graph->CreateTensor(output_spec);

  float temp_data[] = {1.0, 2.0, 3.0, 4.0, 5.0};

  tim::vx::ShapeType conv1_bias_shape({5});
  tim::vx::TensorSpec conv1_bias_spec(tim::vx::DataType::FLOAT32,
                                      conv1_bias_shape,
                                      tim::vx::TensorAttribute::CONSTANT);
  auto conv1_bias = graph->CreateTensor(conv1_bias_spec, temp_data);

  auto mul = graph->CreateOperation<tim::vx::ops::Multiply>();
  (*mul).BindInputs({input, conv1_bias}).BindOutputs({output});

  float* temp_input = (float*)malloc(sizeof(float) * 5);
  float* temp_output = (float*)malloc(sizeof(float) * 5);

  for (int i = 0; i < 5; i++) {
    temp_input[i] = 2;
  }

  for (int i = 0; i < 5; i++) {
    temp_output[i] = 0;
  }

  if (!input->CopyDataToTensor(temp_input, sizeof(float) * 5)) {
    std::cout << "copy to tensor failed" << std::endl;
  }
  if (!graph->Run()) {
    std::cout << "Failed to run" << std::endl;
  }

  if (!output->CopyDataFromTensor(temp_output)) {
    std::cout << "Copy from tensor failed" << std::endl;
  }

  printf("after output.\n");
  for (int i = 0; i < 5; i++) printf("%f,", temp_output[i]);

  return 0;
}
```

I expect 2, 4, 5, 8, 10 but it returns all 0 :(

sunshinemyson, 2021-06-22:
Hi janoslim,

Your code is correct. I can get the expected result on my test board(not vim3). I guess you didn't setup driver correctly. could you share log with "export VSI_NN_LOG_LEVEL=5" ?

janoslim, 2021-06-22:
It works fine after changing ovx version :) Thanks alot!

closed 2021-06-22 by janoslim