VeriSilicon/TIM-VX #123: Simple tflite with vx-delegate example

bkovalenkocomp, 2021-06-22:
Hi, could you provide minimal example, for tflite inference with vx-delegate applied? 
Im not sure how to create and apply vx-delegate to tflite model. Thanks!

sunshinemyson, 2021-06-23:
Yes, we are working on this. The demo will be avaible two weeks later.

bkovalenkocomp, 2021-06-29:
> Yes, we are working on this. The demo will be avaible two weeks later.

looking forward for the example 👍

rniranjan93, 2021-07-21:
I have a .tflite file , I want to build a graph with Delegate class and want to return the graph using the following method
`std::shared_ptr<tim::vx::Graph>& GetGraph() { return graph_; }`
could you provide an example from beginning, i.e, taking .tflite as input -> returning a builded tim::vx::Graph

sunshinemyson, 2021-07-28:
> I have a .tflite file , I want to build a graph with Delegate class and want to return the graph using the following method
> `std::shared_ptr<tim::vx::Graph>& GetGraph() { return graph_; }`
> could you [...]

Not understand your request fully, are you saying use tflite import the tflite file and get compiled vx graph object? Not a demo run tflite model with tflite+vx-delegate? Thanks

rniranjan93, 2021-07-29:
> Not understand your request fully, are you saying use tflite import the tflite file and get compiled vx graph object? Not a demo run tflite model with tflite+vx-delegate? Thanks

Yes

sunshinemyson, 2021-08-02:
What's the purpose for this usage? It's quite strange to me if you want use tflite interpreter to extract model but don't want to inference it with tflite interpreter.

rniranjan93, 2021-08-02:
Wanted to manipulate the graph

sunshinemyson, 2021-08-03:
> 
> 
> Wanted to manipulate the graph

If you want ot manipulate the vx_graph, you have to hack the vx-delegate code. We can not have the vx-delegate graph object with tflite intepreter directly. What kinds of manipulation you want to try with vx_graph?

rniranjan93, 2021-08-05:
No need for it now, thanks.
Could you provide minimal example, for tflite inference with vx-delegate applied?
Im not sure how to create and apply vx-delegate to tflite model. Thanks!

sunshinemyson, 2021-08-07:
> 
> 
> No need for it now, thanks.
> Could you provide minimal example, for tflite inference with vx-delegate applied?
> Im not sure how to create and apply vx-delegate to tflite model. Thanks!

https://github.com/VeriSilicon/tflite-vx-delegate/pull/3 , I'll merge this PR soon. In this PR, I created demo for python/c++ with vx_delegate.

bkovalenkocomp, 2021-08-16:
thanks for the help!

closed 2021-08-16 by bkovalenkocomp