VeriSilicon/TIM-VX #130: tf.transpose segfault

bkovalenkocomp, 2021-06-30:
It seems, that if I try to run an INT8 graph that has tf.transpose op in it, I get segfault.

device: A311D
backend: TFLite with vx-delegate

sunshinemyson, 2021-06-30:
Could you share backtrace for the segfault?

bkovalenkocomp, 2021-06-30:
```
...
2021-06-30 12:56:04.580070: I tensorflow/lite/delegates/vx-delegate/op_map.cc:405] Creating Dequantize op
2021-06-30 12:56:04.580137: I tensorflow/lite/delegates/vx-delegate/op_map.cc:405] Creating Dequantize op
2021-06-30 12:56:04.580203: I tensorflow/lite/delegates/vx-delegate/op_map.cc:405] Creating Dequantize op
2021-06-30 12:56:04.580270: I tensorflow/lite/delegates/vx-delegate/op_map.cc:405] Creating Dequantize op
2021-06-30 12:56:04.580335: I tensorflow/lite/delegates/vx-delegate/op_map.cc:405] Creating Dequantize op
2021-06-30 12:56:04.580402: I tensorflow/lite/delegates/vx-delegate/op_map.cc:405] Creating Dequantize op
2021-06-30 12:56:04.580431: I tensorflow/lite/delegates/vx-delegate/delegate_main.cc:570] Verifying graph
W [HandleLayoutInfer:257]Op 19: Default layout inference pass.
E [GetMapedTensor:140]Tensor has not beed inserted in tensor map.
E [GetMapedTensor:140]Tensor has not beed inserted in tensor map.
E [GetMapedTensor:140]Tensor has not beed inserted in tensor map.
E [GetMapedTensor:140]Tensor has not beed inserted in tensor map.
E [GetMapedTensor:140]Tensor has not beed inserted in tensor map.
E [GetMapedTensor:140]Tensor has not beed inserted in tensor map.
E [GetMapedTensor:140]Tensor has not beed inserted in tensor map.
E [GetMapedTensor:140]Tensor has not beed inserted in tensor map.
E [GetMapedTensor:140]Tensor has not beed inserted in tensor map.
E [GetMapedTensor:140]Tensor has not beed inserted in tensor map.
E [vsi_nn_SetupGraph:670]CHECK STATUS(-10:The supplied parameter information does not match the kernel contract.)
2021-06-30 12:56:04.588468: I tensorflow/lite/delegates/vx-delegate/delegate_main.cc:579] Verified graph
2021-06-30 12:56:04.588552: I tensorflow/lite/delegates/vx-delegate/delegate_main.cc:585] Copying input 595:data:0
2021-06-30 12:56:04.589129: I tensorflow/lite/delegates/vx-delegate/delegate_main.cc:598] Invoking graph
2021-06-30 12:56:04.589595: I tensorflow/lite/delegates/vx-delegate/delegate_main.cc:605] Copying output 596:transpose_6:0

Thread 1 "benchmark_model" received signal SIGSEGV, Segmentation fault.
0x000000000060efbc in ?? ()
```

sunshinemyson, 2021-06-30:
Thanks. I think i know where is the problem. Let me ask the developer to fix this issue.
Root cause: we didn't add layout inference for transpose op yet.

sunshinemyson, 2021-07-07:
Hi， 

We fixed issue with transpose in the layout inference framework. Please try with latest commit. Thanks

closed 2021-07-15 by thezha