VeriSilicon/TIM-VX #131: Is SpatialTransformer on rv1126 supported？

sky-fun, 2021-07-01:
I try to run [SpatialTransformer(mxnet)](https://mxnet.apache.org/versions/1.6/api/r/docs/api/mx.symbol.SpatialTransformer.html) operator on rv1126 using  Tengine, but the Status of "SpatialTransformer" operator in tim-vx is "InternalOnly", and no TIM-VX API implementation.So i can't find a way to add SpatialTransformer npu support to Tengine, i wondered that：
1. Is [SpatialTransformer(mxnet)](https://mxnet.apache.org/versions/1.6/api/r/docs/api/mx.symbol.SpatialTransformer.html) operator on rv1126 npu supported?
2. If it is supported, how can i add it to Tengine?

thezha, 2021-07-02:
SpatialTransformer is supported internally, we will add a TIM-VX API for it, then Tengine can integrate the new API.

thezha, 2021-07-05:
https://github.com/VeriSilicon/TIM-VX/pull/132

sky-fun, 2021-07-05:
@thezha i add the code to Tengine, and run example on  x86_64 simulator platform, got error like 
`Kernel "com.vivantecorp.extension.vxcTransform_setupThres_F16toF16" does not exist     
Kernel "com.vivantecorp.extension.vxcTransform_Gemm_F16toF16" does not exist     
Kernel "com.vivantecorp.extension.vxcTransform_InterP_F16toF16" does not exist`

Do i need to change something about the dependence ？

thezha, 2021-07-05:
Did you specify the runtime compiler header file via "export VIVANTE_SDK_DIR"?

sky-fun, 2021-07-05:
@thezha yes， i did add "export VIVANTE_SDK_DIR=pwd/prebuilt-sdk/x86_64_linux" before i run the example.

thezha, 2021-07-05:
@sky-fun These messages are not errors, they are information. Does your case run OK? Please supply more logs if possible. 
`Kernel "com.vivantecorp.extension.vxcTransform_setupThres_F16toF16" does not exist Kernel "com.vivantecorp.extension.vxcTransform_Gemm_F16toF16" does not exist Kernel "com.vivantecorp.extension.vxcTransform_InterP_F16toF16" does not exist`

sky-fun, 2021-07-05:
@thezha i can run my demo sucess with an unexpect result, so i use "export VIV_VX_DEBUG_LEVEL=4" to see the those informations, then i thought it was error bucause the phrase： "does not exist", and it may not run SpatialTransformer sucess on tim-vx. downblew is all my logs:

```
#productname=VSI SIMULATOR, pid=0xc5
prev_ptrs = 0x55b69df8c080
prev_ptrs = 0x55b69df8cac0
Kernel "com.vivantecorp.extension.vxcTransform_setupThres_F16toF16" does not existKernel "com.vivantecorp.extension.vxcTransform_Gemm_F16toF16" does not existKernel "com.vivantecorp.extension.vxcTransform_InterP_F16toF16" does not exist---------------------------Begin VerifyTiling -------------------------
AXI-SRAM = 0 Bytes VIP-SRAM = 260096 Bytes SWTILING_PHASE_FEATURES[1, 1, 1]
  0 TP [( 112  128    3 1,    43008, 0x0x55b69ded6710(0x0x55b69ded6710, 0x(nil)) ->  112  128    3 1,    86016, 0x0x55b69dfe5d10(0x0x55b69dfe5d10, 0x(nil))) k(0 0    0,        0) pad(0 0) pool(0 0, 1 1)] C[ 11]
  1 NN [( 112  128    3 1,    43008, 0x0x55b69ded6710(0x0x55b69ded6710, 0x(nil)) ->   56   64    3 1,    10752, 0x0x55b69dfce460(0x0x55b69dfce460, 0x(nil))) k(2 2    1,      256) pad(0 0) pool(2 2, 2 2)] C[  2]
  2 NN [(  56   64    3 1,    10752, 0x0x55b69dfce460(0x0x55b69dfce460, 0x(nil)) ->   27   31   24 1,    20088, 0x0x55b69dfd3de0(0x0x55b69dfd3de0, 0x(nil))) k(3 3    3,     1024) pad(0 0) pool(2 2, 2 2)] P[  1] C[  3]
  3 NN [(  27   31   24 1,    20088, 0x0x55b69dfd3de0(0x0x55b69dfd3de0, 0x(nil)) ->   25   29   24 1,    17400, 0x0x55b69dfd8ae0(0x0x55b69dfd8ae0, 0x(nil))) k(3 3   24,     5888) pad(0 0) pool(0 0, 1 1)] P[  2] C[  4]
  4 TP [(  25   29   24 1,    17400, 0x0x55b69dfd8ae0(0x0x55b69dfd8ae0, 0x(nil)) ->   12   14   24 1,     4032, 0x0x55b69dfd9c50(0x0x55b69dfd9c50, 0x(nil))) k(0 0    0,        0) pad(0 0) pool(2 2, 2 2)] P[  3] C[  5]
  5 NN [(  12   14   24 1,     4032, 0x0x55b69dfd9c50(0x0x55b69dfd9c50, 0x(nil)) ->    5    6   48 1,     1440, 0x0x55b69dfcd6e0(0x0x55b69dfcd6e0, 0x(nil))) k(3 3   24,    11520) pad(0 0) pool(2 2, 2 2)] P[  4] C[  6]
  6 TP [(1440    1    1 1,     1440, 0x0x55b69dfcd6e0(0x0x55b69dfcd6e0, 0x(nil)) ->   32    1    1 1,       32, 0x0x55b69dfe2ed0(0x0x55b69dfe2ed0, 0x(nil))) k(0 0    0,        0) pad(0 0) pool(0 0, 1 1)] P[  5] C[  7]
  7 TP [(  32    1    1 1,       32, 0x0x55b69dfe2ed0(0x0x55b69dfe2ed0, 0x(nil)) ->    6    1    1 1,        6, 0x0x55b69dfe4040(0x0x55b69dfe4040, 0x(nil))) k(0 0    0,        0) pad(0 0) pool(0 0, 1 1)] P[  6] C[  8]
  8 TP [(   6    1    1 1,        6, 0x0x55b69dfe4040(0x0x55b69dfe4040, 0x(nil)) ->    6    1    1 1,       12, 0x0x55b69dfe6240(0x0x55b69dfe6240, 0x(nil))) k(0 0    0,        0) pad(0 0) pool(0 0, 1 1)] P[  7] C[  9]
  9 SH [(   6    1    1 1,       12, 0x0x55b69dff0750(0x0x55b69dff0750, 0x(nil)) ->    6    1    1 1,       12, 0x0x55b69dfe7f90(0x0x55b69dfe7f90, 0x(nil))) k(0 0    0,        0) pad(0 0) pool(0 0, 1 1)] P[  8] C[ 10]
 10 SH [(   6    1    1 1,       12, 0x0x55b69dfe7f90(0x0x55b69dfe7f90, 0x(nil)) ->  224  128    1 1,    57344, 0x0x55b69dfe8da0(0x0x55b69dfe8da0, 0x(nil))) k(0 0    0,        0) pad(0 0) pool(0 0, 1 1)] P[  9] C[ 11]
 11 SH [( 112  128    3 1,    86016, 0x0x55b69dfe5d10(0x0x55b69dfe5d10, 0x(nil)) ->  112  128    3 1,    86016, 0x0x55b69dfe6900(0x0x55b69dfe6900, 0x(nil))) k(0 0    0,        0) pad(0 0) pool(0 0, 1 1)] P[  0, 10] C[ 12]
 12 TP [( 112  128    3 1,    86016, 0x0x55b69dfe6900(0x0x55b69dfe6900, 0x(nil)) ->  112  128    3 1,    43008, 0x0x55b69dfe5790(0x0x55b69dfe5790, 0x(nil))) k(0 0    0,        0) pad(0 0) pool(0 0, 1 1)] P[ 11] C[ 13]
 13 NN [( 112  128    3 1,    43008, 0x0x55b69dfe5790(0x0x55b69dfe5790, 0x(nil)) ->  112  128   24 1,   344064, 0x0x55b69e06fe60(0x0x55b69e06fe60, 0x(nil))) k(3 3    3,     1024) pad(1 1) pool(0 0, 1 1)] P[ 12] C[ 14, 17]
 14 TP [( 112  128   24 1,   344064, 0x0x55b69e06fe60(0x0x55b69e06fe60, 0x(nil)) ->   57   65   96 1,   355680, 0x0x55b6a0375d70(0x0x55b6a0375d70, 0x(nil))) k(0 0    0,        0) pad(1 1) pool(0 0, 1 1)] P[ 13] C[ 15]
 15 NN [(  57   65   96 1,   355680, 0x0x55b6a0375d70(0x0x55b6a0375d70, 0x(nil)) ->   56   64   24 1,    86016, 0x0x55b69e02bc30(0x0x55b69e02bc30, 0x(nil))) k(2 2   96,    10496) pad(0 0) pool(0 0, 1 1)] P[ 14] C[ 16]
 16 NN [(  56   64   24 1,    86016, 0x0x55b69e02bc30(0x0x55b69e02bc30, 0x(nil)) ->   56   64   24 1,   172032, 0x0x55b69e08efc0(0x0x55b69e08efc0, 0x(nil))) k(3 3   24,     5888) pad(1 1) pool(0 0, 1 1)] P[ 15] C[ 18]
 17 NN [( 112  128   24 1,   344064, 0x0x55b69e06fe60(0x0x55b69e06fe60, 0x(nil)) ->   56   64   24 1,   172032, 0x0x55b6a1ee6fc0(0x0x55b69e08efc0, 0x0x15000)) k(1 1   24,      768) pad(0 0) pool(2 2, 2 2)] P[ 13] C[ 18]
 18 NN [(  56   64   48 1,   172032, 0x0x55b69e08efc0(0x0x55b69e08efc0, 0x(nil)) ->   56   64   24 1,   172032, 0x0x55b6a1eec5a0(0x0x55b69e0945a0, 0x0x15000)) k(1 1   48,     1536) pad(0 0) pool(0 0, 1 1)] P[ 16, 17] C[ 19, 22]
 19 TP [(  56   64   24 1,   172032, 0x0x55b6a1eec5a0(0x0x55b69e0945a0, 0x0x15000) ->   56   64   24 1,    86016, 0x0x55b69e058fe0(0x0x55b69e058fe0, 0x(nil))) k(0 0    0,        0) pad(0 0) pool(0 0, 1 1)] P[ 18] C[ 20]
 20 NN [(  56   64   24 1,    86016, 0x0x55b69e058fe0(0x0x55b69e058fe0, 0x(nil)) ->   56   64   24 1,    86016, 0x0x55b69e05c740(0x0x55b69e05c740, 0x(nil))) k(3 3   24,     5888) pad(1 1) pool(0 0, 1 1)] P[ 19] C[ 21]
[... cut ...]
vxo_insertHandle(out): old physical address: 00296300, mapIndex: 2, node_id: 15
success prerun_graph
##################VXEngineRun
VXEngineRun 0 4
prev_ptrs = 0x55b69df8cac0

Warning: swapHandel, CMD changed

 NN/TP: pre_physical:0x00296940, new_physical:0x00296940 

 NN/TP: pre_physical:0x00296978, new_physical:0x00296978 

Warning: swapHandel, CMD changed

 NN/TP: pre_physical:0x00296940, new_physical:0x00296940 

Warning: swapHandel, CMD changed

 SH:  pre_physical:0x00296300, new_physical:0x00296300, mapIndex: 2, node_id: 15
prev_ptrs = 0x55b69df8c080

sky-fun, 2021-07-05:
yes, but in the origin network target shape of SpecialTransformer should be (3,128, 112)

nightingalei, 2021-07-05:
@sky-fun What's the datatype and parameters to the SpacialTransformer?

nightingalei, 2021-07-05:
@sky-fun Could please also provide the log with the env "export VSI_NN_LOG_LEVEL=5"? Thanks.

sky-fun, 2021-07-05:
datatype is uint8 and the parameters is (output_h, output_w, false, false, false, false, false, false, 0.0, 0.0, 0.0, 0.0,  0.0, 0.0)

sky-fun, 2021-07-05:
@nightingalei 
```
D [vsi_nn_SetupGraph:617]Sort graph nodes.
no Unprocessed node 
D [setup_node:441]Setup node id[0] uid[1] op[POOL]
D [validate_op_io_types:165]Validate [POOL]
D [print_tensor:146]in(0) : id[   1] vtl[0] const[0] shape[ 112, 128, 3, 1   ] fmt[u8 ] qnt[ASM zp=127, scale=0.007782]
D [print_tensor:146]out(0): id[   4] vtl[1] const[0] shape[ 56, 64, 3, 1     ] fmt[u8 ] qnt[ASM zp=127, scale=0.007805]
D [setup_node:441]Setup node id[1] uid[2] op[CONV2D]
D [validate_op_io_types:165]Validate [CONV2D]
D [print_tensor:146]in(0) : id[   4] vtl[1] const[0] shape[ 56, 64, 3, 1     ] fmt[u8 ] qnt[ASM zp=127, scale=0.007805]
D [print_tensor:146]in(1) : id[   3] vtl[0] const[1] shape[ 3, 3, 3, 24      ] fmt[u8 ] qnt[ASM zp=144, scale=0.008708]
D [print_tensor:146]in(2) : id[   2] vtl[0] const[1] shape[ 24               ] fmt[i32] qnt[ASM zp=  0, scale=0.000068]
D [print_tensor:146]out(0): id[ 105] vtl[1] const[0] shape[ 54, 62, 24, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.014651]
D [setup_node:441]Setup node id[2] uid[3] op[RELU]
D [validate_op_io_types:165]Validate [RELU]
D [print_tensor:146]in(0) : id[ 105] vtl[1] const[0] shape[ 54, 62, 24, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.014651]
D [print_tensor:146]out(0): id[   5] vtl[1] const[0] shape[ 54, 62, 24, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.014651]
D [setup_node:441]Setup node id[3] uid[4] op[POOL]
D [validate_op_io_types:165]Validate [POOL]
D [print_tensor:146]in(0) : id[   5] vtl[1] const[0] shape[ 54, 62, 24, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.014651]
D [print_tensor:146]out(0): id[   8] vtl[1] const[0] shape[ 27, 31, 24, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.014651]
D [setup_node:441]Setup node id[4] uid[5] op[CONV2D]
D [validate_op_io_types:165]Validate [CONV2D]
D [print_tensor:146]in(0) : id[   8] vtl[1] const[0] shape[ 27, 31, 24, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.014651]
D [print_tensor:146]in(1) : id[   7] vtl[0] const[1] shape[ 3, 3, 24, 24     ] fmt[u8 ] qnt[ASM zp=123, scale=0.003316]
D [print_tensor:146]in(2) : id[   6] vtl[0] const[1] shape[ 24               ] fmt[i32] qnt[ASM zp=  0, scale=0.000049]
D [print_tensor:146]out(0): id[ 106] vtl[1] const[0] shape[ 25, 29, 24, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.006747]
D [setup_node:441]Setup node id[5] uid[6] op[RELU]
D [validate_op_io_types:165]Validate [RELU]
D [print_tensor:146]in(0) : id[ 106] vtl[1] const[0] shape[ 25, 29, 24, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.006747]
D [print_tensor:146]out(0): id[   9] vtl[1] const[0] shape[ 25, 29, 24, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.006747]
D [setup_node:441]Setup node id[6] uid[7] op[POOL]
D [validate_op_io_types:165]Validate [POOL]
D [print_tensor:146]in(0) : id[   9] vtl[1] const[0] shape[ 25, 29, 24, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.006747]
D [print_tensor:146]out(0): id[  12] vtl[1] const[0] shape[ 12, 14, 24, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.006747]
D [setup_node:441]Setup node id[7] uid[8] op[CONV2D]
D [validate_op_io_types:165]Validate [CONV2D]
D [print_tensor:146]in(0) : id[  12] vtl[1] const[0] shape[ 12, 14, 24, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.006747]
D [print_tensor:146]in(1) : id[  11] vtl[0] const[1] shape[ 3, 3, 24, 48     ] fmt[u8 ] qnt[ASM zp=132, scale=0.003830]
D [print_tensor:146]in(2) : id[  10] vtl[0] const[1] shape[ 48               ] fmt[i32] qnt[ASM zp=  0, scale=0.000026]
D [print_tensor:146]out(0): id[ 107] vtl[1] const[0] shape[ 10, 12, 48, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.004216]
D [setup_node:441]Setup node id[8] uid[9] op[RELU]
D [validate_op_io_types:165]Validate [RELU]
D [print_tensor:146]in(0) : id[ 107] vtl[1] const[0] shape[ 10, 12, 48, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.004216]
D [print_tensor:146]out(0): id[  13] vtl[1] const[0] shape[ 10, 12, 48, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.004216]
D [setup_node:441]Setup node id[9] uid[10] op[POOL]
D [validate_op_io_types:165]Validate [POOL]
D [print_tensor:146]in(0) : id[  13] vtl[1] const[0] shape[ 10, 12, 48, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.004216]
D [print_tensor:146]out(0): id[  14] vtl[1] const[0] shape[ 5, 6, 48, 1      ] fmt[u8 ] qnt[ASM zp=  0, scale=0.004216]
D [setup_node:441]Setup node id[10] uid[11] op[RESHAPE]
D [print_tensor:146]in(0) : id[  14] vtl[1] const[0] shape[ 5, 6, 48, 1      ] fmt[u8 ] qnt[ASM zp=  0, scale=0.004216]
D [print_tensor:146]out(0): id[  15] vtl[1] const[0] shape[ 1, 1, 1440, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.004216]
D [setup_node:441]Setup node id[11] uid[12] op[FCL2]
D [validate_op_io_types:165]Validate [FCL_RELU]
D [print_tensor:146]in(0) : id[  15] vtl[1] const[0] shape[ 1, 1, 1440, 1    ] fmt[u8 ] qnt[ASM zp=  0, scale=0.004216]
D [print_tensor:146]in(1) : id[  17] vtl[0] const[1] shape[ 1440, 32         ] fmt[u8 ] qnt[ASM zp=167, scale=0.005804]
D [print_tensor:146]in(2) : id[  16] vtl[0] const[1] shape[ 32               ] fmt[i32] qnt[ASM zp=  0, scale=0.000024]
D [print_tensor:146]out(0): id[  18] vtl[1] const[0] shape[ 32, 1            ] fmt[u8 ] qnt[ASM zp=  0, scale=0.005455]
D [setup_node:441]Setup node id[12] uid[13] op[RELU]
D [validate_op_io_types:165]Validate [RELU]
D [print_tensor:146]in(0) : id[  18] vtl[1] const[0] shape[ 32, 1            ] fmt[u8 ] qnt[ASM zp=  0, scale=0.005455]
D [print_tensor:146]out(0): id[  19] vtl[1] const[0] shape[ 32, 1            ] fmt[u8 ] qnt[ASM zp=  0, scale=0.005455]
D [setup_node:441]Setup node id[13] uid[14] op[FCL2]
D [validate_op_io_types:165]Validate [FCL_RELU]
[... cut ...]
D [compute_node:350]Instance node[56] "CONV2D" ...
D [compute_node:350]Instance node[57] "RELU" ...
D [compute_node:350]Instance node[58] "CONV2D" ...
D [compute_node:350]Instance node[59] "ADD" ...
D [vsi_nn_kernel_selector:970]Instance OPENVX node with kernel "add" 
D [compute_node:350]Instance node[60] "RELU" ...
D [compute_node:350]Instance node[61] "POOL" ...
D [compute_node:350]Instance node[62] "RESHAPE" ...
D [compute_node:350]Instance node[63] "L2_NORMALIZE" ...
---------------------------Begin VerifyTiling -------------------------

sky-fun, 2021-07-05:
@nightingalei ok, now i can see the "Register client kernel com.vivantecorp.extension.vxcTransform_setupThres_F16toF16 successfully", thanks;
the unexpected result may comes from somewhere else;

nightingalei, 2021-07-05:
@sky-fun The network seems to be success. Could you please make a unit test for SpatialTransformer from Tengine and make sure it's correct?  The SpatialTransformer may have different implementations from different NN engine, TIM-VX uses this [link](https://github.com/christopher5106/last_caffe_with_stn/blob/master/src/caffe/layers/st_layer.cpp)(https://github.com/christopher5106/last_caffe_with_stn/blob/master/src/caffe/layers/st_layer.cpp).

sky-fun, 2021-07-05:
@nightingalei  sure， i will work on this to find the reason why i got wrong results。

sky-fun, 2021-07-11:
@nightingalei , after check out the mxnet version [SpatialTransformer ](https://github.com/apache/incubator-mxnet/blob/master/src/operator/spatial_transformer.cc),  i find that there is some difference in the SpatialTransformer  implementation:
the order of GridData in vsi_nn_op_spatial_transformer.c is (y,x,1), the order of gemm result is (y,x);
while in our mxnet model, GridData is (x,y,1),  gemm result is (x,y);
so after i manually change the theta data from (t0,t1,t2,t3,t4,t5) to (t4,t3,t5,t1,t0,t2), the SpatialTransformer  result is close to the origin model.

thezha, 2021-07-12:
@sky-fun Could you clarify, did you modify the implementation in tim-vx or the implementation in mxnet?

sky-fun, 2021-07-12:
@thezha In fact, i add `tim::vx::ops::Split` and `tim::vx::ops::Concat` to change the order of second input data(the theta data) before i use the `tim::vx::ops::SpatialTransformer`. So i modify the implementation where tengine-lite add  tim-vx node.

sky-fun, 2021-07-15:
@thezha i found that there is still some difference between the mxnet version and the caffe version：
The grid data in mxnet is coordinate divide (W-1)and(H-1) (https://github.com/apache/incubator-mxnet/blob/8fd17cef2ee854239c6e66f2dd3b9467aa222f79/src/operator/spatial_transformer-inl.h#L99), while caffe version is coordinate divide W and H (https://github.com/christopher5106/last_caffe_with_stn/blob/037482abdfa801f9c606ae878ed8cd3dd2bec675/src/caffe/layers/st_layer.cpp#L118); 
And of course they are different in interp step when they transform gemm result to origin coordinate;

So is there anything i can do to fix this deviation?

thezha, 2021-07-16:
In the next version, we will add align_corners attribute to spatial transform operation. It will be updated end of this month. Thanks!

rniranjan93, 2021-07-26:
I have a .tflite file , I want to build a graph with Delegate class and want to return the graph using the following method
```std::shared_ptr<tim::vx::Graph>& GetGraph() { return graph_; }```
could you provide an example , i.e, taking .tflite as input -> returning a builded tim::vx::Graph

sunshinemyson, 2021-07-29:
> 
> 
> I have a .tflite file , I want to build a graph with Delegate class and want to return the graph using the following method
> `std::shared_ptr<tim::vx::Graph>& GetGraph() { return graph_; }`
> could you provide an example , i.e, taking .tflite as input -> returning a builded tim::vx::Graph

What's the purpose for this usage? It's quite strange to me if you want use tflite interpreter to extract model but don't want to inference it with tflite interpreter.

thezha, 2021-08-03:
@sky-fun align_corners option is added to SpatialTransformer 6a949bb315b7d711cf6f73056f1022ae23d32579

sky-fun, 2021-08-04:
@thezha thanks

closed 2021-08-10 by sky-fun