VeriSilicon/TIM-VX #143: Usage of  vxFlushHandle and vxMapTensorPatch

janoslim, 2021-07-24:
Hello,

I tried to make tensor object with vxCreateTensorFromHandle2() to use host level memory block,
And I recently notice that not calling vxFlushHandle(), the data change is not correctly applied.
And I also see that vxMapTensorPatch() need to be called to access tensor data...
https://www.khronos.org/registry/OpenVX/specs/1.3/html/OpenVX_Specification_1_3.html#_vxmaptensorpatch
Ultimately, I want to create a handle at the host level and apply it every time I change it, and I wonder if it's okay to use only the vxFlushHandle.
Thank you!

sunshinemyson, 2021-07-28:
Yes, It's ok only use vxFlushHandle() after changed the memory.

janoslim, 2021-07-28:
Thank you for your reply.
I wonder what vxFlushHandle() actually do, for example the memory structure between openVX and npu, so that the reason for calling vxFlushHandle().

sunshinemyson, 2021-07-29:
vxFlushHandle() actually just flush the CPU cache back to the memory so that NPU can get right data - because NPU won't reuse cpu cache.

closed 2021-08-02 by sunshinemyson