VeriSilicon/TIM-VX #164: TVM RPC test failed with message: "PLS isn't existed"

Goose-Bomb, 2021-09-01:
I don't know if it's the right place to ask questions about your TVM fork, but I cannot raise issues in that repo.

I followed the guide from [README.VSI.md](https://github.com/VeriSilicon/tvm/blob/vsi_npu/README.VSI.md) to build TVM (on host, using x86_64_linux simulation drivers provided here) and TVM runtime (on target, using vendor-provided VIP NPU drivers), and ran the tests in [test_vsi_npu](https://github.com/VeriSilicon/tvm/tree/vsi_npu/tests/python/contrib/test_vsi_npu), but I got these results: 

logs from TVM C++ RPC tool:
```log
VsiNpuModule::LoadFromBinary
LoadFromBinary: nbg size = 593344
LoadFromBinary: input size = 1
LoadFromBinary: output size = 1
VsiNpuModule : DeSerializeTensorSpec
VsiNpuModule : DeSerializeTensorSpec2
VsiNpuModule : DeSerializeTensorSpec
VsiNpuModule : DeSerializeTensorSpec2
[22:31:35] /home/nullko/Documents/tvm-vsi_npu/apps/cpp_rpc/rpc_env.cc:130: Load module from /home/ubuntu/workspace/rpc/model.so ...
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: tvmgen_default_vsi_npu_0
[     1] PLS isn't existed
E [compute_node:379]Create node[0] NBG fail
Process Graph: 0 ms or 0 us
```
It seemed that TVM is able to compile the NBG on the host, but the target runtime cannot execute it.
I wonder what caused the **"PLS isn't existed"** issue, is it because I didn't set some environment variables on the target platform?

Or maybe your tvm fork is still under development and is not ready to be used yet?

Goose-Bomb, 2021-09-01:
BTW, the target is capable of running NN models converted by your Acuity toolKit, so I guess there's no problem with the hardware.

sunshinemyson, 2021-09-01:
@Goose-Bomb ，

Did you setup VSIMULATRO_TARGET on your target board? I forget to put it into readme.

Goose-Bomb, 2021-09-02:
> @Goose-Bomb ，
> 
> Did you setup VSIMULATRO_TARGET while compile the model with tvm? I forget to put it into readme.

Yes, I did setup the `export VSIMULATOR_TARGET=PID_0xA4` before running the test python script.

xuke537, 2021-09-02:
> > @Goose-Bomb ，
> > Did you setup VSIMULATRO_TARGET while compile the model with tvm? I forget to put it into readme.
> 
> Yes, I did setup the `export VSIMULATOR_TARGET=PID_0xA4` before running the test python script.

Your target should be "VIP8000OI_PID0XA4", please try  `export VSIMULATOR_TARGET=VIP8000OI_PID0XA4`

Goose-Bomb, 2021-09-02:
> Your target should be "VIP8000OI_PID0XA4", please try `export VSIMULATOR_TARGET=VIP8000OI_PID0XA4`

Still has the "PLS isn't existed" problem. It seems that this message is issued from the NPU driver, is my driver version too low?

Goose-Bomb, 2021-09-02:
Here is the dmsg log produced while running the program.
```log
[Dec 8 03:14] [galcore]: gckOS_ReadRegisterEx(1916) GPU[0] external clock off
[  +0.000016] [galcore]: BUG ON @ gckOS_ReadRegisterEx(1917)
[  +0.012540] CPU: 1 PID: 7027 Comm: tvm_rpc Tainted: G           O      4.19.65 #4
[  +0.007454] Hardware name: Orbita OA1801 ARM Cortex A9 Platform
[  +0.005930] [<c0111620>] (unwind_backtrace) from [<c010c7c8>] (show_stack+0x10/0x14)
[  +0.007723] [<c010c7c8>] (show_stack) from [<c0838bec>] (dump_stack+0x88/0x9c)
[  +0.007295] [<c0838bec>] (dump_stack) from [<bf004680>] (gckOS_ReadRegisterEx+0x154/0x168 [galcore])
[  +0.009195] [<bf004680>] (gckOS_ReadRegisterEx [galcore]) from [<bf02d378>] (gckHARDWARE_SetGpuProfiler+0x44/0xc4 [galcore])
[  +0.011264] [<bf02d378>] (gckHARDWARE_SetGpuProfiler [galcore]) from [<bf0141e8>] (gckDEVICE_Profiler_Dispatch+0x214/0x220 [galcore])
[  +0.012037] [<bf0141e8>] (gckDEVICE_Profiler_Dispatch [galcore]) from [<bf00cf90>] (drv_ioctl+0x130/0x334 [galcore])
[  +0.010535] [<bf00cf90>] (drv_ioctl [galcore]) from [<c025d02c>] (do_vfs_ioctl+0xac/0x8c8)
[  +0.008241] [<c025d02c>] (do_vfs_ioctl) from [<c025d87c>] (ksys_ioctl+0x34/0x58)
[  +0.007372] [<c025d87c>] (ksys_ioctl) from [<c0101000>] (ret_fast_syscall+0x0/0x54)
[  +0.007623] Exception stack(0xc9eabfa8 to 0xc9eabff0)
[  +0.005034] bfa0:                   00002710 b021f810 00000005 00007531 b021f7c8 00000000
[  +0.008145] bfc0: 00002710 b021f810 b021f7c8 00000036 00007531 00000188 b021fc84 b021fc68
[  +0.008142] bfe0: b5c1b220 b021f7b4 b5c038c7 b6b30ba8
[  +0.005111] [galcore]: _WriteRegisterEx(1983) GPU[0] external clock off
[  +0.000033] [galcore]: BUG ON @ _WriteRegisterEx(1986)
[  +0.011758] CPU: 1 PID: 7027 Comm: tvm_rpc Tainted: G           O      4.19.65 #4
[  +0.007457] Hardware name: Orbita OA1801 ARM Cortex A9 Platform
[  +0.005922] [<c0111620>] (unwind_backtrace) from [<c010c7c8>] (show_stack+0x10/0x14)
[  +0.007720] [<c010c7c8>] (show_stack) from [<c0838bec>] (dump_stack+0x88/0x9c)
[  +0.007275] [<c0838bec>] (dump_stack) from [<bf003608>] (_WriteRegisterEx.constprop.3+0x120/0x170 [galcore])
[  +0.009886] [<bf003608>] (_WriteRegisterEx.constprop.3 [galcore]) from [<bf02d39c>] (gckHARDWARE_SetGpuProfiler+0x68/0xc4 [galcore])
[  +0.011954] [<bf02d39c>] (gckHARDWARE_SetGpuProfiler [galcore]) from [<bf0141e8>] (gckDEVICE_Profiler_Dispatch+0x214/0x220 [galcore])
[  +0.012036] [<bf0141e8>] (gckDEVICE_Profiler_Dispatch [galcore]) from [<bf00cf90>] (drv_ioctl+0x130/0x334 [galcore])
[  +0.010534] [<bf00cf90>] (drv_ioctl [galcore]) from [<c025d02c>] (do_vfs_ioctl+0xac/0x8c8)
[  +0.008239] [<c025d02c>] (do_vfs_ioctl) from [<c025d87c>] (ksys_ioctl+0x34/0x58)
[  +0.007371] [<c025d87c>] (ksys_ioctl) from [<c0101000>] (ret_fast_syscall+0x0/0x54)
[  +0.007622] Exception stack(0xc9eabfa8 to 0xc9eabff0)
[  +0.005033] bfa0:                   00002710 b021f810 00000005 00007531 b021f7c8 00000000
[  +0.008143] bfc0: 00002710 b021f810 b021f7c8 00000036 00007531 00000188 b021fc84 b021fc68
[  +0.008141] bfe0: b5c1b220 b021f7b4 b5c038c7 b6b30ba8

```

sunshinemyson, 2021-09-02:
You can export VIV_VX_DEBUG_LEVEL=1 on your board, could you share the driver's log?

Goose-Bomb, 2021-09-03:
```log
VsiNpuModule::LoadFromBinary
LoadFromBinary: nbg size = 2880
LoadFromBinary: input size = 1
LoadFromBinary: output size = 1
VsiNpuModule : DeSerializeTensorSpec
VsiNpuModule : DeSerializeTensorSpec2
VsiNpuModule : DeSerializeTensorSpec
VsiNpuModule : DeSerializeTensorSpec2
[19:50:58] /home/nullko/Documents/tvm-vsi_npu/apps/cpp_rpc/rpc_env.cc:130: Load module from /home/ubuntu/workspace/rpc/model.so ...
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: tvmgen_default_vsi_npu_0
[     1] PLS isn't existed
#productname=VIP8000-OI, pid=0xa4
prev_ptrs = 0xaf87cd40
prev_ptrs = 0xaf87d4c0
binary graph format version, 0x1000e
readBinHeader[1492]: binary target: 0xc5, actually target: 0xa4fail to load binary from pointer to create graph
NBG error, please provide genereating NBG logs first
fail to import kernel from VPMN, error code: -10
E [compute_node:354]Create node[0] NBG fail
prev_ptrs = 0xaf87cd40
vxProcessGraph[16152]: Process Graph fail!
vxProcessGraph[16152]: Process Graph fail!
Process Graph: 0 ms or 0 us
VsiNpuModule::GetFunction: size: 2
prev_ptrs = 0xaf87d4c0
prev_ptrs = 0xaf87cd40
prev_ptrs = 0xaf87d4c0
Exit VX Thread: 0xb29d38d0
[19:50:58] /home/nullko/Documents/tvm-vsi_npu/apps/cpp_rpc/rpc_server.cc:322: Finish serving 113.54.149.216:60328
[19:50:58] /home/nullko/Documents/tvm-vsi_npu/apps/cpp_rpc/rpc_server.cc:212: Child pid=3113 exited, Process status =0
[19:50:58] /home/nullko/Documents/tvm-vsi_npu/apps/cpp_rpc/rpc_server.cc:229: Socket Connection Closed
```

sunshinemyson, 2021-09-03:
Yes, please set your target to VIP9000NANOS_PID0XC5 on your host.

Goose-Bomb, 2021-09-03:
> Yes, please set your target to VIP9000NANOS_PID0XC5 on your host.

But the target device is actually VIP8000, what’s the point to change the variable to VIP9000?

Goose-Bomb, 2021-09-03:
It seems that the host simulator driver somehow didn’t recognize that environment variable and didn’t compile the NBG into corresponding target NPU format?

sunshinemyson, 2021-09-03:
Oh, Sorry, My bad. Let me double check it. Would you mind send me a email? xiang.zhang@verisilicon.com

Goose-Bomb, 2021-09-06:
Just tested the TVM RPC on Rockchip RV1126 platform, set `export VSIMULATOR_TARGET=VIP9000NANOSI_PID0XB5` on my host, same problem. The simulator driver still compiled the NBG for `0xc5` target.

logs from host (when running test_operations.py):
```log
#productname=VSI SIMULATOR, pid=0xc5
Testing QNN.ADD                                           1. press any key and continue...
#[version = "0.0.5"]
def @main(%x: Tensor[(1, 96, 96, 64), uint8], %y: Tensor[(64), uint8]) {
  qnn.add(%x, %y, 0.020283f, 112, 0.000316f, 119, 0.020144f, 112)
}

vsi_npu.py --> qnn.add
This is important----> name_node.value() == tvmgen_default_vsi_npu_0
GraphMakerImpl::Create
TensorMakerImpl::InferCall: qnn.add
prev_ptrs = 0x556267de5fc0
prev_ptrs = 0x556268028f40
prev_ptrs = 0x5562680b9000
prev_ptrs = 0x556268149080

---------------------------Begin VerifyTiling -------------------------
AXI-SRAM = 0 Bytes VIP-SRAM = 260096 Bytes SWTILING_PHASE_FEATURES[1, 1, 1]
  0 SH [(49152   12    1 1,   589824, 0x0x556267c45b50(0x0x556267c45b50, 0x(nil)) -> 49152   12    1 1,   589824, 0x0x556267c421a0(0x0x556267c421a0, 0x(nil))) k(0 0    0,        0) pad(0 0) pool(0 0, 1 1)]

 id IN [ x  y  w   h ]   OUT  [ x  y  w  h ] (tx, ty, kpc) (ic, kc, kc/ks, ks/eks, kernel_type)
   0 SH DD 0x(nil) [   0    0        0        0] -> DD 0x(nil) [   0    0        0        0] (  0,   0,   0) (       0,        0, 0.000000%, 0.000000%, NONE)

PreLoadWeightBiases = 0  -nan%
---------------------------End VerifyTiling -------------------------
KernelStreamSize: 0x0, statesSize: 0x540, shShareMemSize: 0x0, shIntrSize: 0x200, shParaSize: 0x1b0000, swParaSize: 0x0, lcdTensorSize: 0x0, shaderStatesSize: 0x4c0, tensorStatic: 0x0
NBG: operationSize: 0x78, nnSize: 0x80, tpSize: 0x80, shSize: 0x8, swSize: 0x0, layerParamSize: 0x0, lcdtSize: 0x48, patchSize: 0x310, icdtSize: 0x0 hwInitOpSize: 0x0, lcdSize 0x1b0840
NBG: entranceSize: 0x1f8, nbIOSize: 0xe8, layeSize: 0x4c, sectionsSize: 0x4d8, inputoutput size: 0x0, InitCommands size: 0x0
NBG: lcdSize: 0x1b0840, headerSize : 0x804
Calculate NBG size : 1773892 bytes
generate NBG into memory start.
vxoBinaryGraph_SaveBinaryEntrance[16689]: collect input count=0, output count=0
vxoBinaryGraph_SaveBinaryEntrance[16771]: total operation count=1
generate NBG, device count=1, core count per-device: 1, 
 input table address: 0x4465c0 
 output table address: 0x566840 
vxoBinaryGraph_SaveBinaryEntranceExt[15868]: graph input/output=1/1, refine input count=1, output count=1
NBG network name field : dummy_network_name
vxoBinaryGraph_SaveBinaryEntranceExt[16463]: header input count=1, output count=1
generate NGB, save initialize commands
generate NBG, map VIP-SRAM start address=0xff000000
generate NBG, map AXI-SRAM size = 0
generate NBG, map AXI-SRAM size=0x0, start address=0x0, end address=0x0
vxo_insertHandle(in): old physical address: 004465c0, mapIndex: 0, node_id: 0
vxo_insertHandle(out): old physical address: 00566840, mapIndex: 3, node_id: 0

Warning: swapHandel, CMD changed

Warning: swapHandel, CMD changed
vxoBinaryGraph_ReSaveInputAndPatchTable[13711]: re-save operation count=3
Generate NBG in memory Actual NBG size : 593344 bytes
generate NBG into memory successfully.
prev_ptrs = 0x5562680b9000
prev_ptrs = 0x556268149080
prev_ptrs = 0x556267de5fc0
prev_ptrs = 0x556268028f40
VsiNpuModule::GetFunction: get_symbol
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: get_const_vars
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: get_const_vars
VsiNpuModule::GetFunction: return early
VsiNpuModule::SaveToBinary
SaveToBinary: nbg size = 593344
SaveToBinary: input size = 1
SaveToBinary: output size = 1
VsiNpuModule : SerializeTensorSpec
VsiNpuModule : SerializeTensorSpec2
VsiNpuModule : SerializeTensorSpec
VsiNpuModule : SerializeTensorSpec2
VsiNpuModule::SaveToBinary2
/tmp/tmpwv3com_y/model.so
```

logs from target RV1126 platform (when running tvm_rpc)
```log
[13:48:08] /home/nullko/Documents/tvm-vsi_npu/apps/cpp_rpc/rpc_server.cc:300: Connection success 113.54.149.237:37966
VsiNpuModule::LoadFromBinary
LoadFromBinary: nbg size = 593344
LoadFromBinary: input size = 1
LoadFromBinary: output size = 1
VsiNpuModule : DeSerializeTensorSpec
VsiNpuModule : DeSerializeTensorSpec2
VsiNpuModule : DeSerializeTensorSpec
VsiNpuModule : DeSerializeTensorSpec2
[13:48:10] /home/nullko/Documents/tvm-vsi_npu/apps/cpp_rpc/rpc_env.cc:130: Load module from /oem/rpc/model.so ...
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: _lookup_linked_param
VsiNpuModule::GetFunction: return early
VsiNpuModule::GetFunction: tvmgen_default_vsi_npu_0
[     1] PLS isn't existed
prev_ptrs = 0x9aab8040
prev_ptrs = 0x9aa27040
binary graph format version, 0x1000e
readBinHeader[1337]: binary target: 0xc5, actually target: 0xb5fail to load binary from pointer to create graph
NBG error, please provide genereating NBG logs first
fail to import kernel from VPMN, error code: -10
E [compute_node:379]Create node[0] NBG fail
prev_ptrs = 0x9aab8040
vxProcessGraph[15000]: Process Graph fail!
vxProcessGraph[15000]: Process Graph fail!
Process Graph: 0 ms or 0 us
VsiNpuModule::GetFunction: size: 2
prev_ptrs = 0x9aa27040
prev_ptrs = 0x9aab8040
prev_ptrs = 0x9aa27040
Exit VX Thread: 0x9b339890
```

sunshinemyson, 2021-09-06:
Please try with VSIMULATOR_CONFIG, we have mistake here before. Thanks

closed 2021-09-06 by Goose-Bomb

leokuo725, 2021-10-05:
@sunshinemyson 

Sorry, I encounter the same problem on TVM. The following is my enviroment:
**TVM Git Branch** 

> https://github.com/VeriSilicon/tvm/tree/upstream/vsi_npu

**Run TVM python test OP from host**

> TVM_FOLDER/tests/python/contrib/test_vsi_npu/test_operations.py

**Device Spec.**

> Mode: Khadas VIM3 Pro​
> SoC: Amlogic A311D with 5 TOPS Performance NPU

**Environment variable (Host)**

> export VSIMULATOR_CONFIG=VIPNANOQI_PID0X88
> export VIV_VX_DEBUG_LEVEL=1 

**Environment variable (VIM3 Pro)**

> export VIV_VX_DEBUG_LEVEL=1 

**sudo dmesg | grep "Galcore"
[   12.481573] Galcore version 6.4.4.3.310723AAA**

Then, I got error from TVM. 
[from_host.log](https://github.com/VeriSilicon/TIM-VX/files/7282323/from_host.log)
[from_device.log](https://github.com/VeriSilicon/TIM-VX/files/7282324/from_device.log)

Could you help me ? 
Thanks.