VeriSilicon/TIM-VX #167: Graph verification errors

nullptr-leo, 2021-09-03:
Retina face quant model, verification failure:

```
D [compute_node:375]Instance node[108] "RESHAPE" ...
D [compute_node:375]Instance node[109] "CONV2D" ...
D [compute_node:375]Instance node[110] "CONV2D" ...
D [compute_node:375]Instance node[111] "RESHAPE" ...
D [compute_node:375]Instance node[112] "CONV2D" ...
D [compute_node:375]Instance node[114] "RESHAPE" ...
D [compute_node:398]compute node finish with ret 0
E [vsi_nn_SetupGraph:752]after compute_node
E [vsi_nn_SetupGraph:759]now will vsi_nn_TrySetupCompleteSignalNode
E [vsi_nn_SetupGraph:764]now will vsi_nn_setup_binary_graph_inputs_outputs
E [vsi_nn_SetupGraph:777]vsi_nn_SetupGraph return with ret = 0
E [vsi_nn_VerifyGraph:795]vsi_nn_VerifyGraph return with ret = -1
Tengine Fatal: Pre-run subgraph(0) on TIMVX failed.
Tengine: Scheduler(sync) prerun failed.
Pre-run graph failed
```

```
root@localhost:/tmp# ldd ./tm_retinaface_timvx
        linux-vdso.so.1 (0xae88d000)
        libtengine-lite.so => /usr/lib/libtengine-lite.so (0xa6a0d000)
        libdl.so.2 => /lib/libdl.so.2 (0xa69fa000)
        libm.so.6 => /lib/libm.so.6 (0xa697b000)
        libCLC.so => /usr/lib/libCLC.so (0xa67c6000)
        libGAL.so => /usr/lib/libGAL.so (0xa6629000)
        libOpenVX.so.1.2 => /usr/lib/libOpenVX.so.1.2 (0xa641b000)
        libOpenVXU.so => /usr/lib/libOpenVXU.so (0xa6405000)
        libVSC.so => /usr/lib/libVSC.so (0xa55bc000)
        libArchModelSw.so => /usr/lib/libArchModelSw.so (0xa557b000)
        libNNArchPerf.so => /usr/lib/libNNArchPerf.so (0xa553a000)
        libstdc++.so.6 => /lib/libstdc++.so.6 (0xa53d1000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xa53a2000)
        libc.so.6 => /lib/libc.so.6 (0xa5260000)
        libpthread.so.0 => /lib/libpthread.so.0 (0xa5237000)
        libgomp.so.1 => /usr/lib/libgomp.so.1 (0xa51ff000)
        /lib/ld-linux-armhf.so.3 (0xa6ef8000)
        librt.so.1 => /lib/librt.so.1 (0xa51e8000)
        libVSC_Lite.so => /usr/lib/libVSC_Lite.so (0xa51c3000)
```

Any idea?

sunshinemyson, 2021-09-03:
Is this issue only happen when you compiled it as a binary graph? Can you provide more detail about your Application usage?

nullptr-leo, 2021-09-04:
> Is this issue only happen when you compiled it as a binary graph? Can you provide more detail about your Application usage?

I want to deploy retinaface on rockchip rv1126 with NPU support via Tengine. But prerun_graph in Tengine failed.

nullptr-leo, 2021-09-06:
With export VIV_VX_DEBUG_LEVEL=4:

```
Set drv optimization level = 3
Error: Make sure the environment variable VIVANTE_SDK_DIR is set to the same directory when compiling and executing.
Failed to initialize Kernel "vivante.nn.softmaxlayer2.layer" of Node 0xe0bc0 (status = -1)
```

sunshinemyson, 2021-09-06:
VIVANTE_SDK_DIR  should direct to our driver sdk root with following structure:
/sdk/
      /include/
     /drivers(lib)/ libOpenVX.so

nullptr-leo, 2021-09-07:
> VIVANTE_SDK_DIR should direct to our driver sdk root with following structure:
> /sdk/
> ----/include/
> ----/drivers(lib)/ libOpenVX.so

Thanks.

It turns out the issue is caused by Rockchip mini library `libVSC_Lite.so`. When `libVSC.so` used, it works fine.

closed 2021-09-07 by nullptr-leo