VeriSilicon/TIM-VX #171: Add interface to create handle tensor using user-allocated buffer

Goose-Bomb, 2021-09-08:
Right now the `Graph::CreateTensor` method can only copy the buffer over instead of using the buffer as the underlying I/O memory of the tensor.

However, the ovxlib did provide the API to do so, but which is not utilized by the wrapper class:
https://github.com/VeriSilicon/TIM-VX/blob/633075f689b2519ee91f770dbd31024df25ad040/src/tim/vx/tensor.cc#L184-L188
where a null pointer is passed.

Is it possible to pass the buffer pointer to the underlying interface to avoid unnecessary copy?

Goose-Bomb, 2021-09-08:
It could be like this:
```c++
  bool isHandleTensor = (spec_.attr_ & TensorAttribute::INPUT) ||
                        (spec_.attr_ & TensorAttribute::OUTPUT);
  
  if (isHandleTensor) {
    id_ = vsi_nn_AddTensorFromHandle(graph_->graph(), VSI_NN_TENSOR_ID_AUTO,
                                     &attr, reinterpret_cast<uint8_t*>(data_));
  } else {
    id_ = vsi_nn_AddTensor(graph_->graph(), VSI_NN_TENSOR_ID_AUTO, &attr,
                           nullptr);
  }

  if (VSI_NN_TENSOR_ID_NA == id_) {
    VSILOGE("Create tensor fail!");
    return false;
  }

  if (data_ != nullptr && !isHandleTensor) {
    if (!CopyDataToTensor(data_, 0)) {
      VSILOGE("Copy data to tensor fail!");
      return false;
    }
  }
```
however, the `data_` member variable is `const void*`, maybe using `const_cast` or change the class interface (e.g., add a `Graph::CreateTensorFromHandle` method to avoid confusion?)

sunshinemyson, 2021-09-16:
@Goose-Bomb , yes, standalone API is prefered in this case.

Graph::CreateTensorFromHandle() is reasonable.

Goose-Bomb, 2021-10-20:
> @Goose-Bomb,
> 
> Are you still active on this item?

Yeah, and I'm willing to contribute to this, but I don't know if this will break the current interface. Similar PR like #181 which replaces some public fields of classes like `TensorSpec` and `Quantization` to setter/getter is not merged. So I'm expecting maintainers from VeriSilicon to make this change. If you guys have no plans on doing this, I can try to contribute on this.

sunshinemyson, 2021-10-21:
#181  already get merged. It just supports the getter with the const modifier. 

We don't have a plan to do this in short term. We are welcome and appreciate it if you can contribute to this.

sunshinemyson, 2022-01-12:
@Goose-Bomb  I would like to close this issue because we decide to not support wrap user memory in TIM-VX. We already support import DMA buffer file descriptor in Tim-vx. We think it should be able to meet most requirements.

closed 2022-01-12 by sunshinemyson