VeriSilicon/TIM-VX #173: Handle tensor double free error on x86_64 simulator driver

Goose-Bomb, 2021-09-09:
I try to use the `vsi_nn_AddTensorFromHandle` to create a tensor using buffer allocated by cv::Mat, every time the program exits, there will be a double free error issued by OpenCV. It seems that the passed buffer is freed by the OpenVX driver when the context is deinitialized, so when the OpenCV try to free the same buffer, it causes a  double free error.

I only encountered this problem with the x86_64 simulator driver, when the program runs on the target device using vendor-provided NPU driver, everything works well.

Here is a short program to reproduce this error:
```c++
#include <iostream>
#include <opencv2/core.hpp>
#include <vsi_nn_pub.h>

int main(int argc, char* argv[]) {
    vsi_status err = VSI_SUCCESS;

    auto matIn = cv::Mat(4, 4, CV_32F);
    auto matOut = cv::Mat(4, 4, CV_32F);

    cv::randu(matIn, -1.0F, 1.0F);
    matOut.setTo(0.0F);

    auto context = vsi_nn_CreateContext();
    auto graph = vsi_nn_CreateGraph(context, 2, 1);
    vsi_nn_SetGraphInputs(graph, nullptr, 1);
    vsi_nn_SetGraphOutputs(graph, nullptr, 1);

    vsi_nn_tensor_attr_t attr = {};
    attr.dtype.fmt = VSI_NN_DIM_FMT_NCHW;
    attr.dim_num = 4;
    attr.size[0] = 4;
    attr.size[1] = 4;
    attr.size[2] = 1;
    attr.size[3] = 1;
    attr.dtype.vx_type = VSI_NN_TYPE_FLOAT32;
    attr.dtype.qnt_type = VSI_NN_QNT_TYPE_NONE;
    attr.is_const = 0;
    attr.vtl = 0;

    auto tensorIn = vsi_nn_AddTensorFromHandle(
        graph, VSI_NN_TENSOR_ID_AUTO, &attr, matIn.data);
    auto tensorOut = vsi_nn_AddTensorFromHandle(
        graph, VSI_NN_TENSOR_ID_AUTO, &attr, matOut.data);

    auto nodeReLU = vsi_nn_AddNode(graph, VSI_NN_OP_RELU, 1, 1, nullptr);
    nodeReLU->uid = 100;
    nodeReLU->input.tensors[0] = tensorIn;
    nodeReLU->output.tensors[0] = tensorOut;

    graph->input.tensors[0] = tensorIn;
    graph->output.tensors[0] = tensorOut;

    err = vsi_nn_SetupGraph(graph, vx_false_e);
    err = vsi_nn_VerifyGraph(graph);
    err = vsi_nn_rnn_RunGraph(graph);

    std::cout << "[Input]\n" << matIn << std::endl;
    std::cout << "[Output]\n" << matOut << std::endl;

    vsi_nn_ReleaseGraph(&graph);
    vsi_nn_ReleaseContext(&context);

    return err;
}
```

sunshinemyson, 2021-09-11:
@Goose-Bomb ,

Could you send an email to ML_Support@verisilicon.com for this issue? We can discuss more detail in email.

Goose-Bomb, 2021-09-11:
Sure

Goose-Bomb, 2021-09-12:
I tried to send an email to this mailbox but it said `authentication required`  and my email won't be delivered.

sunshinemyson, 2021-09-14:
I contacted our IT group. We will fix it. Sorry for inconvenience.

I checked your program, we have an alignment requirement for memory vsi_nn_CreateTensorFromHandle().

Please try follwing logical to see if we can avoid double-free issue or not.
tensor = vsi_nn_AddTensorFromHandle(
        graph, VSI_NN_TENSOR_ID_AUTO, &attr,  nullptr);

void * cpu_addr = 0;
vsi_nn_GetTensorHandle(tensor, &cpu_addr);
memcpy(mainIn.data, cpu_addr, sizeof(...));

vsi_nn_FlushHandle(tensor);

Goose-Bomb, 2021-09-17:
Yes, it did prevent the double free error, but cannot avoid copying. I'll wait for the next simulator driver update.

sunshinemyson, 2021-09-27:
@Goose-Bomb , 

We have a preview version for the simulator. https://github.com/VeriSilicon/TIM-VX/pull/178, you can try it.

Goose-Bomb, 2021-10-03:
> @Goose-Bomb ,
> 
> We have a preview version for the simulator. #178, you can try it.

Still have the problem when using `6.4.8.pre` version of simulator driver.

sunshinemyson, 2021-11-09:
uint8_t* bufferOut = (uint8_t*)aligned_alloc(ALIGNMENT, BUFFER_SIZE);

@Goose-Bomb ， after confirming with the low-level driver, we have an additional requirement for BUFFER_SIZE, please make sure it is aligned to 64Bytes. Thanks