VeriSilicon/TIM-VX #195: vsi_npu tvm compilation issue

sidphbot, 2021-10-20:
I am trying to cross-compile a pytorch model for imx8 with vsi_npu with below code but I am getting the errors noted below, I have also attached a working code for normal linux targets

compilation code:
```
mod, params = relay.frontend.from_pytorch(scripted_model, [(input_name, input_shape)])
target_string = "llvm  -mtriple=aarch64-linux-gnu"              # imx8 host-triple
kwargs = {"cc": "aarch64-linux-gnu-gcc", 'fcompile': False}
disabled_passes = ["AlterOpLayout"]                                     # same error with None
with tvm.transform.PassContext(opt_level=3, disabled_pass=disabled_passes):
        mod = vsi_npu.partition_for_vsi_npu(mod, params)
        lib = relay.build(mod, target_string, params=params)
lib.export_library(build_dir / 'deploy.so',  **kwargs)
```
error: (relay from quantized pytorch trace)
```
This is important----> name_node.value() == tvmgen_default_vsi_npu_593
GraphMakerImpl::Create
TensorMakerImpl::InferCall: qnn.quantize
E [GetMapedTensor:140]Tensor has not beed inserted in tensor map.
python: /code/tim-vx-lib/TIM-VX/src/tim/transform/layout_inference.cc:141: std::shared_ptr<tim::vx::Tensor> tim::transform::layout_inference_impl::LayoutInferContext::GetMapedTensor(const std::shared_ptr<tim::vx::Tensor>&) const: Assertion `false' failed.
Aborted (core dumped)
```
error: (relay from simple pytorch trace)
```
This is important----> name_node.value() == tvmgen_default_vsi_npu_57
GraphMakerImpl::Create
TensorMakerImpl::InferCall: nn.conv2d
TensorMakerImpl::InferCall: add
TensorMakerImpl::InferCall: image.resize2d
TensorMakerImpl::InferCall: add
TensorMakerImpl::InferCall: image.resize2d
TensorMakerImpl::InferCall: add
TensorMakerImpl::InferCall: image.resize2d
W [vsi_nn_SortGraphNode:1378]Unprocessed node 7
W [vsi_nn_SetupGraph:706]Sort graph nodes failure.
Fatal error: compile to binary failed
```

Working Code without `vsi_npu`: (quantized/normal/tuned/notune all cases)
```
mod, params = relay.frontend.from_pytorch(scripted_model, [(input_name, input_shape)])
target_string = "llvm -mtriple=x86_64-linux-gnu"              # linux host-triple
with tvm.transform.PassContext(opt_level=3, disabled_pass=None):
        lib = relay.build(mod, target_string, params=params)
lib.export_library(build_dir / 'deploy.so')
```
I have installed using cmake instructions from `https://github.com/VeriSilicon/tvm/blob/vsi_npu/README.VSI.md` 
Please guide me on the error

sidphbot, 2021-11-05:
@antkillerfarm @sunshinemyson did you get a chance to look into this?

antkillerfarm, 2021-11-08:
I am handling this issue from last two weeks, but still have some problems on pytorch import...

sidphbot, 2021-11-15:
thank you for looking into it..
i don't know if this may help(maybe comparable or can be rebased with supervision) but the vsi_npu partitioning module works for pytorch(un-quantized) models from [NXP tvm fork](https://source.codeaurora.org/external/imx/eiq-tvm-imx)
this fork however is still not tvm stable version 0.7.0 meaning it does not have pytorch changes that allow quantized models as well, hence we are depending on versilicon's fork of tvm as it has vsi_npu module along with the latest tvm pytorch changes

antkillerfarm, 2021-12-07:
@sidphbot You can try this branch:
https://github.com/VeriSilicon/tvm/tree/upstream/vsi_npu
with latest TIM-VX.
torchvision.models.quantization.mobilenet_v2 is OK now.

closed 2022-09-28 by sunshinemyson