VeriSilicon/TIM-VX #217: Can not use Matmul ops with x86 pre-built SDK

zihaomu, 2021-11-11:
Hi,

I got the following error Log. When I run the [Test unit sample](https://github.com/VeriSilicon/TIM-VX/blob/main/src/tim/vx/ops/matmul_test.cc).
But my computer can run "sample/lenet_int8" correctly. 

```bash
W [_matrixmulsetup:77]Call vxBatchGemmNode fail.

(10:0) : error : Error(0,10) : Cannot find the header file cl_viv_vx_ext.h.
(274:0) : error : Error(0,274) : Cannot find the header file cl_viv_vx_ext.h.
(33:0) : error : undefined identifier: 'COPY'
(73:0) : error : undefined identifier: 'COPY'
(280:0) : error : syntax error at 'VXC_512Bits'

ERROR: Failed to compile vx shader. (error: FFFFFFFF)
E [_gpu_register:477]Build program fail.
E [vsi_nn_kernel_create_node:633]Register client kernel com.vivantecorp.extension.evis.gemm_transb_U8U8toU8 fail with -1.
```

Best, 
Zihao

sunshinemyson, 2021-11-11:
Please set VIVANTE_SDK_DIR as

export VIVANTE_SDK_DIR=`pwd`/prebuilt-sdk/x86_64_linux

zihaomu, 2021-11-11:
@sunshinemyson Thanks for your quick reply.
`export VIVANTE_SDK_DIR=pwd/prebuilt-sdk/x86_64_linux` dose not work for me.
I used the CMake to compile the TimVx project. 

Environment:
`export LD_LIBRARY_PATH=`pwd`/../../../prebuilt-sdk/x86_64_linux/lib:$LD_LIBRARY_PATH`
`export VIVANTE_SDK_DIR=pwd/prebuilt-sdk/x86_64_linux`

Now I get the following Error Log when I run `graph->Compile()`:
```bash
W [_matrixmulsetup:77]Call vxBatchGemmNode fail.
```

sunshinemyson, 2021-11-11:
Can you share full console log to me?

zihaomu, 2021-11-12:
@sunshinemyson 

### Compile console Log

<details>
<summary>Console Log</summary>
╭─moo@moo ~/tim-vx/TIM-VX/build ‹main› 
╰─$ make clean
╭─moo@moo ~/tim-vx/TIM-VX/build ‹main› 
╰─$ cmake -DTIM_VX_ENABLE_TEST=ON ..
Processing X86_64.cmake ...
found lib CLC at /home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libCLC.so
--->OVXDRV_LIBRARIES/home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libCLC.so
found lib GAL at /home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libGAL.so
--->OVXDRV_LIBRARIES/home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libGAL.so
found lib OpenVX at /home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libOpenVX.so
--->OVXDRV_LIBRARIES/home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libOpenVX.so
found lib OpenVXU at /home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libOpenVXU.so
--->OVXDRV_LIBRARIES/home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libOpenVXU.so
found lib VSC at /home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libVSC.so
--->OVXDRV_LIBRARIES/home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libVSC.so
found lib ArchModelSw at /home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libArchModelSw.so
--->OVXDRV_LIBRARIES/home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libArchModelSw.so
found lib NNArchPerf at /home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libNNArchPerf.so
--->OVXDRV_LIBRARIES/home/moo/tim-vx/TIM-VX/prebuilt-sdk/x86_64_linux/lib/libNNArchPerf.so
CMake Deprecation Warning at build/_deps/googletest-src/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at build/_deps/googletest-src/googlemock/CMakeLists.txt:45 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at build/_deps/googletest-src/googletest/CMakeLists.txt:56 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
src/tim/vx
src/tim/vx/internal
-- OVXLIB include directory: /home/moo/tim-vx/TIM-VX/src/tim/vx/internal/include
-- Configuring done
-- Generating done
-- Build files have been written to: /home/moo/tim-vx/TIM-VX/build
╭─moo@moo ~/tim-vx/TIM-VX/build ‹main› 
╰─$ make -j8
[  0%] Building CXX object _deps/googletest-build/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[  1%] Building C object src/tim/vx/internal/CMakeFiles/tim_internal.dir/src/vsi_nn_client_op.c.o
[  1%] Building C object src/tim/vx/internal/CMakeFiles/tim_internal.dir/src/vsi_nn_graph_optimization.c.o
[  1%] Building C object src/tim/vx/internal/CMakeFiles/tim_internal.dir/src/vsi_nn_daemon.c.o
[  1%] Building C object src/tim/vx/internal/CMakeFiles/tim_internal.dir/src/vsi_nn_context.c.o
[  1%] Building C object src/tim/vx/internal/CMakeFiles/tim_internal.dir/src/vsi_nn_graph.c.o
[... 700 lines cut ...]
at index:3
[  FAILED  ] AddN.shape_2_2_uint8_Quantized (126 ms)
[----------] 3 tests from AddN (424 ms total)

[----------] 4 tests from AVG
[ RUN      ] AVG.shape_3_3_1_2_fp32_kernel_2_stride_1
[       OK ] AVG.shape_3_3_1_2_fp32_kernel_2_stride_1 (549 ms)
[ RUN      ] AVG.shape_3_3_1_1_fp32_kernel_2_stride_1
[       OK ] AVG.shape_3_3_1_1_fp32_kernel_2_stride_1 (552 ms)
[ RUN      ] AVG.shape_3_3_1_1_uint8_kernel_2_stride_1
[       OK ] AVG.shape_3_3_1_1_uint8_kernel_2_stride_1 (113 ms)
[ RUN      ] AVG.shape_60_52_3_5_fp32_kernel_35_stride_5
[       OK ] AVG.shape_60_52_3_5_fp32_kernel_35_stride_5 (2297 ms)
[----------] 4 tests from AVG (3511 ms total)

[----------] 2 tests from AVG_ANDROID
[ RUN      ] AVG_ANDROID.shape_60_52_3_5_fp32_kernel_35_stride_5
[       OK ] AVG_ANDROID.shape_60_52_3_5_fp32_kernel_35_stride_5 (2283 ms)
[ RUN      ] AVG_ANDROID.shape_60_52_3_5_uint8_kernel_35_stride_5
(3:0) : error : Error(0,3) : Cannot find the header file cl_viv_vx_ext.h.
(5:0) : error : syntax error at 'VXC_512Bits'

ERROR: Failed to compile vx shader. (error: FFFFFFFF)
[1]    23212 segmentation fault (core dumped)  ./unit_test
</details>

zihaomu, 2021-11-12:
Hi, @sunshinemyson, issue was solved.
It's my mistake, not the `W [_matrixmulsetup:77]Call vxBatchGemmNode fail.`
Thanks for your reply.

Friendly reminder, for CMake Compiling:
I need to add two environment Variables to run the `unit_test`.
`export LD_LIBRARY_PATH=`pwd`/../../../prebuilt-sdk/x86_64_linux/lib:$LD_LIBRARY_PATH`
`export VIVANTE_SDK_DIR=`pwd`/prebuilt-sdk/x86_64_linux`

Maybe we should add these two in **README** file.

closed 2021-11-13 by zihaomu