VeriSilicon/TIM-VX #227: How can I debug a tim-vx application using GDB?

xiangjunhuang, 2021-11-25:
Hi,

I run into problem when I'm trying to attach gdb to debug a tim-vx based application with x86 emulator. Same issue can be reproduced with lenet example. 

Here is the problem: I can run lenet successfully in command line, 
$ ./lenet
 --- Top5 ---
  0: 0.999520
  5: 0.000148
  6: 0.000148
  2: 0.000102
  9: 0.000058

But when I attach gdb, it stuck before entering main() with "Unable to connect to server!" message print. Press ctrl+C and check call stack as below:

Starting program: /home/user/work/TIM-VX-main/build/samples/lenet/lenet
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Unable to connect to server!
^C
Program received signal SIGINT, Interrupt.
0x00007ffff73de237 in __libc_accept (fd=3, addr=..., len=0x7fffffffcb0c) at ../sysdeps/unix/sysv/linux/accept.c:26
26      ../sysdeps/unix/sysv/linux/accept.c: No such file or directory.
(gdb) bt
#0  0x00007ffff73de237 in __libc_accept (fd=3, addr=..., len=0x7fffffffcb0c) at ../sysdeps/unix/sysv/linux/accept.c:26
#1  0x00007ffff498cb21 in DebugServer::intialDB() () from /home/user/work/TIM-VX-main/build/install/lib/libEmulator.so
#2  0x00007ffff495fddf in cmoTopLevel::cmoTopLevel(bool, bool, unsigned int, unsigned int, unsigned int, unsigned int) ()
   from /home/user/work/TIM-VX-main/build/install/lib/libEmulator.so
#3  0x00007ffff492b94b in CModelConstructor(Vivante::CEmulator*) () from /home/user/work/TIM-VX-main/build/install/lib/libEmulator.so
#4  0x00007ffff492c18f in Vivante::CEmulator::CEmulator(unsigned int, int, unsigned int, unsigned int) ()
   from /home/user/work/TIM-VX-main/build/install/lib/libEmulator.so
#5  0x00007ffff492cbd1 in CreateEmulator () from /home/user/work/TIM-VX-main/build/install/lib/libEmulator.so
#6  0x00007ffff53e4705 in CreateGAL () from /home/user/work/TIM-VX-main/build/install/lib/libGAL.so
#7  0x00007ffff7fe0b8a in ?? () from /lib64/ld-linux-x86-64.so.2
#8  0x00007ffff7fe0c91 in ?? () from /lib64/ld-linux-x86-64.so.2
#9  0x00007ffff7fd013a in ?? () from /lib64/ld-linux-x86-64.so.2
#10 0x0000000000000001 in ?? ()
#11 0x00007fffffffe2e3 in ?? ()
#12 0x0000000000000000 in ?? ()
(gdb)  

Looks it's sth related with the libEmulator.so, can anyone suggest what is the right way to do debug with GDB? 
Thanks,

sunshinemyson, 2021-11-25:
@xiangjunhuang ，

Yes, it's possible if you export "DISABLE_IDE_DEBUG=1" before use gdb

xiangjunhuang, 2021-11-25:
It works! 

thx! @sunshinemyson

closed 2021-11-25 by xiangjunhuang