VeriSilicon/TIM-VX #228: Compile issue in i.MX 8MP Yocto

michaelnguyen11, 2021-11-26:
Dear Supporters,

The i.MX 8MP BSP  5.10.52-2.1.0 is using TIM-VX 1.1.32, so with this version, the YoloV3 produces wrong output (can't detect object / detect wrong object), compare to TFLite NNAPI produces expected results.

Therefore I'm trying to bring latest TIM-VX version (1.1.34.fix) to i.MX 8MP BSP  5.10.52-2.1.0. However, the compile errors with message:

'''
/imx-yocto-bsp-5_10_52-2_1_0/build-xwayland/tmp/work/cortexa53-crypto-poky-linux/tim-vx/1.1.32-r0/git/src/tim/vx/ops/activations.cc: In constructor 'tim::vx::ops::Gelu::Gelu(tim::vx::Graph*, bool)':
/imx-yocto-bsp-5_10_52-2_1_0/build-xwayland/tmp/work/cortexa53-crypto-poky-linux/tim-vx/1.1.32-r0/git/src/tim/vx/ops/activations.cc:111:24: error: 'VSI_NN_OP_GELU' was not declared in this scope; did you mean 'VSI_NN_OP_ELU'?
  111 |     : Operation(graph, VSI_NN_OP_GELU){
      |                        ^~~~~~~~~~~~~~
      |                        VSI_NN_OP_ELU
/imx-yocto-bsp-5_10_52-2_1_0/build-xwayland/tmp/work/cortexa53-crypto-poky-linux/tim-vx/1.1.32-r0/git/src/tim/vx/ops/activations.cc:112:38: error: 'vsi_nn_nn_param_t' {aka 'union _vsi_nn_nn_param'} has no member named 'gelu'; did you mean 'elu'?
  112 |       this->impl()->node()->nn_param.gelu.approximate = approximate;
      |                                      ^~~~
      |                                      elu
/imx-yocto-bsp-5_10_52-2_1_0/build-xwayland/tmp/work/cortexa53-crypto-poky-linux/tim-vx/1.1.32-r0/git/src/tim/vx/ops/activations.cc: In member function 'virtual std::shared_ptr<tim::vx::Operation> tim::vx::ops::Gelu::Clone(std::shared_ptr<tim::vx::Graph>&) const':
/imx-yocto-bsp-5_10_52-2_1_0/build-xwayland/tmp/work/cortexa53-crypto-poky-linux/tim-vx/1.1.32-r0/git/src/tim/vx/ops/activations.cc:116:70: error: 'vsi_nn_nn_param_t' {aka 'union _vsi_nn_nn_param'} has no member named 'gelu'; did you mean 'elu'?
  116 |   return graph->CreateOperation<Gelu>(this->impl()->node()->nn_param.gelu.approximate);
      |                                                                      ^~~~
      |                                                                      elu
'''

Could you help me solve this issue ? Many thanks in advance !

Best regards,
Hiep Nguyen

sunshinemyson, 2021-11-26:
@michaelnguyen11 I've forward your issue to NXP guys. I don't have environment to reproduce it.

michaelnguyen11, 2021-11-26:
Thank @sunshinemyson for the information. Hope to receive from NXP guys soon.

robert-kalmar, 2021-11-26:
@michaelnguyen11 assume you are using the Yocto to build the TIM-VX. To compile the new version of TIM-VX turn off the TIM_VX_USE_EXTERNAL_OVXLIB CMake switch: 
```
EXTRA_OECMAKE =  " \
   ...
    -DTIM_VX_USE_EXTERNAL_OVXLIB=off \
    ...
```

michaelnguyen11, 2021-11-30:
Hi @robert-kalmar ,

Thank you for your response. By switching DTIM_VX_USE_EXTERNAL_OVXLIB=off, the latest TIM-VX built successfully.
However, when I do ```bitbake imx-image-full``` , the errors appears:

```
ERROR: tim-vx-1.1.32-r0 do_package: QA Issue: tim-vx: Files/directories were installed but not shipped in any package:
  /usr/lib64/libtim-vx.so
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
tim-vx: 1 installed and not shipped files. [installed-vs-shipped]
ERROR: tim-vx-1.1.32-r0 do_package: Fatal QA errors found, failing task.
ERROR: Logfile of failure stored in: /workdir/NXP/Boards/Common/Yocto/imx-yocto-bsp-5_10_52-2_1_0/build-xwayland/tmp/work/cortexa53-crypto-poky-linux/tim-vx/1.1.32-r0/temp/log.do_package.16277
ERROR: Task (/workdir/NXP/Boards/Common/Yocto/imx-yocto-bsp-5_10_52-2_1_0/sources/meta-imx/meta-ml/recipes-libraries/tim-vx/tim-vx_1.1.32.bb:do_package) failed with exit code '1'
```

The latest TIM-VX can't be shipped and packaged, therefore bitbake imx-image-full command failed.

Did you see this issue before ? Could you help me solve this issue ?

Many thanks in advance !.

Regards,
Hiep

michaelnguyen11, 2021-11-30:
I would like to confirm that with ```TIM_VX_USE_EXTERNAL_OVXLIB=off ``` release tag v1.1.34.fix, the Yocto BSP 5.10.52-2.1.0 could built successfully. With latest code, it built failure as above comment.

robert-kalmar, 2021-11-30:
@michaelnguyen11  for the latest code please revert this commit 62a33ecfde063a9c82fd9527eaba63a501081fa9.

michaelnguyen11, 2021-12-01:
Hi @robert-kalmar ,

It worked like a charm. Thank you for your support !

closed 2021-12-01 by michaelnguyen11

SHagerGEL, 2022-01-04:
Just disabling the use of external ovxlib is more a work around than a solution.
I hit the same problem but could is solve by updating the external ovxlib to a new Version which supports the GELU activation.