VeriSilicon/TIM-VX #241: does aarch64_A311D_6.4.8.tgz support Linux Kernel 5.1x?

ideafold, 2021-12-18:
rt.
Thanks.

ideafold, 2021-12-19:
uname -a:
Linux Khadas 5.16.0-rc2 #1.0.9 SMP Fri Dec 17 14:44:28 CST 2021 aarch64 aarch64 aarch64 GNU/Linux

aarch64_A311D_6.4.8 run with tengine lite failed.

gdb bt:
[image]

sunshinemyson, 2021-12-19:
You can check dmesg if it compatible with your kernel or not when execute "insmod"

ideafold, 2021-12-20:
aarch64_A311D_6.4.8/lib# insmod galcore.ko
insmod: ERROR: could not insert module galcore.ko: Invalid module format

dmesg said:
[  306.446813] galcore: version magic '4.9.y SMP preempt mod_unload modversions aarch64' should be '5.16.0-rc2 SMP mod_unload aarch64'

> You can check dmesg if it compatible with your kernel or not when execute "insmod"

aarch64_A311D_6.4.8/lib# insmod galcore.ko
insmod: ERROR: could not insert module galcore.ko: Invalid module format

[  306.446813] galcore: version magic '4.9.y SMP preempt mod_unload modversions aarch64' should be '5.16.0-rc2 SMP mod_unload aarch64'

ideafold, 2021-12-20:
it looks like the aarch64_A311D_6.4.8 is compiled in kernel 4.9. Would you be kind of to compile one use 5.16 kernel?

or tell me the method to compile.

Thanks.

sunshinemyson, 2021-12-20:
@ideafold ,

No plan to do this yet. Please stay with kernel 4.9, kernel module required commercial license, so you cannot compile it by yourself.

sunshinemyson, 2021-12-21:
@ideafold ,

You can ask Khadas for the kernel driver build with 5.16 kernel. They should support this as vendor. Thanks

closed 2021-12-21 by sunshinemyson