VeriSilicon/TIM-VX #242: undefined symbol: vxBatchGemmNode at NXP imx8mpevk

leokuo0725, 2021-12-21:
Hi,

After I built TIM-VX 1.34.fix on libtim-vx.so, it shows that "undefined symbol: vxBatchGemmNode" when I load libtim-vx.so.
Then, use ldd libtim-vx.so to check problem. And all .so files can be found.
	linux-vdso.so.1 (0x0000ffff9a0a6000)
	libCLC.so => /usr/lib/libCLC.so (0x0000ffff9954b000)
	libGAL.so => /usr/lib/libGAL.so (0x0000ffff99384000)
	libOpenVX.so.1 => /usr/lib/libOpenVX.so.1 (0x0000ffff99131000)
	libOpenVXU.so => /usr/lib/libOpenVXU.so (0x0000ffff99118000)
	libVSC.so => /usr/lib/libVSC.so (0x0000ffff9817e000)
	libArchModelSw.so => /usr/lib/libArchModelSw.so (0x0000ffff98156000)
	libNNArchPerf.so => /usr/lib/libNNArchPerf.so (0x0000ffff98131000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x0000ffff97f5a000)
	libm.so.6 => /lib/libm.so.6 (0x0000ffff97eb0000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x0000ffff97e8b000)
	libc.so.6 => /lib/libc.so.6 (0x0000ffff97d16000)
	libpthread.so.0 => /lib/libpthread.so.0 (0x0000ffff97ce7000)
	/lib/ld-linux-aarch64.so.1 (0x0000ffff9a074000)
	libdl.so.2 => /lib/libdl.so.2 (0x0000ffff97cd1000)

I use grep with recursive to find vxBatchGemmNode in /usr/lib/ at NXP imx8mpevk, but get empty.
The version is Linux imx8mpevk 5.10.52-lts.
At VIM3Pro, vxBatchGemmNode is in libOpenVX.so.1.3.0, X86 is also.

sunshinemyson, 2021-12-21:
@leokuo0725 ,

Please set TIM_VX_USE_EXTERNAL_OVXLIB=ON and build it again. After use this flag, we will try to find ovxlib from /usr/lib/. if it cannot detect the library, you can set OVXLIB_INC OVXLIB_LIB to the correct directory manually.

leokuo0725, 2021-12-21:
@sunshinemyson  cmake -DCONFIG=YOCTO -DTIM_VX_USE_EXTERNAL_OVXLIB=ON ..
-- Using YOCTO Project configuration.
src/tim/vx
-- Using external OVXLIB from OVXLIB_LIB-NOTFOUND
CMake Error at src/tim/CMakeLists.txt:26 (message):
  Set OVXLIB_INC if using external OVXLIB (TIM_VX_USE_EXTERNAL_OVXLIB)

-- Configuring incomplete, errors occurred!
See also "/home/root/TIM-VX/TIM-VX/build/CMakeFiles/CMakeOutput.log".

Could you provide OVXLIB_INC OVXLIB_LIB PATH at NXP EVB?

leokuo0725, 2021-12-21:
@sunshinemyson cmake -DCONFIG=YOCTO -DTIM_VX_USE_EXTERNAL_OVXLIB=ON -DOVXLIB_INC="/home/root/TIM-VX/TIM-VX/src/tim/vx/internal/include" ..
-- Using YOCTO Project configuration.
src/tim/vx
-- Using external OVXLIB from /usr/lib
-- OVXLIB include directory: /home/root/TIM-VX/TIM-VX/src/tim/vx/internal/include
-- Configuring done
-- Generating done
-- Build files have been written to: /home/root/TIM-VX/TIM-VX/build

I don't have idea about OVXLIB_LIB.

sunshinemyson, 2021-12-22:
OVXLIB_LIB should be the full path direct to libovxlib.so.

Please don't use internal ovxlib if you enabled external ovxlib. You should be able to find the ovxlib headers/lib from the system such as /usr/include/ and /usr/lib/

sunshinemyson, 2021-12-22:
https://github.com/VeriSilicon/TIM-VX/issues/228 
Please check comments in this issue. It might help you solve your problem.

fengwang, 2022-01-19:
I had a similar problem with OVXLIB, and I had to revert to a previous version to make the program compile. Is there a way to fix this problem without going back to a historical version?

See <https://github.com/OAID/Tengine/issues/1257>

fengwang, 2022-02-16:
A dirty patch to fix this problem by providing empty implementation:

create  a new cpp file with source code below

```cpp
#include <cassert>
#include <cstddef>
#include <cstdint>

extern "C"
{

typedef size_t vx_size;
typedef int32_t vx_int32;
typedef int32_t vx_enum;
typedef struct _vx_node *vx_node;
typedef struct _vx_graph *vx_graph;
typedef struct _vx_tensor_t *vx_tensor;
typedef struct _vx_scalar *vx_scalar;
typedef struct _vx_lut *vx_lut;
typedef struct _vx_node *vx_node;

typedef struct _vx_nn_convolution_3d_params_t
{
    vx_int32 padding_w_left;                 /*!< \brief Number of elements added at each side in the left of w dimension of the input. */
    vx_int32 padding_w_right;                /*!< \brief Number of elements added at each side in the right of w dimension of the input. */
    vx_int32 padding_h_top;                  /*!< \brief Number of elements added at each side in the top of h dimension of the input. */
    vx_int32 padding_h_bottom;               /*!< \brief Number of elements added at each side in the bottom of h dimension of the input. */
    vx_int32 padding_d_front;                /*!< \brief Number of elements added at each side in the front of d dimension of the input. */
    vx_int32 padding_d_rear;                 /*!< \brief Number of elements added at each side in the rear of d dimension of the input. */

    vx_int32 stride_w;                       /*!< \brief  skip w jump for down scale.  */
    vx_int32 stride_h;                       /*!< \brief  skip h jump for down scale.  */
    vx_int32 stride_d;                       /*!< \brief  skip d jump for down scale.  */
    vx_int32 dilation_w;                     /*!< \brief "inflate" the kernel by inserting zeros between the kernel elements in the w direction. The value is the number of zeros to insert.*/
    vx_int32 dilation_h;                     /*!< \brief "inflate" the kernel by inserting zeros between the kernel elements in the h direction. The value is the number of zeros to insert.*/
    vx_int32 dilation_d;                     /*!< \brief "inflate" the kernel by inserting zeros between the kernel elements in the d direction. The value is the number of zeros to insert.*/

    vx_enum pad_mode;                       /*!< \brief A VX_TYPE_ENUM of the <tt> \ref vx_pad_mode_e </tt> enumeration. */
    vx_scalar pad_const;                    /*!< \brief pad const value if setting pad mode to const, the const value is base value, not quantized value. */

    vx_enum overflow_policy;                /*!< \brief A <tt> VX_TYPE_ENUM</tt> of the <tt> vx_convert_policy_e</tt> enumeration. */
    vx_enum rounding_policy;                /*!< \brief A <tt> VX_TYPE_ENUM</tt> of the <tt> vx_round_policy_e</tt> enumeration. */
    vx_enum down_scale_size_rounding;       /*!< \brief Rounding method for calculating output dimensions. See <tt>\ref vx_nn_rounding_type_e</tt> */

    vx_int32 depth_multiplier;              /*!< \brief depthwise multiplier value, if 0, means convolution, elsewise(>=1), the convolution is depthwiseconvolution. */
}vx_nn_convolution_3d_params_t;

vx_node  vxBatchGemmNode(vx_graph graph,
                                                 vx_tensor matrix_a,
                                                 vx_tensor matrix_b,
                                                 vx_tensor matrix_c,
                                                 vx_scalar trans_a,
                                                 vx_scalar trans_b,
                                                 vx_scalar trans_c,
                                                 vx_tensor output)
{
    assert( false && "vxBatchGemmNode is not yet implemented." );
    return NULL;
}

vx_node  vxTensorTableLookupLayer(
    vx_graph graph,
    vx_tensor input,
    vx_lut InLut,
    vx_lut OutLut,
    vx_tensor output)
{
    assert( false && "vxTensorTableLookupLayer is not yet implemented." );
    return NULL;
}

vx_node  vxConv3dLayer(vx_graph graph, vx_tensor inputs, vx_tensor weights, vx_tensor biases, const vx_nn_convolution_3d_params_t *convolution_params, vx_size size_of_convolution_params, vx_tensor outputs)
{
    assert( false && "vxConv3dLayer is not yet implemented." );
    return NULL;
}

}

```

This will temperately fix the problem, providing you will not use these interfaces in your code.

fengwang, 2022-02-21:
@robert-kalmar Thanks for reminding me of this. I was using a relatively old version SDK. I shall upgrade it and come back to you.

fengwang, 2022-03-04:
> 

I upgrade the SDK to the most recent one, with Linux build version `Linux imx8mpevk 5.10.72-lts-5.10.y+ga68e31b63f86  #1 SMP PREEMPT Tue Nov 23 06:02:20 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux`

But unfortunately I still have those symbols not found.

Still, I noticed this macro is **NOT** defined in the header files:

A quick test program
```cpp
#include <algorithm>
#include <iomanip>
#include <iostream>
#include <tuple>
#include <vector>

#include "tim/vx/context.h"
#include "tim/vx/graph.h"
#include "tim/vx/operation.h"
#include "tim/vx/ops/activations.h"
#include "tim/vx/ops/conv2d.h"
#include "tim/vx/ops/fullyconnected.h"
#include "tim/vx/ops/pool2d.h"
#include "tim/vx/ops/softmax.h"
#include "tim/vx/tensor.h"

int main()
{
	#ifdef VX_BATCH_GEMM_API_SUPPORT
	std::cout << "This macro is defined " << std::endl;
	#else
	std::cout << "This macro is not defined " << std::endl;
	#endif

	return 0;
}

```

gives output:

`This macro is not defined`

closed 2022-09-28 by sunshinemyson