VeriSilicon/TIM-VX #246: Div_uint8 returning all zero data

xiangjunhuang, 2021-12-24:
Hi,

Div uint8 is returning all zero data incorrectly. The behavior can be reproduced by following test code:
`

uint8_t data1[] = {1,2,3,4,5,6};
uint8_t data2[] = {2,2,2,2,2,2};

int main(int argc, char** argv) {
  (void) argc, (void) argv;
  auto context = tim::vx::Context::Create();
  auto graph = context->CreateGraph();

  tim::vx::ShapeType input_shape({2, 3, 1, 1});
  tim::vx::Quantization input_quant(tim::vx::QuantType::ASYMMETRIC, 1.0,
                                    0);
  tim::vx::TensorSpec input_spec(tim::vx::DataType::UINT8, input_shape,
                                 tim::vx::TensorAttribute::INPUT, input_quant);
  auto input1 = graph->CreateTensor(input_spec, data1);
  auto input2 = graph->CreateTensor(input_spec, data2);

  tim::vx::ShapeType output_shape({2,3, 1, 1});
  tim::vx::Quantization output_quant(tim::vx::QuantType::ASYMMETRIC, 1.0,
                                    0);
  tim::vx::TensorSpec output_spec(tim::vx::DataType::UINT8, output_shape,
                                  tim::vx::TensorAttribute::OUTPUT, output_quant);
  auto output = graph->CreateTensor(output_spec);

  auto op = graph->CreateOperation<tim::vx::ops::Div>();
  (*op)
      .BindInputs({input1, input2})
      .BindOutputs({output});

  if (!graph->Compile()) {
    std::cout << "Compile graph fail." << std::endl;
    return -1;
  }

  graph->PrintGraph();

  if (!graph->Run()) {
    std::cout << "Run graph fail." << std::endl;
    return -1;
  }

  std::vector<uint8_t> output_data;
  output_data.resize(1 * 10);
  if (!output->CopyDataFromTensor(output_data.data())) {
    std::cout << "Copy output data fail." << std::endl;
    return -1;
  }

  printTopN(output_data.data(), output_data.size(), 6);

  return 0;
}
`

With test code above, below output can be observed:
`
I [vsi_nn_PrintGraph:1573]Graph:
I [vsi_nn_PrintGraph:1574]***************** Tensors ******************
D [print_tensor:150]id[   0] vtl[0] const[0] shape[ 2, 3, 1, 1       ] fmt[u8 ] qnt[ASM zp=  0, scale=1.000000]
D [print_tensor:150]id[   1] vtl[0] const[0] shape[ 2, 3, 1, 1       ] fmt[u8 ] qnt[ASM zp=  0, scale=1.000000]
D [print_tensor:150]id[   2] vtl[0] const[0] shape[ 2, 3, 1, 1       ] fmt[u8 ] qnt[ASM zp=  0, scale=1.000000]
I [vsi_nn_PrintGraph:1583]***************** Nodes ******************
I [vsi_nn_PrintNode:203](          DIVIDE)node[0] [in: 0, 1 ], [out: 2 ] [94c22fa0]
I [vsi_nn_PrintGraph:1592]******************************************
D [_check_swapped_tensors:112]Check swapped tensors
***********dump layer : 0***************
 --- Top6 ---
  0: 0
  1: 0
  2: 0
  3: 0
  4: 0
  5: 0
`

Please help look into this issue.

sunshinemyson, 2021-12-27:
@xiangjunhuang , 

Would you mind share which sw-stack used for this issue? X86 prebuilt or VIM3?

xiangjunhuang, 2021-12-28:
x86 prebuilt.

chxin66, 2021-12-29:
Don’t use CPU software, use our NPU shader will correct

chxin66, 2021-12-29:
Check that the program is executed in SW or SH
[image]

sunshinemyson, 2021-12-29:
@xiangjunhuang ,

We've added your test case to our test suit. [----------] 4 tests from Div and it can pass with our prebuilt x86 simulator. 

You can find it from https://github.com/VeriSilicon/TIM-VX/pull/251. And You can also check debug it locally as @chxin66 suggested. We suppose it should go with "SH"

xiangjunhuang, 2022-01-14:
> @xiangjunhuang Are able to try this again with our suggestion or with out unit_test?

Yes. Retried and confirmed result was good. Thanks!

closed 2022-01-14 by xiangjunhuang