VeriSilicon/TIM-VX #257: Problem during linking when cross compiling with cmake and external Toolchain for arm64

SHagerGEL, 2022-01-04:
I tried to build TIM-VX with an external Toolchain and external ovxlib.

My cmake call:
> ${CMAKE} .. -DOVXLIB_INC=${TARGET_SYSROOT}/usr/include/VX/ovxlib/include/ -DTIM_VX_USE_EXTERNAL_OVXLIB=ON -DOVXLIB_LIB=${TARGET_SYSROOT}/usr/lib/ -DCONFIG=YOCTO -DCMAKE_SYSROOT=${TARGET_SYSROOT}

This gives this output:
>-DOVXLIB_LIB=${TARGET_SYSROOT}/usr/lib/ -DCONFIG=YOCTO -DCMAKE_SYSROOT=${TARGET_SYSROOT}
>-- The C compiler identification is GNU 10.3.0
>-- The CXX compiler identification is GNU 10.3.0
>-- Check for working C compiler: /opt/toolchainxxxx/host/bin/aarch64-buildroot-linux-gnu-gcc
>-- Check for working C compiler: /opt/toolchainxxxx/host/bin/aarch64-buildroot-linux-gnu-gcc -- works
>-- Detecting C compiler ABI info
>-- Detecting C compiler ABI info - done
>-- Detecting C compile features
>-- Detecting C compile features - done
>-- Check for working CXX compiler: /opt/toolchainxxxx/host/bin/aarch64-buildroot-linux-gnu-g++
>-- Check for working CXX compiler: /opt/toolchainxxxx/host/bin/aarch64-buildroot-linux-gnu-g++ -- works
>-- Detecting CXX compiler ABI info
>-- Detecting CXX compiler ABI info - done
>-- Detecting CXX compile features
>-- Detecting CXX compile features - done
>-- Using YOCTO Project configuration.
>src/tim/vx
>-- Using external OVXLIB from /opt/toolchainxxxx/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib
>-- OVXLIB include directory: /opt/toolchainxxxx/host/aarch64-buildroot-linux-gnu/sysroot/usr/include/VX/ovxlib/include/
>-- Configuring done
>-- Generating done
>-- Build files have been written to: /home/xxxxxxx/tim-vx/build

Then calling make starts to build the object files:
>$ make tim-vx
>Scanning dependencies of target tim-vx
>[  1%] Building CXX object src/tim/CMakeFiles/tim-vx.dir/vx/context.cc.o
>[  2%] Building CXX object src/tim/CMakeFiles/tim-vx.dir/vx/direct_map_op.cc.o
> ... and so on until
> [ 98%] Building CXX object src/tim/CMakeFiles/tim-vx.dir/transform/ops/op_layout_inference.cc.o

But the linking process failed with the strange message:

>[100%] Linking CXX shared library libtim-vx.so
>/opt/toolchainxxxx/host/bin/../lib/gcc/aarch64-buildroot-linux-gnu/10.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: error: 
>/opt/toolchainxxxx/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib: read: Is a directory
>collect2: error: ld returned 1 exit status

After starting make again with verbose output the following was printed:
> cd /home/xxxxxxx/tim-vx/build/src/tim &&
> /opt/toolchainxxxx/host/bin/cmake -E cmake_link_script CMakeFiles/tim-vx.dir/link.txt --verbose=1
> /opt/toolchainxxxx/host/bin/aarch64-buildroot-linux-gnu-g++ --sysroot=/opt/toolchainxxxx/host/aarch64-buildroot-linux-gnu/sysroot 
>-fPIC  -Wall -Wextra -Werror  -shared -Wl,-soname,libtim-vx.so -o libtim-vx.so 
>CMakeFiles/tim-vx.dir/vx/context.cc.o CMakeFiles/tim-vx.dir/vx/direct_map_op.cc.o 
> ... and so on ... until
> CMakeFiles/tim-vx.dir/transform/ops/op_layout_inference.cc.o  
>-Wl,-rpath,/usr: -Wl,--no-whole-archive 
>-lCLC 
>-lGAL 
>-lOpenVX 
>-lOpenVXU 
>-lVSC 
>-lArchModelSw 
>-lNNArchPerf 
>/opt/toolchainxxxx/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib        <<<<< STRANGE
>-lCLC 
>-lGAL 
>-lOpenVX 
>-lOpenVXU 
>-lVSC 
>-lArchModelSw 
>-lNNArchPerf 
>/opt/toolchainxxxx/host/bin/../lib/gcc/aarch64-buildroot-linux-gnu/10.3.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: error: >/opt/toolchainxxxx/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib: read: Is a directory
>collect2: error: ld returned 1 exit status

The Linker seems to get the libraries twice and after the first library list there appears a path which is strange.

When i manually type the command 
* without the second block of repeated libraries
* without the strange path 
* and removing the 'colon' in:  -Wl,-rpath,/usr':'
The linking finished.
 
> cd /home/xxxxxxx/tim-vx/build/src/tim &&
> /opt/toolchainxxxx/host/bin/cmake -E cmake_link_script CMakeFiles/tim-vx.dir/link.txt --verbose=1
> /opt/toolchainxxxx/host/bin/aarch64-buildroot-linux-gnu-g++ --sysroot=/opt/toolchainxxxx/host/aarch64-buildroot-linux-gnu/sysroot 
>-fPIC  -Wall -Wextra -Werror  -shared -Wl,-soname,libtim-vx.so -o libtim-vx.so 
>CMakeFiles/tim-vx.dir/vx/context.cc.o CMakeFiles/tim-vx.dir/vx/direct_map_op.cc.o 
> ... and so on ... until
> CMakeFiles/tim-vx.dir/transform/ops/op_layout_inference.cc.o  
>-Wl,-rpath,/usr -Wl,--no-whole-archive 
>-lCLC 
>-lGAL 
>-lOpenVX 
>-lOpenVXU 
>-lVSC 
>-lArchModelSw 
>-lNNArchPerf 

File Type OK:
>$ file libtim-vx.so 
>libtim-vx.so: ELF 64-bit LSB shared object, ARM aarch64, version 1 (GNU/Linux), dynamically linked, with debug_info, not stripped

size:
>$ ll libtim-vx.so
>-rwxr-xr-x 1 aaaaa bbbbb 7774584 Jan  4 15:52 libtim-vx.so

Dependencies seem OK:
>$ readelf -d libtim-vx.so | grep NEEDED
> 0x0000000000000001 (NEEDED)             Shared library: [libCLC.so]
> 0x0000000000000001 (NEEDED)             Shared library: [libGAL.so]
> 0x0000000000000001 (NEEDED)             Shared library: [libOpenVX.so.1]
> 0x0000000000000001 (NEEDED)             Shared library: [libOpenVXU.so]
> 0x0000000000000001 (NEEDED)             Shared library: [libVSC.so]
> 0x0000000000000001 (NEEDED)             Shared library: [libArchModelSw.so]
> 0x0000000000000001 (NEEDED)             Shared library: [libNNArchPerf.so]
> 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
> 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
> 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
> 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]

What is the reason for that?

sunshinemyson, 2022-01-05:
OVXLIB_LIB you need provide libovxlib.so full name for this variable

SHagerGEL, 2022-01-10:
Thanks a lot! This solved the issue

closed 2022-01-10 by SHagerGEL