VeriSilicon/TIM-VX #262: Failing unit tests on target iMX8MP

SHagerGEL, 2022-01-10:
After successfully building TIM-VX i tried to run the demo programs as well as the unit tests on the target iMX8MP.

The following demo programs all run without issues:
* benchmark_test
* lenet
* multi_thread_test

But the unit tests failed many times until it finally stopped with a segmentation fault.

e.g. gelu

> ./unit_test --gtest_filter=Gelu*
> [==========] Running 2 tests from 1 test suite.
> [----------] Global test environment set-up.
> [----------] 2 tests from Gelu
> [ RUN      ] Gelu.shape_5_1_fp32_approximate
> 
> [       OK ] Gelu.shape_5_1_fp32_approximate (115 ms)
> [ RUN      ] Gelu.shape_5_1_uint8_Quantized
> 
> /home/workspacexxx/tim-vx/src/tim/vx/test_utils.h:118: Failure
> The difference between expected[i] and actual[i] is 3, which exceeds abs_error, where
> expected[i] evaluates to 130,
> actual[i] evaluates to 127, and
> abs_error evaluates to 1.
> at index:4
> [  FAILED  ] Gelu.shape_5_1_uint8_Quantized (3531 ms)
> [----------] 2 tests from Gelu (3647 ms total)

e.g. Conv2d.shape_w_h_128_1_ksize_1_1_str:
(This test is repeaded many times with the same result)

> ./unit_test --gtest_filter=Conv2d.shape_w_h_128_1_ksize_1_1_str
> ide_2_int8_QuantizedPerChannelTest
> Running main() from /home/workspacexxx/tim-vx/build/_deps/googletest-src/googletest/src/gtest_main.cc
> Note: Google Test filter = Conv2d.shape_w_h_128_1_ksize_1_1_stride_2_int8_QuantizedPerChannelTest
> [==========] Running 1 test from 1 test suite.
> [----------] Global test environment set-up.
> [----------] 1 test from Conv2d
> [ RUN      ] Conv2d.shape_w_h_128_1_ksize_1_1_stride_2_int8_QuantizedPerChannelTest
> /home/workspacexxx/tim-vx/src/tim/vx/ops/conv2d_test.cc:1626: Failure
> Value of: graph->Compile()
>   Actual: false
> Expected: true
> /home/workspacexxx/tim-vx/src/tim/vx/ops/conv2d_test.cc:1630: Failure
> Value of: graph->Run()
>   Actual: false
> Expected: true
> /home/workspacexxx/tim-vx/src/tim/vx/ops/conv2d_test.cc:1638: Failure
> Expected equality of these values:
>   golden
>     Which is: { '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), '\x7F' (127), ... }
>   output
>     Which is: { '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), '\xFF' (-1), ... }
> /home/workspacexxx/tim-vx/src/tim/vx/ops/conv2d_test.cc:1626: Failure
> Value of: graph->Compile()
>   Actual: false
> Expected: true
> /home/workspacexxx/tim-vx/src/tim/vx/ops/conv2d_test.cc:1630: Failure
> Value of: graph->Run()
>   Actual: false
> Expected: true

Segmentation Fault 1:
> ./unit_test --gtest_filter=TransposeConv2d.shape_4_4_1_1_int8_QuantizedPerChannelOneTest
> Running main() from /home/workspacexxx/tim-vx/build/_deps/googletest-src/googletest/src/gtest_main.cc
> Note: Google Test filter = TransposeConv2d.shape_4_4_1_1_int8_QuantizedPerChannelOneTest
> [==========] Running 1 test from 1 test suite.
> [----------] Global test environment set-up.
> [----------] 1 test from TransposeConv2d
> [ RUN      ] TransposeConv2d.shape_4_4_1_1_int8_QuantizedPerChannelOneTest
> Segmentation fault

Segmentation Fault 2:
> ./unit_test --gtest_filter=TransposeConv2d.shape_4_4_1_1_int8_QuantizedBiasPerChannelTest
> Running main() from /home/workspacexxx/tim-vx/build/_deps/googletest-src/googletest/src/gtest_main.cc
> Note: Google Test filter = TransposeConv2d.shape_4_4_1_1_int8_QuantizedBiasPerChannelTest
> [==========] Running 1 test from 1 test suite.
> [----------] Global test environment set-up.
> [----------] 1 test from TransposeConv2d
> [ RUN      ] TransposeConv2d.shape_4_4_1_1_int8_QuantizedBiasPerChannelTest
> Segmentation fault

After disabling the two segfaulting tests (TransposeConv2d.shape_4_4_1_1_int8_QuantizedPerChannelOneTest, TransposeConv2d.shape_4_4_1_1_int8_QuantizedBiasPerChannelTest) the final results look like this:
> [==========] 138 tests from 47 test suites ran. (171050 ms total)
> [  PASSED  ] 124 tests.
> [  FAILED  ] 14 tests, listed below:
> [  FAILED  ] Gelu.shape_5_1_uint8_Quantized
> [  FAILED  ] Conv2d.shape_w_h_128_1_ksize_1_1_stride_2_int8_QuantizedPerChannelTest
> [  FAILED  ] DeConv1d.no_bias_layout_whcn_depthwise_shape_3_2_1
> [  FAILED  ] DeConv2d.shape_3_3_2_1_float_depthwise
> [  FAILED  ] FloorDiv.shape_5_1_broadcast_uint8
> [  FAILED  ] Erf.shape_3_2_uint8_Quantized
> [  FAILED  ] Matmul.shape_2_3_2_shape_2_3_2_uint8_transpose_a
> [  FAILED  ] MaxpoolWithArgmax.shape_4_4_1_uint8_kernel_2_stride_2
> [  FAILED  ] MaxUnpool2d.shape_2_2_1_uint8_kernel_2_stride_2
> [  FAILED  ] Resize1d.shape_4_2_1_uint8_nearest_whcn
> [  FAILED  ] ScatterND.shape_9
> [  FAILED  ] SpatialTransformer.shape_1_3_3_1_u8
> [  FAILED  ] Tile.shape_3_2_1_int8_multiples_2_2_1
> [  FAILED  ] TransposeConv2d.shape_2_2_1_1_int8_QuantizedPerChannelTwoTest

It guess most of the failed tests are related to (u)int8 operations on the NPU, but not all of them.
For example the following test group succeeded which also contains 8bit Operations (i guess they ran on the NPU because of the long test time caused by NPU Warmup):
> [----------] 16 tests from DepthwiseConv
> [ RUN      ] DepthwiseConv.shape_2_3_2_1_float32_SimpleTest
> [       OK ] DepthwiseConv.shape_2_3_2_1_float32_SimpleTest (18 ms)
> [ RUN      ] DepthwiseConv.shape_2_3_2_1_float32_StrideValidTest
> [       OK ] DepthwiseConv.shape_2_3_2_1_float32_StrideValidTest (18 ms)
> [ RUN      ] DepthwiseConv.shape_2_3_2_1_float32_StrideSameTest
> [       OK ] DepthwiseConv.shape_2_3_2_1_float32_StrideSameTest (18 ms)
> [ RUN      ] DepthwiseConv.shape_2_3_2_1_float32_StrideSameDilationTest
> [       OK ] DepthwiseConv.shape_2_3_2_1_float32_StrideSameDilationTest (18 ms)
> [ RUN      ] DepthwiseConv.shape_2_3_2_1_float32_PaddingTest
> [       OK ] DepthwiseConv.shape_2_3_2_1_float32_PaddingTest (19 ms)
> [ RUN      ] DepthwiseConv.shape_9_9_1_1_float32_DilationValidTest
> [       OK ] DepthwiseConv.shape_9_9_1_1_float32_DilationValidTest (18 ms)
> [ RUN      ] DepthwiseConv.shape_3_3_1_1_float32_DilationSameTest
> [       OK ] DepthwiseConv.shape_3_3_1_1_float32_DilationSameTest (18 ms)
> [ RUN      ] DepthwiseConv.shape_3_3_4_2_float32_BatchValidTest
> [       OK ] DepthwiseConv.shape_3_3_4_2_float32_BatchValidTest (18 ms)
> [ RUN      ] DepthwiseConv.shape_2_2_1_4_float32_BatchSameTest
> [       OK ] DepthwiseConv.shape_2_2_1_4_float32_BatchSameTest (18 ms)
> [ RUN      ] DepthwiseConv.shape_2_3_2_1_uint8_QuantizedTest
> [       OK ] DepthwiseConv.shape_2_3_2_1_uint8_QuantizedTest (242 ms)
> [ RUN      ] DepthwiseConv.shape_9_9_1_1_uint8_QuantizedDilationdValidTest
> [       OK ] DepthwiseConv.shape_9_9_1_1_uint8_QuantizedDilationdValidTest (283 ms)
> [ RUN      ] DepthwiseConv.shape_3_3_1_1_uint8_QuantizedDilationdSameTest
> [       OK ] DepthwiseConv.shape_3_3_1_1_uint8_QuantizedDilationdSameTest (234 ms)
> [ RUN      ] DepthwiseConv.shape_3_2_2_1_int8_PerTensorTest
> [       OK ] DepthwiseConv.shape_3_2_2_1_int8_PerTensorTest (251 ms)
> [ RUN      ] DepthwiseConv.shape_3_2_2_1_int8_PerAxisTest
> [       OK ] DepthwiseConv.shape_3_2_2_1_int8_PerAxisTest (252 ms)
> [ RUN      ] DepthwiseConv.shape_3_3_8_1_int8_PerChannelValidTest
> [       OK ] DepthwiseConv.shape_3_3_8_1_int8_PerChannelValidTest (244 ms)
> [ RUN      ] DepthwiseConv.shape_3_3_8_1_int8_PerChannelSameTest
> [       OK ] DepthwiseConv.shape_3_3_8_1_int8_PerChannelSameTest (244 ms)
> [----------] 16 tests from DepthwiseConv (1915 ms total)

Has anybody run this unit tests on a target so far?
What can i do the make the tests working?

sunshinemyson, 2022-01-11:
@SHagerGEL Please share us which version of NXP bsp you are using. I'll forward this to NXP guys.

SHagerGEL, 2022-01-17:
There is no NXP BSP that i can use, since i need to use buildroot, which is not supported by NXP.
The board i am working on is a custom board which is based on an NXP reference Design with some changes that do not matter when it comes to NPU.
I need to take components close to the Yocto build configuration.

Today i realized that i did not use the latest Version of the iMX8 GPU Drivers which may be the issue for the unit_test errors.
I will upgrade to the latest versions and open a new issue if any failures appear.

closed 2022-01-17 by SHagerGEL

robert-kalmar, 2022-01-21:
@SHagerGEL let us know if the problem get resolved with upgrading the GPU driver.

SHagerGEL, 2022-01-24:
I switched to the latest VeriSilicon GPU Driver package: **imx-gpu-viv-6.4.3.p2.4-aarch64**, Linux Kernel based on **LF5.10.72_2.2.0**;
Now only two unit_tests fail, all the others pass.
But the two failing ones did not just fail, one results in a segmentation fault, the other in a stack smash:

**graph.gen_binary_graph_with_simple_add**

>/usr/local/bin # ./unit_test --gtest_filter=graph.gen_binary_graph_with_simple_add
>Running main() from /home/xxxxx/tim-vx/build/_deps/googletest-src/googletest/src/gtest_main.cc
>Note: Google Test filter = graph.gen_binary_graph_with_simple_add
>[==========] Running 1 test from 1 test suite.
>[----------] Global test environment set-up.
>[----------] 1 test from graph
>[ RUN      ] graph.gen_binary_graph_with_simple_add
>Segmentation fault

**StridedSlice.shape_**
>/usr/local/bin # ./unit_test --gtest_filter=StridedSlice.*
>Running main() from /home/xxxxx/tim-vx/build/_deps/googletest-src/googletest/src/gtest_main.cc
>Note: Google Test filter = StridedSlice.*
>[==========] Running 1 test from 1 test suite.
>[----------] Global test environment set-up.
>[----------] 1 test from StridedSlice
>[ RUN      ] StridedSlice.shape_
>*** stack smashing detected ***: terminated
>Aborted

If i disable this two tests, the rest seems to be ok:

>$ ./unit_test --gtest_filter=-graph.gen_binary_graph_with_simple_add:StridedSlice.*

>[----------] Global test environment tear-down
>[==========] 148 tests from 48 test suites ran. (192599 ms total)
>[  PASSED  ] 148 tests.