VeriSilicon/TIM-VX #263: Resize Op have some problem

lgao-matax, 2022-01-12:
resize have the problem:
1，resize_downsample_scales_linear,input X:type:float32[1,1,2,4],data:[[[[1,2,3,4],[5,6,7,8]]]];scales type:float32[4],data:[1,1,0.6,0.6]; 
onnx attr mode:linear,coordinate_transformation_mode:half_pixel; 
vx attr：reszietype：BILINEAR,factor：0.6,align_corners:false,half_pixel_center:true,target_height:0,target_width:0; 
onnx cpu output Y:type:float32[1,1,1,2],data:[[[[2.6666,4.3333]]]];
vx  output Y：type:float32[1,1,1,2],data:[[[[3.5,5.5]]]].

2，resize_downsample_scales_linear_align_conners,input X:type:float32[1,1,2,4],data:[[[[1,2,3,4],[5,6,7,8]]]];scales type:float32[4],data:[1,1,0.6,0.6]; 
onnx attr mode:linear,coordinate_transformation_mode:align_corners; 
vx attr：reszietype：BILINEAR,factor：0.6,align_corners:true,half_pixel_center:false,target_height:0,target_width:0; 
onnx cpu output Y:type:float32[1,1,1,2],data:[[[[1,3.14]]]];
vx  output Y：type:float32[1,1,1,2],data:[[[[1,4]]]].

3,test_resize_downsample_sizes_nearest:input X:type:float32[1,1,2,4],data:[[[[1,2,3,4],[5,6,7,8]]]];size type:int64[4],data:[1,1,1,3];
onnx attr mode:nearest,coordinate_transformation_mode:half_pixel;
vx attr：reszietype：NEAREST_NEIGHBOR,factor：0,align_corners:false,half_pixel_center:false,target_height:1,target_width:3;
onnx cpu output Y:type:float32[1,1,1,3],data:[[[[1,2,4]]]];
vx  output Ytype:float32[1,1,1,3],data:[[[[1,3,4]]]].

4,how to resize when onnx resize mode is cubic?

sunshinemyson, 2022-01-12:
We will check the detail  and get back to you ASAP. Thanks for your feedback.

sunshinemyson, 2022-01-14:
@gao,

Here is the feedback:
For item 1 & 2, it because  we calculate resize factor with output div by input. This could be different with ONNX since it asked us calculate output with input by scale factor.
For item 3, we have different policy for  integer rounding, we use floor.
For item 4, this is not implemented yet.

lgao-matax, 2022-01-14:
con't follow “ it because we calculate resize factor with output div by input”，when onnx give input scale ，how can we do  vx resize？
"For item 3, we have different policy for integer rounding, we use floor", may not ceil?  vx is f(1.33,0)->f(2,0)?

sunshinemyson, 2022-01-17:
@lgao-matax ,
In our internal implementation, we always calculate output shape by floor(2*0.6) = 1, then we think factor is 0.5 during resize.  We never use the factor from the ONNX. Then we have different behavior here (0.5 vs 0.6).

During resize, we need calculate coordinate (x,y), if coordinate cannot be integer, we use floor().

closed 2022-09-28 by sunshinemyson