VeriSilicon/TIM-VX #266: Softmax bug

bkovalenkocomp, 2022-01-12:
Hi, I think there is a bug in the softmax layer.

I have detector model (the same ;-) ), here are probabilities that are higher than `0.5`:

```
FP32 vx-delegate
Score: 0.656692
Score: 0.75131
Score: 0.980652
Score: 0.930825
Score: 0.716989
Score: 0.522719

INT8 vx-delegate
Score: 0.605469
Score: 0.746094
Score: 0.984375
Score: 0.945312
Score: 0.734375

INT8 arm
Score: 0.84375
Score: 0.996094
Score: 0.988281
Score: 0.992188
Score: 0.996094
Score: 0.996094
Score: 0.996094
Score: 0.949219
Score: 0.898438
Score: 0.996094
Score: 0.992188
Score: 0.84375
Score: 0.996094
Score: 0.988281
Score: 0.992188
Score: 0.996094
Score: 0.996094
Score: 0.996094
Score: 0.949219
Score: 0.898438
Score: 0.996094
Score: 0.992188

INT8 x86
Score: 0.828264
Score: 0.999085
Score: 0.987032
Score: 0.993541
Score: 0.997998
Score: 0.999625
Score: 0.9995
Score: 0.93271
Score: 0.906202
Score: 0.999204
Score: 0.991705
Score: 0.828264
Score: 0.999085
Score: 0.987032
Score: 0.993541
Score: 0.997998
Score: 0.999625
Score: 0.9995
Score: 0.93271
Score: 0.906202
Score: 0.999204
Score: 0.991705
```

here are 2 more people with detectors that experience similar problem
https://github.com/VeriSilicon/tflite-vx-delegate/issues/35

I think we all have the same problem - Softmax is calculated over wrong axes.

sunshinemyson, 2022-01-19:
@bkovalenkocomp This have been fixed a02900d13519869456ba42c4b95330bf24c71e7f and unit test also get merged.

bkovalenkocomp, 2022-01-19:
> @bkovalenkocomp This have been fixed [a02900d](https://github.com/VeriSilicon/TIM-VX/commit/a02900d13519869456ba42c4b95330bf24c71e7f) and unit test also get merged.

thank you very much!

closed 2022-01-21 by sunshinemyson