VeriSilicon/TIM-VX #271: Matmul cant't get expected values when input is 1dx1d

ahqzy, 2022-01-17:
Hi,
   I'm testing matmul in timvx(opencl). It works abnormally when input dimension is 1dx1d. For example, I wrote this test cpp file:
-----------------------------------------------------------------------------------------------------------------------------------------------
#include "tim/vx/context.h"
#include "tim/vx/graph.h"
#include "tim/vx/ops/matmul.h"
#include "test_utils.h"
#include "gtest/gtest.h"

TEST(Matmul, shape_1_shape_1_float) {
    auto ctx = tim::vx::Context::Create();
    auto graph = ctx->CreateGraph();

    tim::vx::ShapeType a_shape({3});
    tim::vx::ShapeType b_shape({3});
    tim::vx::ShapeType out_shape({1});
    tim::vx::TensorSpec a_spec(tim::vx::DataType::FLOAT32,
                    a_shape, tim::vx::TensorAttribute::INPUT);
    tim::vx::TensorSpec b_spec(tim::vx::DataType::FLOAT32,
                    b_shape, tim::vx::TensorAttribute::INPUT);
    tim::vx::TensorSpec out_spec(tim::vx::DataType::FLOAT32,
                            out_shape, tim::vx::TensorAttribute::OUTPUT);

    auto a_tensor = graph->CreateTensor(a_spec);
    auto b_tensor = graph->CreateTensor(b_spec);
    auto out_tensor = graph->CreateTensor(out_spec);

    std::vector<float> a_data = {
        1, 2, 3
    };
		
    std::vector<float> b_data = {
		3, 3, 3
    };
    std::vector<float> golden = {
		18
    };

    EXPECT_TRUE(a_tensor->CopyDataToTensor(a_data.data(), a_data.size() * sizeof(float)));
    EXPECT_TRUE(b_tensor->CopyDataToTensor(b_data.data(), b_data.size() * sizeof(float)));

    auto op = graph->CreateOperation<tim::vx::ops::Matmul>();
    (*op).BindInputs({a_tensor, b_tensor}).BindOutputs({out_tensor});

    EXPECT_TRUE(graph->Compile());
    EXPECT_TRUE(graph->Run());

    std::vector<float> output(golden.size());
    EXPECT_TRUE(out_tensor->CopyDataFromTensor(output.data()));

	  for (size_t i = 0; i < golden.size(); ++i){
        std::cout<<"i: " <<i<<", golen: "<<golden[i]<<", output: "<<output[i]<<std::endl;
    }
   
	
    EXPECT_TRUE(ArraysMatch(golden, output, 1e-5f));
}
	
int main(int argc, char** argv)
	 
{
	 
		::testing::InitGoogleTest(&argc, argv);
		 
			return RUN_ALL_TESTS();
			 
}

-----------------------------------------------------------------------------------------------------------------------------------------------
After running,  I got errors like this:
---------------------------------------------------------------------------------
> ../test_utils.h:118: Failure
> The difference between expected[i] and actual[i] is 18, which exceeds abs_error, where
> expected[i] evaluates to 18,
> actual[i] evaluates to 0, and
> abs_error evaluates to 9.9999997473787516e-06.
> at index:0
> [  FAILED  ] Matmul.shape_1_shape_1_float (120 ms)
---------------------------------------------------------------------------------
So, what's the reason?

sunshinemyson, 2022-01-17:
@ahqzy ,

Thanks for your report, We will check it ASAP.

chxin66, 2022-01-19:
the shape is wrong, above is correct:
    tim::vx::ShapeType a_shape({3, 1});
    tim::vx::ShapeType b_shape({1, 3});
    tim::vx::ShapeType out_shape({1, 1});

ahqzy, 2022-01-19:
@chxin66 The above shape is 2dx2d，timvx doesn't support 1dx1d(or can't convert automatically)?  When I wrote it  in python, such as:
-------------------------------------------------------------
import numpy as np

a = np.random.randn(3).astype(np.float32)
b = np.random.randn(3).astype(np.float32)
c = np.matmul(a, b)

print('-----------------------------------------------')
print(a)

print('-----------------------------------------------')
print(b)

print('-----------------------------------------------')
print(c)
------------------------------------------------------------  

I can got this output correctly:
> -----------------------------------------------
> [-0.22248997 -2.0454798  -1.4306251 ]
> -----------------------------------------------
> [-0.6248956  -0.50848556 -0.42812803]
> -----------------------------------------------
> 1.7916206

So，in timvx,  when I input two 1d data， I must specify them to 2d ??

chxin66, 2022-01-19:
yes, you should write as 2d pattern. the case i made will push later

closed 2022-02-02 by sunshinemyson