VeriSilicon/TIM-VX #272: Whether include pad pixels when calculating AveragePool values for the edges

wdian, 2022-01-17:
When I was calculating averagePool, I noticed that the pad value sometimes computed, sometimes not computed.

[image]

As shown above, the result is 19 if the pad value is not computed, and 6.84 if the pad value is computed.

sunshinemyson, 2022-01-17:
Thanks for your feedback. 

What do you mean "sometimes"? Are you saying this is a random issue?

wdian, 2022-01-17:
Not random, I tested two models， The following two models, the first one calculates the pad value, the other does not.

[image]
[image]

wdian, 2022-04-12:
Is this problem solved?

wdian, 2022-05-24:
averagepool only supported ceil_mode=1 for onnx,  the default parameters ceil_mode=0  do not support.

closed 2022-09-28 by sunshinemyson