VeriSilicon/TIM-VX #277: Deconv2d result error when kernel_shape c>1

MESeraph, 2022-01-18:
Hi,
I add a test case in deconv2d_test.cc.
```c++
TEST(DeConv2d, shape_3_3_2_1_float) {
    auto ctx = tim::vx::Context::Create();
    auto graph = ctx->CreateGraph();

    tim::vx::ShapeType input_shape ({3, 3, 1, 1});  //whcn
    tim::vx::ShapeType kernel_shape({3, 3, 2, 1});  //whc1 same as depthwise convolution
    tim::vx::ShapeType output_shape({5, 5, 2, 1});  //whcn

    tim::vx::TensorSpec input_spec  (tim::vx::DataType::FLOAT32, input_shape, tim::vx::TensorAttribute::INPUT);
    tim::vx::TensorSpec kernel_spec (tim::vx::DataType::FLOAT32, kernel_shape, tim::vx::TensorAttribute::CONSTANT);
    tim::vx::TensorSpec output_spec (tim::vx::DataType::FLOAT32, output_shape, tim::vx::TensorAttribute::OUTPUT);

    auto input_tensor = graph->CreateTensor(input_spec);
    auto output_tensor = graph->CreateTensor(output_spec);
    auto kernel_tensor = graph->CreateTensor(kernel_spec);

    std::vector<float> input_data = {
        3.0f, 8.0f, 1.0f, 9.0f, 5.0f, 7.0f, 3.0f, 2.0f, 3.0f,
    };

    std::vector<float> kernel_data = {
        9.0f, 0.0f, 3.0f, 0.0f, 0.0f, 0.0f, 1.0f, 0.0f, 2.0f,
        9.0f, 0.0f, 3.0f, 0.0f, 0.0f, 0.0f, 1.0f, 0.0f, 2.0f,
    };

    std::vector<float> output_data(element_count(output_shape));

    EXPECT_TRUE(input_tensor->CopyDataToTensor(input_data.data(), input_data.size()*4));
    EXPECT_TRUE(kernel_tensor->CopyDataToTensor(kernel_data.data(), kernel_data.size()*4));

    auto op = graph->CreateOperation<tim::vx::ops::DeConv2d>(
        2,
        tim::vx::PadType::SAME,
        std::array<uint32_t, 2>({3, 3}),    /*ksize*/
        std::array<uint32_t, 2>({1, 1}),    /*stride*/
        std::array<uint32_t, 2>({1, 1}),    /*dilation*/
        std::array<uint32_t, 4>({0, 0, 0, 0}), /*pad*/
        1/*group*/);
    (*op).BindInputs({input_tensor, kernel_tensor}).BindOutputs({output_tensor});

    EXPECT_TRUE(graph->Compile());
    EXPECT_TRUE(graph->Run());

    EXPECT_TRUE(output_tensor->CopyDataFromTensor(output_data.data()));
    std::vector<float> golden = {
        27.0f, 72.0f, 18.0f, 24.0f, 3.0f,  81.0f, 45.0f, 90.0f, 15.0f,
        21.0f, 30.0f, 26.0f, 43.0f, 22.0f, 11.0f, 9.0f,  5.0f,  25.0f,
        10.0f, 14.0f, 3.0f,  2.0f,  9.0f,  4.0f,  6.0f,
        27.0f, 72.0f, 18.0f, 24.0f, 3.0f,  81.0f, 45.0f, 90.0f, 15.0f,
        21.0f, 30.0f, 26.0f, 43.0f, 22.0f, 11.0f, 9.0f,  5.0f,  25.0f,
        10.0f, 14.0f, 3.0f,  2.0f,  9.0f,  4.0f,  6.0f,
    };

    EXPECT_EQ(golden, output_data) << "Result mismatch";
}

```

unit_test result:
```c++
shuliu@shuliu-ThinkPad-E14-Gen-2:~/Downloads/TIM-VX/host_build/src/tim$ ./unit_test --gtest_filter="DeConv2d.shape_3_3_2_1_float"
Running main() from /home/shuliu/Downloads/googletest/googletest/src/gtest_main.cc
Note: Google Test filter = DeConv2d.shape_3_3_2_1_float
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from DeConv2d
[ RUN      ] DeConv2d.shape_3_3_2_1_float
#productname=VSI SIMULATOR, pid=0xc5
I [vsi_nn_CreateGraph:562]OVXLIB_VERSION==1.1.37
graph gpuCount=1 interConnectRingCount=0
NN ring buffer is disabled
D [vsi_nn_SetupGraph:702]Sort graph nodes.
D [setup_node:466]Setup node id[0] uid[1] op[DECONVOLUTION]
D [print_tensor:155]in(0) : id[   0] vtl[0] const[0] shape[ 3, 3, 1, 1       ] fmt[f32] qnt[NONE]
D [print_tensor:155]in(1) : id[   2] vtl[0] const[1] shape[ 3, 3, 2, 1       ] fmt[f32] qnt[NONE]
D [print_tensor:104]in(2) : None
D [print_tensor:155]out(0): id[   1] vtl[0] const[0] shape[ 5, 5, 2, 1       ] fmt[f32] qnt[NONE]
D [optimize_node:410]Backward optimize neural network
D [optimize_node:417]Forward optimize neural network
I [compute_node:346]Create vx node
D [compute_node:375]Instance node[0] "DECONVOLUTION" ...
---------------------------Begin VerifyTiling -------------------------
AXI-SRAM = 0 Bytes VIP-SRAM = 259584 Bytes SWTILING_PHASE_FEATURES[1, 1, 1]
  0 SH [(   3    3    1 1,       36, 0x0x55b755539a70(0x0x55b755539a70, 0x(nil)) ->    9    5    5 1,      900, 0x0x55b75553fdb0(0x0x55b75553fdb0, 0x(nil))) k(0 0    0,        0) pad(0 0) pool(0 0, 1 1)] C[  1]
  1 SH [(   9    5    5 1,      900, 0x0x55b75553fdb0(0x0x55b75553fdb0, 0x(nil)) ->    5    5    2 1,      200, 0x0x55b75553a990(0x0x55b75553a990, 0x(nil))) k(0 0    0,        0) pad(0 0) pool(0 0, 1 1)] P[  0]

 id IN [ x  y  w   h ]   OUT  [ x  y  w  h ] (tx, ty, kpc) (ic, kc, kc/ks, ks/eks, kernel_type) NNT(in, out)

 id | opid IN [ x  y  w   h ]   OUT  [ x  y  w  h ] (tx, ty, kpc) (ic, kc, kc/ks, ks/eks, kernel_type) NNT(in, out)
  0 |   0 SH DD 0x0 [   0    0        0        0] -> DD 0x0 [   0    0        0        0] (  0,   0,   0) (       0,        0, 0.00%, 0.00%, NONE) (       0,        0)
  1 |   1 SH DD 0x0 [   0    0        0        0] -> DD 0x0 [   0    0        0        0] (  0,   0,   0) (       0,        0, 0.00%, 0.00%, NONE) (       0,        0)

PreLoadWeightBiases = 0  -nan%
---------------------------End VerifyTiling -------------------------
D [_check_swapped_tensors:112]Check swapped tensors
***********dump layer : 0***************
***********dump layer : 1***************
/home/shuliu/Downloads/TIM-VX/src/tim/vx/ops/deconv2d_test.cc:195: Failure
Expected equality of these values:
  golden
    Which is: { 27, 72, 18, 24, 3, 81, 45, 90, 15, 21, 30, 26, 43, 22, 11, 9, 5, 25, 10, 14, 3, 2, 9, 4, 6, 27, 72, 18, 24, 3, 81, 45, ... }
  output_data
    Which is: { 27, 72, 18, 24, 3, 81, 45, 90, 15, 21, 30, 26, 43, 22, 11, 9, 5, 25, 10, 14, 3, 2, 9, 4, 6, 0, 0, 0, 0, 0, 0, 0, ... }
Result mismatch
[  FAILED  ] DeConv2d.shape_3_3_2_1_float (978 ms)
[----------] 1 test from DeConv2d (979 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (979 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] DeConv2d.shape_3_3_2_1_float

 1 FAILED TEST
```

sunshinemyson, 2022-01-18:
@MESeraph ,

Thanks for your feedback. We will check this ASAP.

sunshinemyson, 2022-01-19:
@MESeraph ，

We have found the root cause of this internally, it will require sdk update. I'll put it to our quarterly release plan (End of 2022 Q1).

MESeraph, 2022-01-24:
@sunshinemyson What does sdk refer to? Does it refer to libopenvx.so?

sunshinemyson, 2022-01-25:
@MESeraph ,

Yes.

closed 2022-05-07 by sunshinemyson