VeriSilicon/TIM-VX #279: Does timvx support MatmMulInteger?

ahqzy, 2022-01-20:
Hi,
   I'm testing MatMulInteger in Timvx， I meet some tough problems：
   (1) Timvx doesn't support format U8/U8/I32
   (2 )If I modify the check in vsi_nn_op_matmul.c，add this:
>          IO_TYPE(D_U8|Q_ASYM,  D_U8|Q_ASYM,  D_I32|Q_ASYM) 

      When running, I got errors like this:
>  W [vsi_nn_kernel_selector:1029]No valid kernel for matrixmul
> E [/local_ws_nobackup2/zqiu/TIM-VX-main/src/tim/vx/internal/src/vsi_nn_graph.c:compute_node:379]Create node[0] MATRIXMUL fail
> matmul_test.cc:354: Failure
> Value of: graph->Compile()
>   Actual: false
> Expected: true
> matmul_test.cc:355: Failure
> Value of: graph->Run()
>   Actual: false
> Expected: true
> free(): invalid pointer
> Aborted (core dumped)

So, what can I do?

Below is the whole test file:
------------------------------------------------------------------------------------------------------------
#include "tim/vx/context.h"
#include "tim/vx/graph.h"
#include "tim/vx/ops/matmul.h"
#include "test_utils.h"
#include "gtest/gtest.h"

TEST(Matmul, shape___2_3_2_shape_2_3_2_uint8_transpose_a) {
    auto ctx = tim::vx::Context::Create();
    auto graph = ctx->CreateGraph();

    tim::vx::ShapeType a_shape({2, 3, 2});
    tim::vx::ShapeType b_shape({2, 3, 2});
    tim::vx::ShapeType out_shape({2, 2, 2});

    tim::vx::Quantization input_quant(tim::vx::QuantType::ASYMMETRIC, 1, 6);
    //tim::vx::Quantization output_quant(tim::vx::QuantType::ASYMMETRIC, 1, 0);

    tim::vx::TensorSpec a_spec(tim::vx::DataType::UINT8,
                    a_shape, tim::vx::TensorAttribute::INPUT, input_quant);

    tim::vx::TensorSpec b_spec(tim::vx::DataType::UINT8,
                    b_shape, tim::vx::TensorAttribute::INPUT, input_quant);

    tim::vx::TensorSpec out_spec(tim::vx::DataType::INT32,
                    out_shape, tim::vx::TensorAttribute::OUTPUT);

    auto a_tensor = graph->CreateTensor(a_spec);
    auto b_tensor = graph->CreateTensor(b_spec);
    auto out_tensor = graph->CreateTensor(out_spec);

    std::vector<uint8_t> a_data = {
         7,  8,
         9, 10,
        11, 12,
         5,  4,
         3,  2,
         1,  0,
    };
    std::vector<uint8_t> b_data = {
        12, 11,
        10,  9,
         8,  7,
         0,  1,
         2,  3,
         4,  5,
    };
    std::vector<uint8_t> golden = {
        28, 19,
        40, 28,
        28, 19,
        40, 28,
    };

    EXPECT_TRUE(a_tensor->CopyDataToTensor(a_data.data(), a_data.size()));
    EXPECT_TRUE(b_tensor->CopyDataToTensor(b_data.data(), b_data.size()));

    auto op = graph->CreateOperation<tim::vx::ops::Matmul>(true);
    (*op).BindInputs({a_tensor, b_tensor}).BindOutputs({out_tensor});

    EXPECT_TRUE(graph->Compile());
    EXPECT_TRUE(graph->Run());

    std::vector<uint8_t> output(golden.size());
    EXPECT_TRUE(out_tensor->CopyDataFromTensor(output.data()));
    EXPECT_EQ(golden, output);
}

int main(int argc, char** argv)
         
{
         
                ::testing::InitGoogleTest(&argc, argv);
                 
                        return RUN_ALL_TESTS();
                         
}
--------------------------------------------------------------------------------------------------------------

sunshinemyson, 2022-01-25:
@ahqzy ，

Thanks for your feedback, We will check it ASAP.

ZongwuYang, 2022-01-26:
@ahqzy ,

We have checked that our hardware does not support u8 * u8 -> int32 temporarily, but we have planed to support the it in our new hardware.

ahqzy, 2022-01-26:
ok, 3ks.

closed 2022-02-02 by sunshinemyson