VeriSilicon/TIM-VX #283: Why is layout of  weight_shape 'whio'  in  depthwiseConv_test.cc ?

MESeraph, 2022-01-28:
I can not understand the note of weight_shape in https://github.com/VeriSilicon/TIM-VX/blob/main/src/tim/vx/ops/depthwiseConv_test.cc.
Why is layout of  weight_shape 'whoi'?
 
Another, depthwise convolution is a special case of grouped convolution. 
Shouldn't it be calling groupedconv2d? 
I can not understand the test data in depthwiseConv_test.cc.
It is not a correct depthwise input / output.

sunshinemyson, 2022-02-02:
Hi MESeraph,

Layout is Filter_Weight, Filter_Height, Out_Channel, In_Channel, and In_Channel should always set to 1. You can take android nnapi  as a reference.
https://developer.android.google.cn/ndk/reference/group/neural-networks?hl=zh-cn 

Do you mean "shape_2_3_2_1_float32_SimpleTest" is not a correct depthwise convolution? This test case have multiplier = 2. Which is correct configuration for depthwise conv.

Depthwise is special case of group conv, since we have special speed-up for depthwise, we treat depthwise different way with general conv2d.

MESeraph, 2022-02-14:
@sunshinemyson  I got it. Thank you.

closed 2022-02-17 by sunshinemyson