VeriSilicon/TIM-VX #309: How can I get the intermediate output of a model?

ahqzy, 2022-02-28:
Hi 
    I'm debuging some onnx model, I want to compare the result between cpu and gpu., so I need get all intermediate output. I can get them when in cpu mode like this article: https://blog.csdn.net/Artyze/article/details/116270215.
    But In the gpu mode, I only get the last reslut, can't get any intermediate output. why?

sunshinemyson, 2022-03-01:
@ahqzy ,

We don't provide this capability from TIM-VX level. I suggest you can implement this feature from ONNXRT side by:

If debugging per layer dump enabled:
    create Tim-vx tensor with attribute = OUTPUT instead of TRANSIENT

Then you can dump tensor with CopyDataFromTensor()

ahqzy, 2022-03-02:
@sunshinemyson  Sorry, maybe I didn't express my requirement clearly. For exampe, I have a model as below:

[image]

How can I get all intermediate node(MaxPool、Conv...... Relu) output when the model has been completed？

sunshinemyson, 2022-03-03:
How did you create tensor for those intermediate nodes? what kinds of attribute?

Only tensor with In/output as its attribute can be read. You need to change the model construction code.

xuke537, 2022-03-07:
@ahqzy  You can append the intermediate outputs to the onnx graph outputs to check the results.
eg.
import onnx
from onnx import helper
model = onnx.load('xx.onnx')

intermediate_tensor_name = 'xxx'
model_path = 'xx_new.onnx'
intermediate_layer_value_info = helper.ValueInfoProto()
intermediate_layer_value_info.name = intermediate_tensor_name
model.graph.output.append(intermediate_layer_value_info)
onnx.save(model, model_path)

ahqzy, 2022-03-08:
@xuke537  Using that method, I can get  intermediate output for some models. But I can't get  intermediate output for another models(such as above picture).  I guess the reason is that VSI optimize some operator(for exampe, conv+bn)，so we can't get the intermediate output for conv. Do you have idea about it(or is there any method to disable the optimize for debug)?

xuke537, 2022-03-08:
For debug， you can cut your model.

closed 2022-09-28 by sunshinemyson