VeriSilicon/TIM-VX #317: Sota pose net needing!

2050airobert, 2022-03-10:
hi,
1 Any plan of the lite-hrnet coverting to vim3 a311d NPU?
2 How about the sota of pose model just like tinypose ? Could we covert tinypose to the vim3 NPU?

Tks

sunshinemyson, 2022-03-10:
if you have tflite model, you don't have to convert it. We have a vx-delegate for tflite can handle this.

2050airobert, 2022-03-10:
Any restrict requirement of the tflite？ For example， I will translate the third party model to onnx model, then convert the onnx model to the tensorflow model , then convert the tf model to the tflite model.  Any lose or inaccuracy occurs in this coverting process？ And most of all， we can  handle this tflite mode also?

2050airobert, 2022-03-16:
Tks, could you share any webchat group like other company which make us communicate conveniently？ 
As we are working on the a311d ，we want to share or know more similar problem with the TIM-VX framework on the chip.
Looking forward to your reply!

sunshinemyson, 2022-03-22:
@2050airobert ,

we don't have it right now. It will take some time because company policy. Before that, you can email to our support email.

ML_Support@verisilicon.com

closed 2022-09-28 by sunshinemyson