VeriSilicon/TIM-VX #318: max_pool compute error

MESeraph, 2022-03-10:
Hi,
I add a maxpool test case:
```cpp
#include "tim/vx/context.h"
#include "tim/vx/graph.h"
#include "tim/vx/ops/pool2d.h"
#include <iostream>
#include "gtest/gtest.h"
#include "test_utils.h"

TEST(MAX, shape_6_6_1_1_fp32_kernel_3_stride_2) {
    auto ctx = tim::vx::Context::Create();
    auto graph = ctx->CreateGraph();

    tim::vx::ShapeType in_shape({6, 6, 1, 1});
    tim::vx::ShapeType out_shape({3, 3, 1, 1});
    tim::vx::TensorSpec input_spec(tim::vx::DataType::FLOAT32,
                            in_shape, tim::vx::TensorAttribute::INPUT);
    tim::vx::TensorSpec output_spec(tim::vx::DataType::FLOAT32,
                            out_shape, tim::vx::TensorAttribute::OUTPUT);

    auto input_tensor = graph->CreateTensor(input_spec);
    auto output_tensor = graph->CreateTensor(output_spec);

    std::vector<float> in_data = {
        -1, -2, -3, -4, -5, -5,
        -6, -7, -8, -9, -10, -10,
        -11, -12, -13, -14, -15, -15,
        -16, -17, -18, -19, -20, -20,
        -21, -22, -23, -24, -25, -20, 
        -21, -22, -23, -24, -25, -20
        };
    std::vector<float> golden = {
        -1, -3, -5,
        -11, -13, -15,
        -21, -23, -20
        };

    EXPECT_TRUE(input_tensor->CopyDataToTensor(in_data.data(), in_data.size()*4));
    std::array<uint32_t, 2> ksize = {3, 3};
    std::array<uint32_t, 2> stride = {2, 2};
    auto round_type = tim::vx::RoundType::CEILING;
    auto op = graph->CreateOperation<tim::vx::ops::Pool2d>(tim::vx::PoolType::MAX,
        tim::vx::PadType::AUTO, ksize, stride, round_type);
    (*op).BindInputs({input_tensor}).BindOutputs({output_tensor});

    EXPECT_TRUE(graph->Compile());
    EXPECT_TRUE(graph->Run());

    std::vector<float> output(golden.size());
    EXPECT_TRUE(output_tensor->CopyDataFromTensor(output.data()));
    EXPECT_EQ(golden, output);
}

```
but the result  mismatch: 
```sh
Running main() from /home/ubuntu/github/googletest/googletest/src/gtest_main.cc
Note: Google Test filter = MAX.*
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from MAX
[ RUN      ] MAX.shape_6_6_1_1_fp32_kernel_3_stride_2
/home/ubuntu/github/TIM-VX/src/tim/vx/ops/max_pool_test.cc:72: Failure
Expected equality of these values:
  golden
    Which is: { -1, -3, -5, -11, -13, -15, -21, -23, -20 }
  output
    Which is: { -1, -3, -5, -11, -13, -15, -21, -23, 0 }
[  FAILED  ] MAX.shape_6_6_1_1_fp32_kernel_3_stride_2 (687 ms)
[----------] 1 test from MAX (687 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (687 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] MAX.shape_6_6_1_1_fp32_kernel_3_stride_2

 1 FAILED TEST
```

sunshinemyson, 2022-03-15:
@MESeraph ,

Thanks for feedback. We will check it.

sunshinemyson, 2022-03-22:
@MESeraph ,

SAME pad is not valid for MAXPool, we're going to refine the API. You can provide pad with concrete size explicitly. It can work. Thanks

MESeraph, 2022-03-25:
@sunshinemyson 
I modify test by provide pad with concrete size explicity:
```cpp
EXPECT_TRUE(input_tensor->CopyDataToTensor(in_data.data(), in_data.size()*4));
    std::array<uint32_t, 2> ksize = {3, 3};
    std::array<uint32_t, 2> stride = {2, 2};
    auto round_type = tim::vx::RoundType::CEILING;
    std::array<uint32_t, 4> padd_array = {0, 0, 0, 0};

    auto op = graph->CreateOperation<tim::vx::ops::Pool2d>(tim::vx::PoolType::MAX,
        padd_array, ksize, stride, round_type);
    (*op).BindInputs({input_tensor}).BindOutputs({output_tensor});
```
but the result still mismatch:
```sh
Running main() from /home/ubuntu/github/googletest/googletest/src/gtest_main.cc
Note: Google Test filter = MAX.*
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from MAX
[ RUN      ] MAX.shape_6_6_1_1_fp32_kernel_3_stride_2
/home/ubuntu/github/TIM-VX/src/tim/vx/ops/max_pool_test.cc:74: Failure
Expected equality of these values:
  golden
    Which is: { -1, -3, -5, -11, -13, -15, -21, -23, -20 }
  output
    Which is: { -1, -3, -5, -11, -13, -15, -21, -23, 0 }
[  FAILED  ] MAX.shape_6_6_1_1_fp32_kernel_3_stride_2 (699 ms)
[----------] 1 test from MAX (699 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (700 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] MAX.shape_6_6_1_1_fp32_kernel_3_stride_2

 1 FAILED TEST
```
I think that the reason for its mismatch may be the same as issue #272

sunshinemyson, 2022-03-28:
@MESeraph , 
Not exactly same with #272, in 272, padding value provided explicitly. In your case, we just not support SAME for pad_type, because for SAME, we can not calculate correct pad value internally.

sunshinemyson, 2022-04-15:
fixed in latest v1.1.42

closed 2022-04-15 by sunshinemyson