VeriSilicon/TIM-VX #324: The compilation error: variable-sized object may not be initialized

hong19860320, 2022-03-19:
When I compile this library with android-ndk-r20b for android armeabi-v7a, the following error occurs:
`
tim-vx/src/tim/vx/ops/custom_base.cc:165:44: error: variable-sized object may not be initialized
    vsi_nn_kernel_node_param_t node_params[param_num] = {NULL};
` 

So I found the following solution:
`
    vsi_nn_kernel_node_param_t node_params[param_num];
    memset(node_params, 0, size(vsi_nn_kernel_node_param_t) * param_num);
`

sunshinemyson, 2022-03-22:
@hong19860320 ,

Thanks for your feedback. I think it because clang have different warning rule compare to GCC. We will check if we can replace it with memset or not - maybe can do it in another approach.

Thanks

sunshinemyson, 2022-03-22:
Fixed by 08500158ba96ce1a6f0f4e2ac6fc289fca4f3c11

closed 2022-03-22 by sunshinemyson