VeriSilicon/TIM-VX #330: Test failed in ArgMax

ahqzy, 2022-03-23:
Hi，
   I cound't find a unit test about ArgMax in TimVX code, I wrote it, but test faild. The source is:

`#include "tim/vx/context.h"
#include "tim/vx/graph.h"
#include "tim/vx/ops/arg.h"
#include "test_utils.h"
#include "gtest/gtest.h"

TEST(ArgMax, one_input) {
    auto ctx = tim::vx::Context::Create();
    auto graph = ctx->CreateGraph();

    tim::vx::ShapeType a_shape({2, 2});
    tim::vx::ShapeType out_shape({2});

    tim::vx::TensorSpec a_spec(tim::vx::DataType::FLOAT32,
                    a_shape, tim::vx::TensorAttribute::INPUT);

    tim::vx::TensorSpec out_spec(tim::vx::DataType::INT32,
                            out_shape, tim::vx::TensorAttribute::OUTPUT);

    auto a_tensor = graph->CreateTensor(a_spec);
    auto out_tensor = graph->CreateTensor(out_spec);

    std::vector<float> a_data = {
        2, 1,
        3, 10
    };

    std::vector<float> golden = {
        0, 1
    };

    EXPECT_TRUE(a_tensor->CopyDataToTensor(a_data.data(), a_data.size() * sizeof(float)));

    auto op = graph->CreateOperation<tim::vx::ops::ArgMax>(1);
    (*op).BindInputs({a_tensor}).BindOutputs({out_tensor});

    EXPECT_TRUE(graph->Compile());
    EXPECT_TRUE(graph->Run());

    std::vector<float> output(golden.size());
    EXPECT_TRUE(out_tensor->CopyDataFromTensor(output.data()));
    EXPECT_TRUE(ArraysMatch(golden, output, 1e-5f));
}

int main(int argc, char** argv)
	 
{
	 
		::testing::InitGoogleTest(&argc, argv);
		 
			return RUN_ALL_TESTS();
			 
}`

The result after running is:
----
[image]
------------------------------------------------------------
What's the problem?

sunshinemyson, 2022-03-24:
@ahqzy ,

Thanks for report. Could you just create PR with your code. Then once we fixed it in case or implementation, we can merge it.

Thanks

ahqzy, 2022-03-24:
@sunshinemyson  Pull Request has been submitted, please handle it. Thank you.

sunshinemyson, 2022-03-25:
@ahqzy 

Thanks to @xuke537 , he refined your case and get it merged.

closed 2022-03-25 by sunshinemyson