VeriSilicon/TIM-VX #344: will support python interface?

zjd1988, 2022-04-06:


sunshinemyson, 2022-04-06:
Hi @zjd1988 , 

So far, no plan to support python. Could you share more background why you interested with python interface? I can promote your information to the team. Thanks

zjd1988, 2022-04-06:
hi @sunshinemyson, thanks for your reply. 
I have a rv1109 borad, and converted onnx to rknn model by RK tools. In order to test the rknn model's accuracy with a test dataset , I use rk's x64 simulator to infer model, but it takes too long to infer a image. 
I know RK use VeriSilicon‘s npu ip, So I am trying to convert rknn model to tim-vx model to simulate.
Currently,I can parse the rknn model with layer's detail info( weight, scale, zero point),
and  next trying to use the tim-vx's python api( not supported yet) to construct model

sunshinemyson, 2022-04-06:
@zjd1988 ,

Sorry that RV1109 is not actively supported from our side, even if you can overcome the conversion issue, their driver stack is not verified with TIM-VX.  I would like to recommend VIM3/VIM3L dev-kit to you, they are active and verified by us.

The vx-delegate is preferred for inference on board. Hope you can enjoy it. Thanks.

zjd1988, 2022-05-08:
@sunshinemyson  I have implemented a simple python binding for timvx，verified with lenet example
https://github.com/zjd1988/TIM-VX-python/tree/main/src/pytim

sunshinemyson, 2022-05-09:
@zjd1988,

Great job, Thanks for sharing. Would you like merge it to tim-vx?

zjd1988, 2022-05-09:
@sunshinemyson I‘m not sure’whether the code quality is good enough to merge. You may check it out first.

zjd1988, 2022-05-09:
Hi @sunshinemyson, I have a simple question, weights tensor's data layout is NCHW or WHCN in lenet_asymu8_weights.h?

zjd1988, 2022-05-10:
Hi @sunshinemyson,  Is it convenient to communication with wechat (xd-zjd)

sunshinemyson, 2022-05-10:
> Hi @sunshinemyson, I have a simple question, weights tensor's data layout is NCHW or WHCN in lenet_asymu8_weights.h?

For convolution's kernel, data layout in memory is Oc,Ic,H,W, In VX, it descripted in shape as C_Array[W,H,Ic,Oc] - column major description.

fengwang, 2022-05-10:
> > Hi @sunshinemyson, I have a simple question, weights tensor's data layout is NCHW or WHCN in lenet_asymu8_weights.h?
> 
> For convolution's kernel, data layout in memory is Oc,Ic,H,W, In VX, it descripted in shape as C_Array[W,H,Ic,Oc] - column major description.

@sunshinemyson  Does it necessarily mean all the data layout description in tim-vx is column major? Or have you documented this somewhere I missed out?

sunshinemyson, 2022-06-02:
@fengwang ,

Sorry for late. https://github.com/VeriSilicon/TIM-VX/blob/main/docs/Programming_Guide.md#memory-layout
Here is the doc about memory layout.