VeriSilicon/TIM-VX #346: depthwiseconv weights layout whoi  or whoi?

MESeraph, 2022-04-06:
I try modify a unit test case of the depthwiseConv_test.cc as follows:
[image]
I just modify weight_shape from 'whoi' from 'whio'.
But the unit test case can run successfully.
Don't get me wrong. ``I really want this effect.``
Because other model weight tensor shape is 'whio'.
I just want to make sure: 
``Is this an optimization ?  I hope no matter how weight_ shape is 'whio' or 'whoi', it can run correctly.``
Please confirm this.This is very important to us.
Thank You!

sunshinemyson, 2022-04-08:
For depthwise , Out channel always equal to input channel. When we say whoi for depthwise conv, it is actually WHO1. Then WH1O doesn't much different than WHO1. 

You can check android nnapi's definition as reference https://developer.android.google.cn/ndk/reference/group/neural-networks?hl=zh-cn.

MESeraph, 2022-04-19:
@sunshinemyson I got. Thank you!

closed 2022-04-19 by MESeraph