VeriSilicon/TIM-VX #368: max_pool compute error

MESeraph, 2022-04-19:
I verified your fix for issue #318. My unit test and relevant model is still wrong.
My unit is modify in [max_pool_test file](https://github.com/VeriSilicon/TIM-VX/blob/main/src/tim/vx/ops/max_pool_test.cc)
pad:[0,0,0,0]--->[0,1,0,1]
In fact, the pad of my maxpool OP are [0, 0, 0, 0].
The maxpool operator of my onnx mobilenetV1 model is as follows:
[image]
Looking forward to your reply, Thank you!

xuke537, 2022-04-22:
@MESeraph 

Please check apply the patch for your unit test 
diff --git a/src/tim/vx/ops/max_pool_test.cc b/src/tim/vx/ops/max_pool_test.cc
index c245414..691a6cb 100644
--- a/src/tim/vx/ops/max_pool_test.cc
+++ b/src/tim/vx/ops/max_pool_test.cc
@@ -35,12 +35,13 @@ TEST(MAX, shape_6_6_1_1_fp32_kernel_3_stride_2) {

   EXPECT_TRUE(
       input_tensor->CopyDataToTensor(in_data.data(), in_data.size() * 4));
-  std::array<uint32_t, 4> pad = {0, 1, 0, 1};
+  //std::array<uint32_t, 4> pad = {0, 1, 0, 1};
   std::array<uint32_t, 2> ksize = {3, 3};
   std::array<uint32_t, 2> stride = {2, 2};
   auto round_type = tim::vx::RoundType::CEILING;
+  auto  pad_type = tim::vx::PadType::SAME;
   auto op = graph->CreateOperation<tim::vx::ops::Pool2d>(
-      tim::vx::PoolType::MAX, pad, ksize, stride, round_type);
+      tim::vx::PoolType::MAX, pad_type, ksize, stride, round_type);
   (*op).BindInputs({input_tensor}).BindOutputs({output_tensor});

   EXPECT_TRUE(graph->Compile());

MESeraph, 2022-04-26:
Thank you! It work successfully by RoundType::CEILING and PadType::SAME.
But it still compute error when i run my onnx model.
I try to write a actual data maxpool UT from my onnx model.

closed 2022-05-23 by sunshinemyson