VeriSilicon/TIM-VX #369: What is the correct tensor attribute corresponding to the layer output that will be used more than once?

fengwang, 2022-04-19:
For a model such as

```
y1 = L0(x)
y2 = L1(y1)
y3 = L2(y1)
y = add( y2, y3)
```

in which `x` is the input tensor and `y` is the output tensor. In the forward propagation (inference) time, the  intermediate output tensor `y1` from layer `L0` will be used twice by layers `L1` and `L2`.
In this case, how to correctly setup the tensor [attribute](https://github.com/VeriSilicon/TIM-VX/blob/main/include/tim/vx/types.h#L45) for the output tensor `y1`?

fengwang, 2022-04-21:
For the tensor `x`, I should set its attribute to ` tim::vx::TensorAttribute::INPUT`, and for the tensor `y` I should set its attribute to  `tim::vx::TensorAttribute::OUTPUT`
But for the intermediate tensor `y1`, which will be used in two branches L1 and L2, can I set its attribute to `tim::vx::TensorAttribute::TRANSIENT`? Or should it be  `tim::vx::TensorAttribute::VARIABLE`?

Given the fact that residual connections are popular in network architectures, such as U-Net and Resnet, I think it is important to clarify which attribute to use for the intermediate tensors that will be used more than once.

sunshinemyson, 2022-04-21:
@fengwang ,

For Y1, recommend tim::vx::TensorAttribute::TRANSIENT for any intermediate output tensor.

fengwang, 2022-04-22:
@sunshinemyson 

Does this necessarily mean that, during the inference time, all the tensors with attribute `TRANSIENT` are well kept and there is no possibility of memory overlapping/reuse for them?

fengwang, 2022-04-22:
@sunshinemyson 

For a intermediate tensor `y_n`, if there is only one activation layer such as Relu or Tanh behind it, we will have

```
y_n1 = Relu( y_n );
```

and `y_n` will not be cited elsewhere.

Is it ture that the tensor stored at `y_n` are copy/modified to new memory hold by `y_n1`, rather than applying the `relu` activation inplace at `y_n` and then pointing `y_n1` to `y_n`?

sunshinemyson, 2022-04-22:
yes, your understanding is correct. However, we have global memory optimization inside to reuse those intermediate buffer.

closed 2022-04-22 by fengwang