VeriSilicon/TIM-VX #378: run custom_op_test got "segmentation fault" when add two params in ParamTuple

zhnin, 2022-04-29:
**Describe**
sample path: `MACA-VX/samples/custom_op_test` 

Run `./custom_op_test` got **"Segmentation fault (core dumped)"**, when I add **two params** in ParamTuple

[image]

**Steps/Code to reproduce**
1. Modified two places in `custom_gemm.h`, `ParamTuple` and `kernel_resource`
[image]
2. Modified `tuple_list` in `custom_op_test.cc`
[image]

**Expected behavior**
1. When I add one param, it works well. whether have some rule about param nums?

sunshinemyson, 2022-04-29:
@zhnin 

Thanks for report. 
Please share your patch  and base version?  It seems you didn't update kernel source code with same parameters - https://github.com/VeriSilicon/TIM-VX/blob/main/docs/customized_op.md  please check this doc about the parameter definition.

zhnin, 2022-04-29:
> @zhnin
> 
> Thanks for report. Please share your patch and base version? It seems you didn't update kernel source code with same parameters - https://github.com/VeriSilicon/TIM-VX/blob/main/docs/customized_op.md please check this doc about the parameter definition.

Hi: @sunshinemyson , kernel source code also added in opencl input args. the patch and version is
- branch: main
- commitid: [d108661](https://github.com/VeriSilicon/TIM-VX/commit/d108661a03a12a6ada53ad8cbf4d2686c257c9e5) (latest)

**git patch** 
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index a266d0b..e866cb1 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -2,17 +2,17 @@ cmake_minimum_required (VERSION 3.14)
 project(tim-vx LANGUAGES C CXX)
 
 option(BUILD_SHARED_LIBS                "Build using shared libraries"          ON)
-option(TIM_VX_ENABLE_CUSTOM_OP          "Enable custom op support"              OFF)
+option(TIM_VX_ENABLE_CUSTOM_OP          "Enable custom op support"              ON)
 option(TIM_VX_ENABLE_TEST               "Build the unit test"                   OFF)
 option(TIM_VX_ENABLE_LAYOUT_INFER       "Enable layout inference support"       ON)
 option(TIM_VX_ENABLE_NBG_PARSER         "Enable NBG parser"                     OFF)
 option(TIM_VX_CODE_COVERAGE             "Run code coverage with gconv(gcc only" OFF)
 option(TIM_VX_USE_EXTERNAL_OVXLIB       "Use external OVXLIB"                   OFF)
-option(TIM_VX_BUILD_EXAMPLES            "Build demos show general usage"        OFF)
+option(TIM_VX_BUILD_EXAMPLES            "Build demos show general usage"        ON)
 option(TIM_VX_ENABLE_VIPLITE            "Enable lite driver api support"        OFF)
 option(TIM_VX_ENABLE_40BIT              "Enable large memory support"           OFF)
 option(TIM_VX_ENABLE_PLATFORM           "Enable multi devices support"          OFF)
-option(TIM_VX_ENABLE_CUSTOM_OP          "Enable customized op example"          OFF)
+option(TIM_VX_ENABLE_CUSTOM_OP          "Enable customized op example"          ON)
 
 set(CMAKE_CXX_STANDARD 14)
 set(CMAKE_POSITION_INDEPENDENT_CODE ON)
diff --git a/samples/custom_op_test/custom_gemm.h b/samples/custom_op_test/custom_gemm.h
index 5fbb8f8..e12fa51 100644
--- a/samples/custom_op_test/custom_gemm.h
+++ b/samples/custom_op_test/custom_gemm.h
@@ -39,6 +39,8 @@ class CustomGemm : public CustomOpBase {
   using ParamTuple = std::tuple<int,   /* M */
                                       int,   /* K */
                                       int,   /* N */
+                                      float, /* test1 */
+                                      float, /* test2 */
                                       int,   /* ac2zero */
                                       int,   /* bc2zero */
                                       float, /* scale_a */
@@ -66,6 +68,8 @@ class CustomGemm : public CustomOpBase {
     int M,\n\
     int K,\n\
     int N,\n\
+    float test1,\n\
+    float test2,\n\
     int ac2zero,\n\
     int bc2zero,\n\
     float scale_a,\n\
@@ -154,4 +158,4 @@ class CustomGemm : public CustomOpBase {
 }  // namespace ops
 }  // namespace vx
 }  // namespace tim
-#endif
\ No newline at end of file
+#endif
diff --git a/samples/custom_op_test/custom_op_test.cc b/samples/custom_op_test/custom_op_test.cc
index 07441ca..d4f7aaa 100644
--- a/samples/custom_op_test/custom_op_test.cc
+++ b/samples/custom_op_test/custom_op_test.cc
@@ -65,7 +65,7 @@ void custom_gemm_single_test(){
     a_tensor->CopyDataToTensor(a_data.data(), a_data.size() * sizeof(float));
     b_tensor->CopyDataToTensor(b_data.data(), b_data.size() * sizeof(float));
 
-    tim::vx::ops::CustomGemm::ParamTuple tuple_list(2,6,6,0,0,1.0,0,1.0,0,1.0,0);
+    tim::vx::ops::CustomGemm::ParamTuple tuple_list(2,6,6,5.0,6.0,0,0,1.0,0,1.0,0,1.0,0);
 
     auto op = graph->CreateOperation<tim::vx::ops::CustomGemm>(
         false,false,tuple_list);
@@ -143,7 +143,7 @@ void custom_gemm_op_and_add_op_test(){
     auto op_add = graph->CreateOperation<tim::vx::ops::AddN>(2);
     (*op_add).BindInputs({a_tensor, b_tensor}).BindOutputs({c_tensor});
 
-    tim::vx::ops::CustomGemm::ParamTuple tuple_list(2,6,6,0,0,1.0,0,1.0,0,1.0,0);
+    tim::vx::ops::CustomGemm::ParamTuple tuple_list(2,6,6,5.0,6.0,0,0,1.0,0,1.0,0,1.0,0);
 
     auto op_gemm = graph->CreateOperation<tim::vx::ops::CustomGemm>(
         false,false,tuple_list);
@@ -210,7 +210,7 @@ void custom_gemm_op_and_custom_gemm_op_test(){
     b_tensor->CopyDataToTensor(b_data.data(), b_data.size() * sizeof(float));
     d_tensor->CopyDataToTensor(d_data.data(), d_data.size() * sizeof(float));
 
-    tim::vx::ops::CustomGemm::ParamTuple tuple_list(2,2,2,0,0,1.0,0,1.0,0,1.0,0);
+    tim::vx::ops::CustomGemm::ParamTuple tuple_list(2,2,2,5.0,6.0,0,0,1.0,0,1.0,0,1.0,0);
 
     auto op_gemm = graph->CreateOperation<tim::vx::ops::CustomGemm>(
         false,false,tuple_list);
@@ -239,4 +239,4 @@ int main(){
     custom_gemm_op_and_add_op_test();
     custom_gemm_op_and_custom_gemm_op_test();
     return 1;
-}
\ No newline at end of file
+}

```

sunshinemyson, 2022-05-06:
Fixed in #385

closed 2022-05-06 by sunshinemyson