VeriSilicon/TIM-VX #387: [QST] custom op : Some question about kernel resource input data type

zhnin, 2022-05-05:
Hi: I have some question about custom op kernel resource.
1. Can I use `__global float4 *inputA` or `__global float *inputA` instead of `__read_only image2d_t inputA`, If so, how should I get the data from inputA (`inputA[0]`?)
2. Whether the `__read_only image3d_t inputA` or `__read_only image1d_t inputA`  data type supported?

sherlock9991, 2022-05-06:
> 

Hi zhnin,

1. of course you can.
2. image2d_array_t is the 3d image. There isn't any image3d_t or image2d_t.

zhnin, 2022-05-06:
example code: https://github.com/VeriSilicon/TIM-VX/blob/main/samples/custom_op_test/custom_gemm.h#L62-L93

In this example, kernel resource function argument and read/write method is :
```
// function arguments
__read_only image2d_t   inputA,
__write_only image2d_t  output,

// read data
int2 coord = (int2)(0, 0);
float4 tempA0 = read_imagef(inputA, coord);  // input[0] = 1
// write data
write_imagef(output, coord, tempA0);         // output[0] = 1
```
I want to process multi dims input(1, 2, 3, 4,.... ), so I try to use float* replace image2d_t, and use index to get value, but result value is wrong
```
// function arguments
__global float *inputA,
__global float *output,

// read data
float tempA0 = inputA[0];                       // input[0] = 1
// write data
output[0] = tempA0;                             // got the wrong result, got 0
```
Could you give me some suggest, thanks very much.

sherlock9991, 2022-05-09:
> inputA
if you want to use a pointer in a kernel, like __global float *inputA, you should define  a vx_array to the kernel parameter in the host side.
Also you can find the address of the image2d_t,

// function arguments
__read_only image2d_t   inputA,
__write_only image2d_t  output,

// define a function to get the pointer of the image2d_t
inline uchar* get_address_from_image2d(image2d_t input)
{
    int8 desc;
    _viv_asm(COPY, desc, input, sizeof(desc));
    return (uchar*)desc.s0;
}

// kernel
__global float* inputA_ptr = (__global float*)get_address_from_image2d(inputA);
__global float* output_ptr = (__global float*)get_address_from_image2d(output);
output_ptr[0] = inuttA_ptr[0];

small-v, 2022-05-09:
> example code: https://github.com/VeriSilicon/TIM-VX/blob/main/samples/custom_op_test/custom_gemm.h#L62-L93
>
> In this example, kernel resource function argument and read/write method is :
>
> ```
> // [...]

A common practice is to create a vx_array object and use pointer access.

zhnin, 2022-05-10:
> inputA
> if you want to use a pointer in a kernel, like __global float *inputA, you should define  a vx_array to the kernel parameter in the host side.

> A common practice is to create a vx_array object and use pointer access.

Hi, thanks for your reply. Do we have demo or easy example: How to create vx_array to kernel parameter and use pointer access?

sherlock9991, 2022-05-11:
> Hi, I have another question to ask: How to define opencl kernel parameter, If the input is 1-dimensional scale tensor of size C ? eg: input type is `float32[3]`

Hi,

1. No such vx_array demo.
2. you can define it with shape [3, 1], dim is 2.

zhnin, 2022-05-16:
Thanks very much.

closed 2022-05-16 by zhnin