VeriSilicon/TIM-VX #391: Is Tile only support 3 dimensions and below?

MESeraph, 2022-05-11:
https://github.com/VeriSilicon/TIM-VX/blob/main/src/tim/vx/internal/src/kernel/evis/tile_evis.c
```c
static vsi_bool _is_supported_axis(vsi_size_t* multiples, vsi_size_t multiples_num)
{
    vsi_size_t i = 0;

    if ( multiples_num < 4)
    {
        return TRUE;
    }
    else if ( multiples_num > 4)
    {
        return FALSE;
    }

    for ( i = 3;  i < multiples_num;  i++)
    {
        if (multiples[i] > 1)
        {
            return FALSE;
        }
    }

    return TRUE;
}
```

small-v, 2022-05-11:
After shape optimization of input, multiples and output，the rank of multiples support 3 dimensions and below

MESeraph, 2022-05-17:
@small-v Will 4D be supported in the future？

small-v, 2022-05-18:
> @small-v Will 4D be supported in the future？

No plans for now, what kind of 4D shape did you use?

MESeraph, 2022-05-29:
In fact, i have not encountered this situation now. I'm just trying to make the operator coverage more comprehensive.
Thank You!

closed 2022-05-29 by MESeraph