VeriSilicon/TIM-VX #398: Key issue about TIM-VX's easy use!

2050airobert, 2022-05-17:
Hi，
     Two question：
     1、TIM-VX/ops   TIM-VX/internal/ops  这两个目录下的算子有什么区别呢？是不是TIM-VX/ops 就可以直接让用户跑模型的时候拿来当做基础算子搭建自己的网络模型呢？
     2、使用TIM-VX 必须用acuity tool产生的模型吗？acuity tool就是amlogic提供的这套给用户使用的工具，还是amlogic在TIM-VX上又包了一层，就像tengine一样？
     3、可不可以不使用acuity tool 或者tengine 这种又封装了一层的包，直接使用TIM-VX ，自己在TIM-VX源码工程中搭建自己的网络，就像 TIM-VX\samples\lenet\lenet_asymu8.cc TIM-VX\samples\lenet_lite\lenet_lite_asymu8.cc一样，自己cc文件实现复杂网络比如resnet50 等？
BR

sunshinemyson, 2022-05-18:
Hi @2050airobert ，

1、TIM-VX/ops TIM-VX/internal/ops 这两个目录下的算子有什么区别呢？是不是TIM-VX/ops 就可以直接让用户跑模型的时候拿来当做基础算子搭建自己的网络模型呢？
[VSI: tim-vx/ops 是对internal/ops的一个API封装。TIM-VX/ops是开放出来搭建网络的]
2、使用TIM-VX 必须用acuity tool产生的模型吗？acuity tool就是amlogic提供的这套给用户使用的工具，还是amlogic在TIM-VX上又包了一层，就像tengine一样？
[VSI: Acuity tool 是VSI开发的模型转换和优化工具。你可以从AML取得（我们的客户）。也可以使用我们开放的PIP 包 - https://pypi.org/project/acuitylite/。 除了使用acuity转换模型，你也可以直接使用vx-delegate来直接运行tflite模型(可以参考一下我们的CI设定）]
3、可不可以不使用acuity tool 或者tengine 这种又封装了一层的包，直接使用TIM-VX ，自己在TIM-VX源码工程中搭建自己的网络，就像 TIM-VX\samples\lenet\lenet_asymu8.cc TIM-VX\samples\lenet_lite\lenet_lite_asymu8.cc一样，自己cc文件实现复杂网络比如resnet50 等？
[VSI: 是可以这样做的。只是对于复杂网络使用tim-vx的API直接来搭建会显得比较繁琐]
4、如果上述问题3 可行，那么我只要在 TIM-VX修改代码即可，不需要依赖其他的转换和优化工具了吧？比如acuity covert or optimization tool etc？right？
[VSI: 是的，看你实际的需求. 我建议使用tflite格式的模型，这样你可以和tflite-runtime的实现做精度对比]

sunshinemyson, 2022-06-27:
@zjd1988 ,

你可以参考https://tensorflow.google.cn/lite/performance/quantization_spec 这里面提到的int8的op列表，他这里面的OP我们都可以支持量化成U8_Asymm.

后面我们会再弄一个文档出来。

zjd1988, 2022-06-27:
@sunshinemyson  嗯嗯，好的，谢谢啦。

closed 2022-09-28 by sunshinemyson