VeriSilicon/TIM-VX #407: Get the data from INPUT tensor

MESeraph, 2022-05-29:
[image]
I need the data of INPUT tensor.  
But I see that you have added restrictions on INPUT/OUTPUT tensor recently.
If it is removed, will it have other effects?

sunshinemyson, 2022-05-29:
@MESeraph ,

Is it possible use CopyDataToTensor() in your case? In principle, we should keep  original API definition. But this change is raised by @gdh1995 who would like to use "TENSOR_HANDLE", I need to make a PR and send you and @gdh1995 to review.

MESeraph, 2022-05-29:
@sunshinemyson 
CopyDataToTensor() will not used  according to the current structure of my case.

gdh1995, 2022-05-29:
Hello. In the old APIs, `data_` stores a pointer to original input data blob, which is a "wild pointer" now that a user may free the memory block immediately after `CreateTensor` returns.

Therefore, if you don't need `CopyDataToTensor`, but just want to access the input blob again, then I suggest you change `CreateTensor({::Input, ...}, input_ptr)` into `CreateIOTensor(...)`. The new API keeps `data_` there, and it can work almost the same as `CreateTensor`, if only you don't modify or release the input blob on `input_ptr`.

If your tensor is not `TensorAttribute::INPUT`, could you share your detailed use case then?

MESeraph, 2022-05-30:
@gdh1995 Thank You! I Got.
I need modify  input data of some Op. For example，indices of onehot/gatherelements, TIM-VX is not support negative.
But it just be used in my Op unittest.

gdh1995, 2022-05-30:
I'm not sure what you meant. But if just to modify input data and re-run forwarding, you may call `Graph::CreateIOTensor`, and then call `Tensor::FlushCacheForHandle` everytime you have modified the memory area of a input blob.

gdh1995, 2022-05-30:
BTW, as far as I've understand from TIM-VX source code, with the old APIs, if you modified the memory area of a input tensor's `data_`, the real input data used in graph forwarding would not be updated. You'll have to use `CopyDataToTensor` to sync your modification.

MESeraph, 2022-05-30:
@gdh1995 Thank you! I will check how use these APIs.

closed 2022-06-07 by MESeraph