VeriSilicon/TIM-VX #408: What are the limitations of TopK support?

MESeraph, 2022-06-07:
When I use a TopK Op with 512*1 input, can not find valid kernel.
I find some kernel limited in [topk_cl.c](https://github.com/VeriSilicon/TIM-VX/blob/main/src/tim/vx/internal/src/kernel/cl/topk_cl.c)
[image]
According to  num_stages:
```int32_t num_stages = (int32_t)ceil(log10(block_size / 2.0f) / log10(2.0f));```
Only 256 or less input is supported.
And only FP32/U32/I32 is supported. Not support FP16.
**Are there any other limitations?  Are there plan to eliminate these limitations?**

Thanks

sunshinemyson, 2022-06-09:
@MESeraph ,

Sorry for late reply. In current version, the limitation should num_stages <= 6 where input vector size <= 128.
FP16 should already supported, you can check _query(), FP16 supported by Float32 kernel.

MESeraph, 2022-06-11:
I got. Thank you!

closed 2022-06-11 by MESeraph