VeriSilicon/TIM-VX #418: Why the evis verson of pre-process-rgb fails to register

gdh1995, 2022-06-27:
I want to use the `PRE_PROCESS` node to convert `nhwc-uint8-from-cv_imread` into `nchw-uint8-quant_per_tensor`, but with the prebuilt SDK of 6.4.10.2, an A311D board reports such an error:
``` log
Kernel "com.vivantecorp.extension.evis.pre_process_rgb_copy_U8toU8" does not exist
E [TIM-VX/src/tim/vx/internal/src/kernel/vsi_nn_kernel.c:_kernel_init_obj:757]Add parameter 0 to kernel com.vivantecorp.extension.evis.pre_process_rgb_copy_U8toU8 fail. with -12.
E [TIM-VX/src/tim/vx/internal/src/kernel/vsi_nn_kernel.c:_kernel_init_obj:770]Finalize kernel com.vivantecorp.extension.evis.pre_process_rgb_copy_U8toU8 fail with -12.
E [TIM-VX/src/tim/vx/internal/src/kernel/vsi_nn_kernel.c:_kernel_init_obj:775]Remove kernel com.vivantecorp.extension.evis.pre_process_rgb_copy_U8toU8 fail with -10.
E [TIM-VX/src/tim/vx/internal/src/kernel/vsi_nn_kernel.c:vsi_nn_kernel_create_node:873]Register client kernel com.vivantecorp.extension.evis.pre_process_rgb_copy_U8toU8 fail with -10.
```

I've debugged the code, but it seems `obj = vxAddKernelInProgram(...)` has contains some errors - an inserted `vxGetStatus((vx_reference)obj)` returns `-1`, although `obj` is not NULL.

The CPU version named `com.vivantecorp.extension.cpu.pre_process_rgb_sw` can be created by `vxAddUserKernel` without any error.

So I wonder why the evis version breaks, and how I can work around it?

sunshinemyson, 2022-06-27:
@gdh1995 ，

Thanks for report this. Can you share your patch with a PR? We'd like try it locally. Thanks

gdh1995, 2022-06-27:
Um, I didn't make any meaningful change. All the logic keeps the same as commit f8741b4 .

My test code looks like:
``` cpp
// src/tim/vx/internal/src/kernel/vsi_nn_kernel.c:523
static vsi_status _gpu_register
{
    // ...
    obj = vxAddKernelInProgram(
        program,
        info->name,
        info->enumeration,
        info->numParams,
        info->validate,
        info->initialize,
        info->deinitialize
        );
    // region: I only added these 5 lines
    status = vxGetStatus( (vx_reference)obj );
    if (VSI_SUCCESS != status)
    {
        VSILOGE( "vxAddKernelInProgram %s fail: %d.", info->name, (int) status);
    }
    // endregion
    if( obj )
    {
        status = _kernel_init_obj( info, obj );
        // here status is -10.
        //vxReleaseKernel( &obj );
    }
}
```

sherlock9991, 2022-06-28:
> Um, I didn't make any meaningful change. All the logic keeps the same as commit [f8741b4](https://github.com/VeriSilicon/TIM-VX/commit/f8741b4704ab9caffd49f6fc15f72616962ba1d1) .
> 
> My test code looks like:
> 
> ```c++
> // src/tim/vx/internal/src/kernel/vsi_nn_kernel.c:523
> static vsi_status _gpu_register
> {
>     // ...
>     obj = vxAddKernelInProgram(
>         program,
>         info->name,
>         info->enumeration,
>         info->numParams,
>         info->validate,
>         info->initialize,
>         info->deinitialize
>         );
>     // region: I only added these 5 lines
>     status = vxGetStatus( (vx_reference)obj );
>     if (VSI_SUCCESS != status)
>     {
>         VSILOGE( "vxAddKernelInProgram %s fail: %d.", info->name, (int) status);
>     }
>     // endregion
>     if( obj )
>     {
>         status = _kernel_init_obj( info, obj );
>         // here status is -10.
>         //vxReleaseKernel( &obj );
>     }
> }
> ```

Hi,

It looks like the first parameter is invalid, please check the data type

gdh1995, 2022-06-28:
The original code without `vxGetStatus( (vx_reference)obj );` still stopped in the below `_kernel_init_obj(info, obj)`, in which `vxAddParameterToKernel` returns `-12`, and then `vxRemoveKernel` return `-10`.

sherlock9991, 2022-06-28:
vxAddKernelInProgram returns failure, please check the parameters info and program.