VeriSilicon/TIM-VX #423: avgeragepool compute error

MESeraph, 2022-07-04:
I have a onnx case [averagepool](https://github.com/VeriSilicon/TIM-VX/files/9037105/averagepool.zip ) UT that can passed  on onnxruntime.

I add a UT for averagepool based on the onnx case : (need put the [averagepool](https://github.com/VeriSilicon/TIM-VX/files/9037105/averagepool.zip) folder under **host_build/sr/tim**)
avg_pool_onnx_test.cc
```
/****************************************************************************
*
*    Copyright (c) 2021 Vivante Corporation
*
*    Permission is hereby granted, free of charge, to any person obtaining a
*    copy of this software and associated documentation files (the "Software"),
*    to deal in the Software without restriction, including without limitation
*    the rights to use, copy, modify, merge, publish, distribute, sublicense,
*    and/or sell copies of the Software, and to permit persons to whom the
*    Software is furnished to do so, subject to the following conditions:
*
*    The above copyright notice and this permission notice shall be included in
*    all copies or substantial portions of the Software.
*
*    THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
*    IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
*    FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
*    AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
*    LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
*    FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER
*    DEALINGS IN THE SOFTWARE.
*
*****************************************************************************/
#include "tim/vx/context.h"
#include "tim/vx/graph.h"
#include "tim/vx/ops/pool2d.h"
#include <iostream>
#include <fstream>
#include <numeric>
#include "gtest/gtest.h"
#include "test_utils.h"

union float32_bits {
  unsigned int u;
  float f;
};

struct Float16 {
  uint16_t val;

  Float16() : val(0) {}
  explicit Float16(uint16_t x) : val(x) {}
  //explicit MLFloat16(float f);

  float ToFloat() const
  {
    const float32_bits magic = { 113 << 23 };
    const unsigned int shifted_exp = 0x7c00 << 13; // exponent mask after shift
    float32_bits o;

    o.u = (val & 0x7fff) << 13;             // exponent/mantissa bits
    unsigned int exp = shifted_exp & o.u;   // just the exponent
    o.u += (127 - 15) << 23;                // exponent adjust

    // handle exponent special cases
    if (exp == shifted_exp) {     // Inf/NaN?
        o.u += (128 - 16) << 23;    // extra exp adjust
    } else if (exp == 0) {        // Zero/Denormal?
        o.u += 1 << 23;             // extra exp adjust
        o.f -= magic.f;             // renormalize
    }

    o.u |= (val & 0x8000) << 16;    // sign bit
    return o.f;
  }

  operator float() const {
    return ToFloat();
  }
};

TEST(AVG, shape_11_11_12_11_fp16_kernel_5_stride_1) {
    auto ctx = tim::vx::Context::Create();
    auto graph = ctx->CreateGraph();

    tim::vx::ShapeType in_shape({11, 11, 12, 11});
    tim::vx::ShapeType out_shape({8, 9, 12, 11});
    tim::vx::TensorSpec input_spec(tim::vx::DataType::FLOAT16,
                            in_shape, tim::vx::TensorAttribute::INPUT);
    tim::vx::TensorSpec output_spec(tim::vx::DataType::FLOAT16,
                            out_shape, tim::vx::TensorAttribute::OUTPUT);

    auto input_tensor = graph->CreateTensor(input_spec);
    auto output_tensor = graph->CreateTensor(output_spec);

    
    auto in_count = std::accumulate(in_shape.begin(),
                                        in_shape.end(), 1, std::multiplies<int64_t>());
    u_int16_t in_data[in_count] = {0};
    std::ifstream in("averagepool/test_data_set0/input_0.bin", std::ios::in | std::ios::binary);
    if(in.is_open())
    {
        in.read((char *) &in_data, sizeof(in_data)*sizeof(u_int16_t));
        in.close();
    }
    else
    {
        std::cout<<"open input_0.bin fail!"<<std::endl;
    }
    

    auto out_count = std::accumulate(out_shape.begin(),
                                        out_shape.end(), 1, std::multiplies<int64_t>());
    u_int16_t golden[out_count] = {0};
    std::vector<float> golden_float(out_count);
    std::ifstream out("averagepool/test_data_set0/output_0.bin", std::ios::in | std::ios::binary);
    if(out.is_open())
    {
        out.read((char *) &golden, sizeof(golden)*sizeof(u_int16_t));
        for(auto i=0; i<out_count; i++)
        {   
            golden_float[i] = Float16(golden[i]).ToFloat();
        }
        out.close();
    }
    else
    {
        std::cout<<"open output_0.bin fail!"<<std::endl;
    }
    
    EXPECT_TRUE(input_tensor->CopyDataToTensor((void*)in_data, sizeof(in_data)*sizeof(u_int16_t)));
    std::array<uint32_t, 4> pad = {0, 1, 1, 1};
    std::array<uint32_t, 2> ksize = {5, 5};
    std::array<uint32_t, 2> stride = {1, 1};
    auto op = graph->CreateOperation<tim::vx::ops::Pool2d>(tim::vx::PoolType::AVG, pad, ksize, stride, tim::vx::RoundType::CEILING);
    (*op).BindInputs({input_tensor}).BindOutputs({output_tensor});

    EXPECT_TRUE(graph->Compile());
    EXPECT_TRUE(graph->Run());

    std::vector<u_int16_t> output(sizeof(golden)*sizeof(u_int16_t));
    std::vector<float> output_float(out_count);
    EXPECT_TRUE(output_tensor->CopyDataFromTensor(output.data()));
    for(auto i=0; i<out_count; i++)
    {   
        output_float[i] = Float16(output[i]).ToFloat();
    }
    EXPECT_TRUE(ArraysMatch(golden_float, output_float,1e-4f));
}
```
After running, it is inconsistent with the expected result.

[image]

sunshinemyson, 2022-07-06:
Thanks for feedback. We will handle it.

xuke537, 2022-07-07:
@MESeraph  How can I get your input golden data?

xuke537, 2022-07-21:
@MESeraph We should take care the attr count_include_pad of ONNX avgpool.
From ONNX op doc

> count_include_pad: int64
> Whether include pad pixels when calculating values for the edges. Default is 0, doesn't count include pad.

For TIM-VX, PoolType::AVG will count include pad, PoolType::AVG_ANDROID doesn't count include pad.
In this case, count_include_pad is default value(0).  So the pool type should be AVG_ANDROID.

MESeraph, 2022-07-21:
I just re-verified, UT has passed. 
Thank you!

closed 2022-07-21 by MESeraph